open62541/open62541 #454: Use size_t for all array lengths

jpfr, 2015-10-23:
Hey team,

our current implementation is a 1:1 adaptation of the OPC UA standard. Array lengths are int32 values and -1 indicates the absence of an array (which is different from an array of length 0).

Signed array lengths are a bad idea. They enable lots of subtle bugs.
https://www.securecoding.cert.org/confluence/display/c/INT01-C.+Use+rsize_t+or+size_t+for+all+integer+values+representing+the+size+of+an+object
So we should switch to size_t array length before the 0.2 is frozen.

**What will change**
Easy. UA_String would look like that:
```c
typedef struct {
    size_t length;
    UA_Byte *data;
} UA_String;
```
But we still need to differentiate between absent and zero-length arrays.
For that, we can use a sentinel pointer value, e.g.
```c
#define UA_ZERO_LENGTH_ARRAY_SENTINEL (void*)0x01
```
But only a few places will check for `0x01`. For-loops can be used as usual, since the array length is zero and the pointer is never dereferenced.

**How is that better**
- We avoid many many subtle bugs with integer overflows, comparison, ...
- We can replace every _init function with `memset(ptr, sizeof(type))` and every _new function with `calloc(1, sizeof(type))`. All we need to do is zero out the region to initialize the type.
- `UA_Array_new` would become more malloc-like and return a pointer
- Many double-conditions in if-clauses (if ptr != 0 and length > 0) will become easier

Do you have any more arguments in favor or against this proposition?

Stasik0, 2015-10-24:
well it is okay for me, a sentinel pointer is 1337-enough for us.

jpfr, 2015-12-04:
Done as of 2f4a056607fd10a8fa2b396c5b3fd4e580aeeb87

closed 2015-12-04 by jpfr