open62541/open62541 #458: sample server crash (current master branch)

StalderT, 2015-10-26:
Hello,

In some cases the sample server crash.

Step to reproduce : 

1) Start server
2) Start UA_Expert and connect to server
3) Add "current time" node to Data Access View
4) Right click on "current time" in Data Access View and click "Subscription Settings".
5) Change "Publishing Interval" to 100 and click "OK".
6) Delete "current time" node from Data Access View.
7) Add "current time" node to Data Access View

The server crash.

Valgrind results (for info, the first invalid read just happens when deleting the node, step 6) :
==1805== Invalid read of size 4
==1805==    at 0x405A463: Subscription_timedUpdateNotificationsJob (open62541.c:14491)
==1805==    by 0x4050DC6: processJobs (open62541.c:8919)
==1805==    by 0x4050FA5: processRepeatedJobs (open62541.c:9189)
==1805==    by 0x4050FA5: UA_Server_run_mainloop (open62541.c:9433)
==1805==    by 0x4051109: UA_Server_run (open62541.c:9507)
==1805==    by 0x804B454: main (server.c:428)
==1805==  Address 0x45dbce8 is 48 bytes inside a block of size 100 free'd
==1805==    at 0x402A7A8: free (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==1805==    by 0x4051318: SubscriptionManager_deleteSubscription (open62541.c:14873)
==1805==    by 0x40513B7: Service_DeleteSubscriptions (open62541.c:14207)
==1805==    by 0x4040B73: invoke_service (open62541.c:8109)
==1805==    by 0x40500F0: processMSG (open62541.c:8293)
==1805==    by 0x4050ACB: UA_Server_processBinaryMessage (open62541.c:8383)
==1805==    by 0x4050E1A: processJobs (open62541.c:8907)
==1805==    by 0x4050F2E: UA_Server_run_mainloop (open62541.c:9464)
==1805==    by 0x4051109: UA_Server_run (open62541.c:9507)
==1805==    by 0x804B454: main (server.c:428)
==1805== 

After step 7

==1805== Invalid read of size 4
==1805==    at 0x405A46A: Subscription_timedUpdateNotificationsJob (open62541.c:14495)
==1805==    by 0x4050DC6: processJobs (open62541.c:8919)
==1805==    by 0x4050FA5: processRepeatedJobs (open62541.c:9189)
==1805==    by 0x4050FA5: UA_Server_run_mainloop (open62541.c:9433)
==1805==    by 0x4051109: UA_Server_run (open62541.c:9507)
==1805==    by 0x804B454: main (server.c:428)
==1805==  Address 0x45dbd18 is 24 bytes before a block of size 48 in arena "client"
==1805== 
==1805== Invalid read of size 4
==1805==    at 0x4059F22: MonitoredItem_QueuePushDataValue (open62541.c:14705)
==1805==    by 0x405A481: Subscription_timedUpdateNotificationsJob (open62541.c:14496)
==1805==    by 0x4050DC6: processJobs (open62541.c:8919)
==1805==    by 0x4050FA5: processRepeatedJobs (open62541.c:9189)
==1805==    by 0x4050FA5: UA_Server_run_mainloop (open62541.c:9433)
==1805==    by 0x4051109: UA_Server_run (open62541.c:9507)
==1805==    by 0x804B454: main (server.c:428)
==1805==  Address 0x98 is not stack'd, malloc'd or (recently) free'd
==1805== 
==1805== 
==1805== Process terminating with default action of signal 11 (SIGSEGV)
==1805==  Access not within mapped region at address 0x98
==1805==    at 0x4059F22: MonitoredItem_QueuePushDataValue (open62541.c:14705)
==1805==    by 0x405A481: Subscription_timedUpdateNotificationsJob (open62541.c:14496)
==1805==    by 0x4050DC6: processJobs (open62541.c:8919)
==1805==    by 0x4050FA5: processRepeatedJobs (open62541.c:9189)
==1805==    by 0x4050FA5: UA_Server_run_mainloop (open62541.c:9433)
==1805==    by 0x4051109: UA_Server_run (open62541.c:9507)
==1805==    by 0x804B454: main (server.c:428)
==1805==  If you believe this happened as a result of a stack
==1805==  overflow in your program's main thread (unlikely but
==1805==  possible), you can try to increase the size of the
==1805==  main thread stack using the --main-stacksize= flag.
==1805==  The main thread stack size used in this run was 8388608.
==1805== 
==1805== HEAP SUMMARY:
==1805==     in use at exit: 82,868 bytes in 1,287 blocks
==1805==   total heap usage: 12,096 allocs, 10,809 frees, 173,911,978 bytes allocated
==1805== 
==1805== LEAK SUMMARY:
==1805==    definitely lost: 0 bytes in 0 blocks
==1805==    indirectly lost: 0 bytes in 0 blocks
==1805==      possibly lost: 0 bytes in 0 blocks
==1805==    still reachable: 82,868 bytes in 1,287 blocks
==1805==         suppressed: 0 bytes in 0 blocks
==1805== Rerun with --leak-check=full to see details of leaked memory
==1805== 
==1805== For counts of detected and suppressed errors, rerun with: -v
==1805== ERROR SUMMARY: 60 errors from 3 contexts (suppressed: 0 from 0)
Segmentation fault

GDB results:

Program received signal SIGSEGV, Segmentation fault.
0xb7fc2f22 in MonitoredItem_QueuePushDataValue (server=0x804d008, monitoredItem=0x19b201d1) at /data/linux/open62541/src/build/open62541.c:14705
14705	    if(!monitoredItem || monitoredItem->lastSampled + monitoredItem->samplingInterval > UA_DateTime_now())

Bests regards,

Thomas

Stasik0, 2015-12-14:
@ichrispa Chris, could you please look into this one?

StalderT, 2015-12-15:
With current master branch : 

```
==1047== Memcheck, a memory error detector
==1047== Copyright (C) 2002-2015, and GNU GPL'd, by Julian Seward et al.
==1047== Using Valgrind-3.11.0 and LibVEX; rerun with -h for copyright info
==1047== Command: ./a.out
==1047== 
[12/15/2015 09:52:22.882] info/network	TCP network layer listening on opc.tcp://devel:16664
[12/15/2015 09:52:27.136] info/network	New Connection 4 over TCP from 192.168.1.4:16664
[12/15/2015 09:52:27.198] info/network	Closing the Connection 4
[12/15/2015 09:52:27.201] info/network	New Connection 5 over TCP from 192.168.1.4:16664
==1047== Invalid read of size 4
==1047==    at 0x40551C3: Subscription_timedUpdateNotificationsJob (open62541.c:14307)
==1047==    by 0x405644C: processJobs (open62541.c:8686)
==1047==    by 0x405662D: processRepeatedJobs (open62541.c:8958)
==1047==    by 0x405662D: UA_Server_run_mainloop (open62541.c:9207)
==1047==    by 0x40567A9: UA_Server_run (open62541.c:9281)
==1047==    by 0x804B35D: main (in /data/linux/open62541/src/open62541-master/examples/a.out)
==1047==  Address 0x49d46d8 is 48 bytes inside a block of size 104 free'd
==1047==    at 0x402B2F8: free (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==1047==    by 0x4056F38: SubscriptionManager_deleteSubscription (open62541.c:14663)
==1047==    by 0x4056FD7: Service_DeleteSubscriptions (open62541.c:14037)
==1047==    by 0x4055A88: processMSG (open62541.c:8124)
==1047==    by 0x405616C: UA_Server_processBinaryMessage (open62541.c:8180)
==1047==    by 0x40564AF: processJobs (open62541.c:8674)
==1047==    by 0x40565B6: UA_Server_run_mainloop (open62541.c:9238)
==1047==    by 0x40567A9: UA_Server_run (open62541.c:9281)
==1047==    by 0x804B35D: main (in /data/linux/open62541/src/open62541-master/examples/a.out)
==1047==  Block was alloc'd at
==1047==    at 0x402A11C: malloc (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==1047==    by 0x40539C5: UA_Subscription_new (open62541.c:14110)
==1047==    by 0x4053F3A: Service_CreateSubscription (open62541.c:13778)
==1047==    by 0x4055A88: processMSG (open62541.c:8124)
==1047==    by 0x405616C: UA_Server_processBinaryMessage (open62541.c:8180)
==1047==    by 0x40564AF: processJobs (open62541.c:8674)
==1047==    by 0x40565B6: UA_Server_run_mainloop (open62541.c:9238)
==1047==    by 0x40567A9: UA_Server_run (open62541.c:9281)
==1047==    by 0x804B35D: main (in /data/linux/open62541/src/open62541-master/examples/a.out)
==1047== 

==1047== Invalid read of size 4
==1047==    at 0x40551CA: Subscription_timedUpdateNotificationsJob (open62541.c:14311)
==1047==    by 0x405644C: processJobs (open62541.c:8686)
==1047==    by 0x405662D: processRepeatedJobs (open62541.c:8958)
==1047==    by 0x405662D: UA_Server_run_mainloop (open62541.c:9207)
==1047==    by 0x40567A9: UA_Server_run (open62541.c:9281)
==1047==    by 0x804B35D: main (in /data/linux/open62541/src/open62541-master/examples/a.out)
==1047==  Address 0x49d470c is 20 bytes inside a block of size 24 in arena "client"
==1047== 
==1047== Invalid read of size 4
==1047==    at 0x405429E: Subscription_updateNotifications (open62541.c:14171)
==1047==    by 0x405644C: processJobs (open62541.c:8686)
==1047==    by 0x405662D: processRepeatedJobs (open62541.c:8958)
==1047==    by 0x405662D: UA_Server_run_mainloop (open62541.c:9207)
==1047==    by 0x40567A9: UA_Server_run (open62541.c:9281)
==1047==    by 0x804B35D: main (in /data/linux/open62541/src/open62541-master/examples/a.out)
==1047==  Address 0x49d46c8 is 0 bytes after an unallocated block of size 32 in arena "client"
==1047== 
==1047== Invalid read of size 4
==1047==    at 0x40542A1: Subscription_updateNotifications (open62541.c:14171)
==1047==    by 0x405644C: processJobs (open62541.c:8686)
==1047==    by 0x405662D: processRepeatedJobs (open62541.c:8958)
==1047==    by 0x405662D: UA_Server_run_mainloop (open62541.c:9207)
==1047==    by 0x40567A9: UA_Server_run (open62541.c:9281)
==1047==    by 0x804B35D: main (in /data/linux/open62541/src/open62541-master/examples/a.out)
==1047==  Address 0x49d46d0 is 8 bytes after an unallocated block of size 32 in arena "client"
==1047== 
==1047== Invalid read of size 4
==1047==    at 0x40542A4: Subscription_updateNotifications (open62541.c:14171)
[... 333 lines cut ...]
==1047==    by 0x405662D: UA_Server_run_mainloop (open62541.c:9207)
==1047==    by 0x40567A9: UA_Server_run (open62541.c:9281)
==1047==    by 0x804B35D: main (in /data/linux/open62541/src/open62541-master/examples/a.out)
==1047==  If you believe this happened as a result of a stack
==1047==  overflow in your program's main thread (unlikely but
==1047==  possible), you can try to increase the size of the
==1047==  main thread stack using the --main-stacksize= flag.
==1047==  The main thread stack size used in this run was 8388608.
==1047== 
==1047== HEAP SUMMARY:
==1047==     in use at exit: 98,102 bytes in 1,137 blocks
==1047==   total heap usage: 37,639 allocs, 36,502 frees, 735,952,904 bytes allocated
==1047== 
==1047== LEAK SUMMARY:
==1047==    definitely lost: 128 bytes in 5 blocks
==1047==    indirectly lost: 264 bytes in 8 blocks
==1047==      possibly lost: 0 bytes in 0 blocks
==1047==    still reachable: 97,710 bytes in 1,124 blocks
==1047==         suppressed: 0 bytes in 0 blocks
==1047== Rerun with --leak-check=full to see details of leaked memory
==1047== 
==1047== For counts of detected and suppressed errors, rerun with: -v
==1047== ERROR SUMMARY: 961 errors from 20 contexts (suppressed: 0 from 0)
Segmentation fault
```

jpfr, 2016-01-28:
Fixed in 28f8fca.
Sorry this took so long.

commit 28f8fca7 (2016-01-28): fix #458
closed 2016-01-28 by jpfr