open62541/open62541 #459: Server leaks with multithreading option activated

FlorianPalm, 2015-10-27:
==30773== HEAP SUMMARY:
==30773==     in use at exit: 280 bytes in 2 blocks
==30773==   total heap usage: 5,463 allocs, 5,461 frees, 6,550,797 bytes allocated
==30773== 
==30773== 152 bytes in 1 blocks are possibly lost in loss record 2 of 2
==30773==    at 0x402D109: calloc (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==30773==    by 0x401117E: allocate_dtv (dl-tls.c:296)
==30773==    by 0x40118EB: _dl_allocate_tls (dl-tls.c:460)
==30773==    by 0x40877A2: pthread_create@@GLIBC_2.1 (allocatestack.c:589)
==30773==    by 0x4257C74: call_rcu_data_init (urcu-call-rcu-impl.h:435)
==30773==    by 0x42590F9: get_default_call_rcu_data_memb (urcu-call-rcu-impl.h:562)
==30773==    by 0x4259163: get_call_rcu_data_memb (urcu-call-rcu-impl.h:591)
==30773==    by 0x42591B7: call_rcu_memb (urcu-call-rcu-impl.h:712)
==30773==    by 0x405649B: UA_NodeStore_replace (ua_nodestore_concurrent.c:278)
==30773==    by 0x404E3CE: UA_Server_editNode (ua_services_attribute.c:437)
==30773==    by 0x404E9FA: Service_AddReferences_single (ua_services_nodemanagement.c:768)
==30773==    by 0x404ED15: UA_Server_addExistingNode (ua_services_nodemanagement.c:75)
==30773== 
==30773== LEAK SUMMARY:
==30773==    definitely lost: 0 bytes in 0 blocks
==30773==    indirectly lost: 0 bytes in 0 blocks
==30773==      possibly lost: 152 bytes in 1 blocks
==30773==    still reachable: 128 bytes in 1 blocks
==30773==         suppressed: 0 bytes in 0 blocks

jpfr, 2015-10-27:
Hey @FlorianPalm,

imho this is an artifact from the liburcu library. It seems the "possibly lost" size is invariant to how many nodes are created / deleted.

WONTFIX for now?

Stasik0, 2015-10-27:
+1 for wontfix since the leak is possible and constant

FlorianPalm, 2015-10-28:
Okay with me

closed 2015-10-28 by Stasik0