open62541/open62541 #464: Can not build open62541.c on MSVC 2015

mentospcg, 2015-11-06:
The first error is :  const UA_QualifiedName myIntegerName = UA_QUALIFIEDNAME(1, "the answer");

It shows: IntelliSense a value of type UA_String cannot be used to initialize an entity of type UA_Int32.

I have been trying for more than a  week, please help me....

ichrispa, 2015-11-06:
Hi @mentospcg,

IntelliSense is more of a suggestive error prediction then an actual error. Try disabling intellisense.

If you get errors from generated namespaces ("String too long" or "line length exceeds..."), remove the 2 ByteString certificates from the Namespace.xml.

After IntelliSense is off, if you still get real errors, feel free to post those and I'll try to help :)

Regards,

ichrispa

HolgerJeromin, 2015-11-06:
Which version of open62541 did you tried?
We build every commit with Visual Studio on appveyor since june:
https://ci.appveyor.com/project/Stasik0/open62541/branch/master

jpfr, 2015-11-08:
Hey @mentospcg,

yes. MSVC 2015 is not fully C99 compliant.

We changed the code to MSVC compatible some time ago in the master branch.
A bleading edge build of the single-file release can be found here: http://open62541.org/releases/a0e73ff02f.zip

Somebody else ran into the same problem before: http://stackoverflow.com/questions/30931865/macro-error-in-c99-and-ms-visual-studio-2015

Also note that IntelliSense is a tool to assist the user. The real error came from the compiler.

closed 2015-11-08 by jpfr