open62541/open62541 #465: Clang installation breaks on travis

jpfr, 2015-11-12:
The clang build exits with
```
==> Pouring swig-3.0.2.mavericks.bottle.tar.gz
  /usr/local/Cellar/swig/3.0.2: 665 files, 6.7M
brew install --HEAD valgrind
==> Checking out svn://svn.valgrind.org/valgrind/trunk
==> Downloading https://gist.githubusercontent.com/jacknagel/cd26a902d72aabd0b51
curl: (22) The requested URL returned error: 404 Not Found
Error: Failed to download resource "valgrind--patch"
Download failed: https://gist.githubusercontent.com/jacknagel/cd26a902d72aabd0b51d/raw/1a61a328a87a728dccbeef0594f6fe335e9bf917/valgrind-sdk-paths-Makefile-am.diff
```

I will disable the clang build for now. We can re-enable by doing all the grunt testing work in a separate branch.

jpfr, 2015-12-06:
Now gcc broke as well. It seems some packages have changed names.
```
$ if [ ${TRAVIS_OS_NAME} == 'linux' ]; then . ./tools/travis_linux_before_install.sh; fi
sudo apt-get install binutils-mingw-w64-i686 gcc-mingw-w64-i686 mingw-w64
Reading package lists... Done
Building dependency tree       
Reading state information... Done

E: Unable to locate package binutils-mingw-w64-i686
E: Unable to locate package gcc-mingw-w64-i686
E: Unable to locate package mingw-w64
```

Stasik0, 2015-12-06:
This is the windows cross-compiler.
Software aging at its best.

HolgerJeromin, 2015-12-06:
The failing builds are from the new infrastructure:
https://blog.travis-ci.com/2015-11-27-moving-to-a-more-elastic-future

closed 2016-02-09 by Stasik0