open62541/open62541 #466: First steps with open62541-server don't work properly

Spezies31, 2015-11-18:
Hello,

i am trying to learn about the usage of open62541 via the Tutorials on open62541.org. Therefore i wanted to do the first steps with an open62541-server. During this Tutorial i encountered many errors for which i haven't any solution. The first error appears while i am trying to verify my build environment. When i am typing the command ":open62541/build> cmake ../ " i will get the following error message:

pi@raspberrypi-y5zOL1oFue ~/open62541/build $ cmake ../
-- The C compiler identification is GNU 4.6.3
-- Check for working C compiler: /usr/bin/gcc
-- Check for working C compiler: /usr/bin/gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Found PythonInterp: /usr/bin/python (found version "2.7.3") 
-- Found Git: /usr/bin/git (found version "1.7.10.4") 
-- Git version: v0.1.0-RC4-784-g5f6beed
-- CMAKE_BUILD_TYPE not given; setting to 'RelWithDebInfo'.
CMake Error at CMakeLists.txt:301 (target_compile_definitions):
  Unknown CMake command "target_compile_definitions".

-- Configuring incomplete, errors occurred!

I handled this by putting an "#" infront of "target_compile_definitions..." in the lines 301, 305 and 306. After this it compiles fine but i get completely different messages from my shell as described in the Tutorial. I think that it could have something to do with the Version of GNU, gcc and Python in the tutorial they use GNU 4.8.3, Python 2.7.8 and Git 2.1.4. I am using GNU 4.6.3, Python 2.7.3 and Git 1.7.10.4 right now. I am working at updating them, because with "apt-get install ..." and "apt-get upgrade" and "apt-get update" i don't get any newer versions.
Like mentioned before it compiles just fine and the minimal server application runs like in the tutorial.

My Second issue appears while compiling "myServer.c" with the added code tu exit cleanly (like in "examples/server_firstSteps.c") . I get following messages after trying to compile "myServer.c":

pi@raspberrypi-y5zOL1oFue ~/myServerApp $ gcc -Wl,-rpath=./ -L./ -I ./include -o myServer myServer.c -lopen62541
myServer.c: In function 'main':
myServer.c:21:10: warning: assignment makes pointer from integer without a cast [enabled by default]
/tmp/ccEMq1qC.o: In function `main':
myServer.c:(.text+0x70): undefined reference to `Logger_Stdout_new'
collect2: ld returned 1 exit status

Again i solved the error by putting "//" infront of "logger = Logger_Stdout_new();" .
I looked in the Header "logger_stdout.h" and recognized that the function is named "Logger_Stdout" and not "Logger_Stdout_new". But nothing changes after changing the name of the function.

It is the first time that i work with a raspberry pi and with open62541 so i would be happy if someone knows how to solve these problems.

greetings 
Spezies31

Stasik0, 2015-11-18:
issue #1 please type "cmake --version", you need a newer CMake Version >3
if you are on an older ubuntu try

```sudo add-apt-repository ppa:kalakris/cmake -y```
```sudo apt-get update```
```sudo apt-get install cmake```

issue #2 the problem is that the logger is not included in the .so file (even if it is in the amalgamated file open62541.c) you need to compile example/logger_stdout.c with your executable like
```gcc -Wl,-rpath=./ -L./ -I ./include -o myServer myServer.c examples/logger_stdout.h -lopen62541```

feel free to edit make a pull-request to fix/extend the tutorials... they are located here https://github.com/acplt/open62541/tree/master/doc

HolgerJeromin, 2015-11-18:
Thanks! Should we enforce 2.8.11 or 3? :-)

uleon, 2015-11-19:
I would always prefer to implement an earliest compatible version philosophy. If both 2.8.11 and 3 make the job then 2.8.11 would be the minimum  required, wouldn't it?

ichrispa, 2015-11-19:
I would highly disadvise using "bleeding edge" features in anything; backwards compatibility says a lot about the quality of a project. I would highly support going for cmake 2.8.x

Stasik0, 2015-11-19:
@ichrispa in this case please idenify what is incompatible btw how to
workaround it :)

Spezies31, 2015-11-25:
Thanks for the many answers.

@Stasik0:
if i put "sudo add-apt-repository ppa:kalakris/cmake -y" in my shell i get the message:
"sudo: add-apt-repository: command not found"
Therefore i cannot try your solution for the first issue but i will search for a way to update cmake to a newer Version. If you know why my shell don't know this command feel free to help me (again).

HolgerJeromin, 2015-11-25:
the command add-apt-repository is ubuntu (and newer debian) specific. Your raspberry seems to run another linux distribution. So you need to look for a way to upgrade your cmake in this distro.

Stasik0, 2015-11-25:
try to follow this guide
http://askubuntu.com/questions/38021/how-to-add-a-ppa-on-a-server

the manual way is described in the 4th answer
http://askubuntu.com/a/38029/168906

Spezies31, 2015-11-25:
I tried to follow the guide. But it doesn't work. I only get the following messages:

pi@raspberrypi-y5zOL1oFue ~ $ sudo apt-get install software-properties-common
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
E: Paket software-properties-common kann nicht gefunden werden.
pi@raspberrypi-y5zOL1oFue ~ $ sudo apt-get install python-software-properties
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
E: Paket python-software-properties kann nicht gefunden werden.
pi@raspberrypi-y5zOL1oFue ~ $ sudo apt-get install apt-file
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
E: Paket apt-file kann nicht gefunden werden.

I try to make the manual way but i have problems with the first step because i don't know what he means by "Visit the PPA's overview page in Launchpad. Look for the heading that reads Adding this PPA to your system and click the Technical details about this PPA link." 

By the way does anybody know how i can check which Systemversion i have installed? I only found "uname -a" but this only calls the Firmwareversion. I'll try my best in the meantime.

Tanks for the help i really appreciate it.

Spezies31, 2015-11-25:
I'm at the point at which i think i can say that the manual way don't work. At least with the guide in the fourth answer. Because i have Raspbian 7.8 installed and no Ubuntu System.

Stasik0, 2015-11-25:
Then just get a newer cmake from their website

Spezies31, 2015-11-25:
Thanks for the awesome help i am happy to say that the first issue is solved.
I installed cmake 2.8.11 with this guide. I think the main problem was me because i was in the wrong directory. The guide install cmake into the directory 

$ /opt/cmake/bin/cmake

but i needed it in the directory $ /usr/bin/cmake.
So back to the first sulotion for the second issue from Stasik0.

When i put the command for the compiler into the shell i get this error message:

pi@raspberrypi-y5zOL1oFue ~/myServerApp $ gcc -Wl,-rpath=./ -L./ -I ./include -o myServer myServer.c examples/logger_stdout.c -lopen62541
gcc: error: examples/logger_stdout.c: Datei oder Verzeichnis nicht gefunden

I understand that the raspberry can't find the directory examples/logger_stdout.c but isn't it a standard directory afterall?
Again thanks for the awesome help.

Spezies31, 2015-11-25:
After I changed the compile command into this:

gcc -Wl,-rpath=./ -L./ -I ./include -o myServer myServer.c /home/pi/open62541/examples/logger_stdout.c -lopen62541

i still get the same error message:

pi@raspberrypi-y5zOL1oFue ~/myServerApp $ gcc -Wl,-rpath=./ -L./ -I ./include -o myServer myServer.c /home/pi/open62541/examples/logger_stdout.c -lopen62541
myServer.c: In function 'main':
myServer.c:21:10: warning: assignment makes pointer from integer without a cast [enabled by default]
/tmp/ccfl9mag.o: In function `main':
myServer.c:(.text+0x70): undefined reference to `Logger_Stdout_new'
collect2: ld returned 1 exit status

Stasik0, 2015-11-25:
As you already noticed there is no _new function -> use Logger_Stdout instead

Spezies31, 2015-11-25:
After i changed the name of the funktion i got this message:

pi@raspberrypi-y5zOL1oFue ~/myServerApp $ gcc -Wl,-rpath=./ -L./ -I ./include -o myServer myServer.c /home/pi/open62541/examples/logger_stdout.c -lopen62541
myServer.c: In function 'main':
myServer.c:21:3: error: too few arguments to function 'Logger_Stdout'
./include/logger_stdout.h:16:16: note: declared here

It makes sense to me because the logger_stdout Function is declared in the header logger_stdout.h like this:

UA_EXPORT void Logger_Stdout (UA_LogLevel level, UA_LogCategory category, const char *msg, ...)

and the Funktion is used in this way:

logger = Logger_Stdout();

clearly there are variables that should be given to the function but how can i find out which one i have to give to it? Am I wrong with this solution?

Stasik0, 2015-11-25:
Compare the code with examples/server_firstSteps.c

Spezies31, 2015-12-01:
I compared the code of the examples/server_firstSteps.c with myServer.c and noticed that they 
don't use the function anymore. So i deleted it an the server is running fine.
The only difference between myServer and the Server in the Tutorial are the Messages i get in the Shell.
When i type ./myServer & into the shell i only get:

pi@raspberrypi-y5zOL1oFue ~/myServerApp $ ./myServer &
[1] 5677

instead of:

:myApp> ./myServer &
[xx/yy/zz aa:bb:cc.dd.ee] info/communication        Listening on opc.tcp://localhost:16664
[1] 2114

Is this because i have no Client communicating with the server yet?
Because i don't understand why my messages are different. The compiling worked fine and there were zero error messages.

Again Thanks for the help

closed 2015-12-02 by Spezies31

rashed043, 2017-10-25:
hello everyboby!! i am not sure this is the right place to ask my question regarding OPC ua in vxworks but any body from you can help me then i will be grateful. My request is : has there any body implemented the OPC ua multithreading in Vxworks?

HolgerJeromin, 2017-10-25:
>  i am not sure this is the right place to ask my question regarding OPC ua in vxworks

Please open a separate issue. 
Commenting in a two year old closed issue will probably not be seen by many devs.