open62541/open62541 #467: Allocating dynamic nodeIds fails if namespace is not 0 or 1

ichrispa, 2015-11-20:
Hi all,

I just tried to write an application where I use namespaces 0, 1 and 2. All requests to UA_NODEID_NUMERIC(2,0) for dynamic node allocation fail (BADNODEIDINVALID).

It's caused via ua_services_nodemanagement.c:16... shouldn't this part only apply to external namespaces?

Regards,

ichrispa

jpfr, 2015-11-21:
The namespace needs to exist in order to use it.
```c
/** @brief Add a new namespace to the server. Returns the index of the new namespace */
UA_UInt16 UA_EXPORT UA_Server_addNamespace(UA_Server *server, const char* name);
```

ichrispa, 2015-11-22:
The numerical identifier of the namespace is just incremented with each "add"? I would expect there to be a UA_Int32 requestedNamespaceIndex in the call.

Suppose I have a field device that comes pre-modelled in namespace "23"... the user would have to add 22 "non-related" namespaces to get to 23?

Stasik0, 2015-11-22:
The number of the namespace does not play any role. Only the position of namespace's urn in the namespace array counts. Inho there is no need to add a namespace with a particular number.

ichrispa, 2015-11-23:
Well, 2 against 1 ^^ I consider this issue closed.

closed 2015-11-23 by ichrispa