open62541/open62541 #469: Android build and unaligned access.

Aleskey78, 2015-11-25:
Hello,

I have spent some hours today trying to figure out why a simple client program crashes in a SecureChannelHandshake->…->UA_encodeBinary->UA_UInt64_encodeBinary function trying to encode UA_TYPES_DATETIME.

Maybe it is not really a bug, but a predefined macro __ARM_FEATURE_UNALIGNED works somehow weird on GCC 4.9, Qt 5.5.0. 

QMAKE_CFLAGS += -mno-unaligned-access solved the problem.

Tested on ARMv7 (Samsung SGS5 and Asus Nexus7 2013)

Best regards,
Aleskey

jpfr, 2015-11-27:
Hey @Aleskey78,

what do you mean by "works somehow weird"?

If you can change the behavior with a compiler flag, maybe you need to set __ARM_FEATURE_UNALIGNED manually..

The relevant code portions are in ua_config.h.in
```c
#if defined(__arm__) && !defined(__ARM_FEATURE_UNALIGNED)
# define UA_ALIGNED_MEMORY_ACCESS
#endif
```
and in ua_types_encoding_binary.c

```c
static UA_StatusCode
UInt64_encodeBinary(UA_UInt64 const *src, const UA_DataType *_,
                    UA_ByteString *dst, size_t *UA_RESTRICT offset) {
    if(*offset + sizeof(UA_UInt64) > dst->length)
        return UA_STATUSCODE_BADENCODINGERROR;
#if defined(UA_NON_LITTLEENDIAN_ARCHITECTURE) && !defined(UA_ALIGNED_MEMORY_ACCESS)
    UA_UInt64 le_uint64 = htole64(*src);
    src = &le_uint64;
#endif

#ifdef UA_ALIGNED_MEMORY_ACCESS
    dst->data[(*offset)++] = (*src & 0x00000000000000FF) >> 0;
    dst->data[(*offset)++] = (*src & 0x000000000000FF00) >> 8;
    dst->data[(*offset)++] = (*src & 0x0000000000FF0000) >> 16;
    dst->data[(*offset)++] = (*src & 0x00000000FF000000) >> 24;
    dst->data[(*offset)++] = (*src & 0x000000FF00000000) >> 32;
    dst->data[(*offset)++] = (*src & 0x0000FF0000000000) >> 40;
    dst->data[(*offset)++] = (*src & 0x00FF000000000000) >> 48;
    dst->data[(*offset)++] = (*src & 0xFF00000000000000) >> 56;
#else
    *(UA_UInt64*) &dst->data[*offset] = *src;
    *offset += 8;
#endif
    return UA_STATUSCODE_GOOD;
}
```

There is not much that can go wrong there, imho.

Aleskey78, 2015-11-30:
Hi @jpfr,

it looks like this problem has nothing to do with an open6254 project itself, sorry.

The __ARM_FEATURE_UNALIGNED flag is set automatically from GCC and it is always defined/set, whether the -mno-unaligned-access flag was set or not. Actually it should be active by default for ARMv7-A as it stated here: https://gcc.gnu.org/gcc-4.7/changes.html

This part of code caused unalignment fault:
```c
#else
    *(UA_UInt64*) &dst->data[*offset] = *src;
    *offset += 8;
#endif
```

By forcing compiler to generate an unaligned access code with a -mno-unaligned-access flag the problem was solved.

Regards / Gruß
Aleskey

closed 2015-11-30 by Aleskey78

Stasik0, 2015-11-30:
@Aleskey78, @jpfr meant that alternatively you can unset -mno-unaligned-access in the compiler and set ```# define UA_ALIGNED_MEMORY_ACCESS``` in our code - in this case no unaligned access will be made as well. It would interest me which approach will be a faster one.

Aleskey78, 2015-11-30:
@Stasik0, unaligned memory access should be much slower, please take a look here: https://github.com/andikleen/snappy-c/pull/4

Stasik0, 2015-11-30:
@Aleskey78 i meant which is slower, the compiler workaround via switch -mno-unaligned-access or the direct modification in our code by disabling UA_ALIGNED_MEMORY_ACCESS

Aleskey78, 2015-12-01:
@Stasik0, could you provide some benchmarking code or server_readspeed.c example will be enough?

Aleskey78, 2015-12-08:
@Stasik0; sorry, with or w/o AMALGAMATION cannot compile server_readspeed.c:
Fehler: undefined reference to `UA_encodeBinary(void const*, UA_DataType const*, UA_String*, unsigned int*)'
Fehler: undefined reference to `UA_decodeBinary(UA_String const*, unsigned int*, void*, UA_DataType const*)'
Fehler: undefined reference to `Service_Read(UA_Server*, UA_Session*, UA_ReadRequest const*, UA_ReadResponse*)'