open62541/open62541 #473: someone please tell me how to implement a message in opc ua server through tcp port ,with some codes please and is it possible through netbeans IDE

manikandankasi, 2015-12-04:


FlorianPalm, 2015-12-05:
Could you please tell us a little more specific what you are planning to do? I am not sure what you mean with "implement a message"?

manikandankasi, 2015-12-06:
how to create a server and client for OPC UA  with the help of java in Netbeans and this was the assigned work for me and if you have a message in client side and we should receive through TCP port in client side and am really sorry if the inforamtion is not necessary and am new to this field and am learning day by day,please tell how opc ua works and how to learn about this stuff

Stasik0, 2015-12-06:
Open62541 is a C-based stack. So you either have to look into a jna binding for java e.g. with jnaerator, or to look for a java stack alternative. There is a link in our wiki.

manikandankasi, 2015-12-06:
thanks for the answers,but still  not able to get a proper solution for running a java program for opc ua server-client in netbeans??????

HolgerJeromin, 2015-12-06:
Stasik said you probably should use one of those stacks listed as Java:
https://github.com/acplt/open62541/wiki/List-of-Open-Source-OPC-UA-Implementations

manikandankasi, 2015-12-08:
i have already got some multichat application with server and multi clients and transferring files too,but i want to make it secure through aes encryption and rsa for opc ua,please suggest me some solution

jpfr, 2015-12-09:
@manikandankasi 
1. If you want to write chat applications, use IRC or Jabber.
2. We don't support encryption yet.
3. Our stack is in C. I think you should get a trial of the Prosys Java SDK: https://www.prosysopc.com/products/opc-ua-java-sdk/

closed 2015-12-09 by jpfr

manikandankasi, 2015-12-09:
already i have Proxys Java SDK but not able to understand how it works and how to run a java application in this SDK and i need some help for this

FlorianPalm, 2015-12-09:
Unfortunately, we are not experts of Prosys. But you can find their support forum here: 
http://forum.prosysopc.com/

manikandankasi, 2015-12-10:
okay ,thank you for that,can you just tell me how a java apllication can run in a server with aes encryption

jpfr, 2015-12-10:
No.

I hope this does not come across as rude. But we are unable to provide support on topics that are not directly related to the open62541 project. Our lives are busy enough.

manikandankasi, 2015-12-10:
can you send me a sample java application running in opc ua server with creating certificates,signature and all the security methods,to make it secret and secured,if you have something like that,please help me by sending the codes,thanking you for your kind reply