open62541/open62541 #474: package-architecture (amd64) doesn't match to the system (armhf)

Spezies31, 2015-12-07:
Hello,

i am trying to use the UAExpert Client from Unified Automation to view my Client in a more pleasent way.
I downloaded the UAExpert Client for Debian (v1.4.1) and tried to install it on my Respbian System. After downloading it i unpacked it with:

tar -xvzf <Dateiname>.tar.gz

then i wanted to install it with:

sudo dpkg -i <Dateiname>.deb

But then i got the following error message:

dpkg: Fehler beim Bearbeiten von uaexpert-bin-debian7-x64-gcc4.7.2-v1.4.1-240.deb (--install):
 Paket-Architektur (amd64) passt nicht zum System (armhf)
Fehler traten auf beim Bearbeiten von:
 uaexpert-bin-debian7-x64-gcc4.7.2-v1.4.1-240.deb

The English translation is:

package-architecture (amd64) doesn't match to the system (armhf)

I googled and found nothing useful, so i hope that you could tell me what's the problem. Isn't it possible to install the UAExpert Client for Debian on a Raspbian System?

Greetings Spezies31

Stasik0, 2015-12-07:
This problem is not really open62541 related, afaik UnifiedAutomation does not distribute any arm binaries, therefore you can only run UAExpert on x86 systems. 

Btw, are you sure you want to install the client on the raspberry pi and not on your PC? Normally the embedded system acts as a server and the PC as the client.

Spezies31, 2015-12-08:
I just try to learn how to build up a communication via OPC UA. Therefore i tried the open62541 stack and now i am trying (after recommondation) Unified Architecture with the programs on their website.
I talked to a few people and i think that there is no way to use the unified automation programs on a raspberry pi 2 model B properly because of the hardware architecture.

closed 2015-12-08 by Stasik0