open62541/open62541 #475: Mailing List issue "UA_SubscriptionSettings and Subscription for "long-named" identifiers"

jpfr, 2015-12-08:
Opening up an issue for the mailing list thread "UA_SubscriptionSettings and Subscription for "long-named" identifiers"

https://groups.google.com/forum/#!topic/open62541/i3zc9UexPYo

Oleksiy posted a gist with an example that shows the bug.
We have no excuse not to get rid of the bug fast.

@ichrispa Can you help in pinpointing the source of this behavior? Maybe the fix is trivial after that.

ichrispa, 2015-12-08:
Will do. I'll try to reproduce this tomorrow and commit a fix if it is easily solvable

ichrispa, 2015-12-09:
@Stasik0, @jpfr:  Apart from the Gist clientside, can Alekesy please also provide a serverside example that delivers a reproducable failure? I have CodeSys with a patch level 3 floating around, but I am not going to learn how to use the OPC UA Server of patchlevel 7 just to provide support. 

We have been having several problems with WAGO OPC UA servers (particular with subscriptions), includuing an ARM based WAGO PLC running a certain OPC UA capable Beta firmware. As far as I can tell it's the PLC side server mucking things up, not us.

ichrispa, 2015-12-09:
I just tested this by placing a string named node into our own server.c (replacing currenttime). The subs work as expected.

Stasik0, 2015-12-09:
@ichrispa i think it is some server-client issue, we will need to analyse the wireshark traffic. @Aleskey78 do you have any means to hang the opc server into a DMZ so we can access it over ip? Does anyone have a better idea?

Stasik0, 2015-12-09:
@ichrispa or should we take somethink like this? http://store.codesys.com/systeme/codesys-control-for-raspberry-pi-sl.html

ichrispa, 2015-12-09:
I don't really know if that's the way to go. Before I go anywhere into a WAGO direction, I would first like to established what's happening. If this is our client API misbehaving, I can get my hands on the new Codesys or respective PLCs around here too with relative ease.

I would like to see a wireshark dump of this problem, from transport HELLO to CLOSE. I am especially interested if any publishResponses reach the client at all (and if they do, with what StatusCode). If there are no relevant publishResponses for ChangedItems, I consider this a serverside bug and I can only offer to communicate this bug to WAGO.

We have seen this bug (or a very similar one) around here before during a thesis. That's why I don't want anyone to dive into the subject before we can establish that we can fix it.

Stasik0, 2015-12-09:
Alexey, could you get a wireshark dump?

Aleskey78, 2015-12-10:
Hello,
here is the first test sequence (with 300ms delay) and a dump:

server ip: opc.tcp://10.1.8.60:4840
client ip: 10.1.3.204
Server variables:
   |var|CODESYS Control RTE .Application.HMI.stPagesVend.stPageFavorites.stFavorite_1.sName
   |var|CODESYS Control RTE .Application.HMI.stPagesVend.stPageFavorites.stFavorite_2.sName
   |var|CODESYS Control RTE .Application.HMI.stPagesVend.stPageFavorites.stFavorite_3.sName
   |var|CODESYS Control RTE .Application.HMI.stPagesVend.stPageFavorites.stFavorite_4.sName
   |var|CODESYS Control RTE .Application.HMI.stPagesVend.stPageFavorites.stFavorite_5.sName
   |var|CODESYS Control RTE .Application.HMI.stPagesVend.stPageFavorites.stFavorite_6.sName

Server PLS program does not change any of them.

UaExpert runs on a client PC parallel to Qt test client (gist: https://gist.github.com/Aleskey78/b2ed116b3975bb9d1f33)

Variables were manually set in the UaExpert:
stFavorite_1.sName set to “1” -> client receives  notification: yes
stFavorite_2.sName set to “2” -> client receives  notification: no
stFavorite_3.sName set to “3” -> client receives  notification: no
stFavorite_4.sName set to “4” -> client receives  notification: no
stFavorite_5.sName set to “5” -> client receives  notification: no
stFavorite_6.sName set to “6” -> client receives  notification: no

Link to the Wireshark dump: https://gist.github.com/Aleskey78/b2ed116b3975bb9d1f33/raw/a8cbc18899cbecedde0650055156f0d0ad638a72/opcua_test1_marked.pcapng

Thanks
Oleksiy

Aleskey78, 2015-12-10:
here is the second test without delay, wireshark dump: https://gist.github.com/Aleskey78/b2ed116b3975bb9d1f33/raw/cfe6a98b9aa9a1841f888a59152a2b2aac2f54b5/opcua_test2_marked.pcapng

The client works as expected:
Variables were manually set in the UaExpert:
stFavorite_1.sName set to “11” -> client receives notification: yes
stFavorite_2.sName set to “22” -> client receives notification: yes
stFavorite_3.sName set to “33” -> client receives notification: yes
stFavorite_4.sName set to “44” -> client receives notification: yes
stFavorite_5.sName set to “55” -> client receives notification: yes
stFavorite_6.sName set to “66” -> client receives notification: yes

Aleskey78, 2015-12-10:
@Stasik0, one more example and a dump:

Codesys OPC UA server and a test client are running on the same PC.

Client code, PLC program, a wireshark dump and a client log: https://gist.github.com/Aleskey78/613865f6e9a72e804eab

Stasik0, 2015-12-10:
okay, let me learn subscriptions. 

Regarding the last dump: https://gist.github.com/Aleskey78/613865f6e9a72e804eab
What i observe:
* Packet 39: client issues a publish request for subscription number 4
* Packet 40: server answers, but with a notification for subscription number 3
-> the subscription in our client is 'locked' since it wait for subscription id forever :)

can this line be a problem, @ichrispa? https://github.com/open62541/open62541/blob/b3459d09d78ca68097c1e4c033d132924caad626/src/client/ua_client_highlevel_subscriptions.c#L263

ichrispa, 2015-12-10:
@Stasik0: Publish Requests are not targeted at a subscription. 

@Stasik0: The line you marked is absolutely valid. I only add Publish ACKs to the list when I verify that they are not in it. A publish ACK is identified by both the seqNo and the subID (subscription 1,2,3 and 4 may all have the seqNo 8, so one of the two is not enough).

The messages seen in the in the package dump are keepalive messages emitted by the server. The spec says that if a subscription keepalive interval is reached without new data being available, the subscription shall emit a keepalive response. The response consists of the subscriptions ID, the *next* seq. number to be used and a null sized/empty array of notificationItems.

As far as I interpret the package dump, the server does not generate any changedItemNotifications.

What is interesting from my point of view is the second publish response from the server - that response appears to be a keepalive, but the server incremented the sequencenumber for some reason.

From what I can see this is a serverside fault. 

@Aleskey78: Maybe you could play around with the publishInterval of the subscription and the SamplingInterval of the monitoredItem req? Try increasing both to some higher value, like 500ms and 2 subs a 250ms samplingInterval? Maybe the server can be coerced into working by slowing down the polling.

Aleskey78, 2015-12-10:
Hi @Stasik0,

thanks for your suggestion to create only one subscription, this helped. 

This test case seems has nothing to do with a real-life situation, it is just a banal cut’n’paste mistake, piece of code blindly copied several times from a client example.
Anyway a server could handle it in the right way...

Thanks again for your help!
Cheers

Stasik0, 2015-12-11:
@Aleskey78 where does this info come from? Why do they not limit the number of subscriptions by a proper error-message?

ichrispa, 2015-12-11:
@Aleskey78: You should still be able to create multiple monitoredItems and assign them all to the same subscription. Sorry we can't fix this from the open62541 side.

Maybe, as a workaround, you could spawn multiple client threads and aggregate the subscription replies for processing (with a thread safe/mutex controlled queue)? The publish handlers would only append the replies to the queue and you could handle them externally?

Aleskey78, 2015-12-11:
@Stasik0: these are only my observations, based on the tests with many "exe" - clients simultaneously fetching the same data from the same server. I didn’t get in contact with 3S yet.

@ichrispa: for our project the workaround with only one subscription should be enough. 
Increasing requestedMaxKeepAliveCount or maxNotificationsPerPublish makes a “UA_Client_Subscriptions_manuallySendPublishRequest(client)” call blocking till a new notification comes. It makes not possible to use read/write operations from another thread until ...manuallySendPublishRequest is finished. Should i consider using two UA_Clients: one for RW and one more only for subs?

OFF-TOPIC:  is there any way to get notified about connection drops , watchdog timeouts?

thx

Stasik0, 2015-12-11:
@Aleskey78 the service layer returns UA_STATUSCODE_BADCONNECTIONCLOSED if the connection was dropped. I am not sure whether it is propagated up by the subscription interface, client api of the read service does it.

Stasik0, 2016-02-09:
closed due to expiration, feel free to reopen

closed 2016-02-09 by Stasik0