open62541/open62541 #476: FOREACH_SAFE does not compile.

NetoLe, 2015-12-08:
Hello there,

I issued a problem with compiling the parts, where FOREACH_SAFE is used instead of FOREACH.
A sample compiler error is this one: 

`src/open62541.c:14326:5: warning: implicit declaration of function 'LIST_FOREACH_SAFE'`
`src/open62541.c:14326:60: error: 'listEntry' undeclared (first use in this function)`
`src/open62541.c:14326:60: note: each undeclared identifier is reported only once for each function it appears in`
`src/open62541.c:14326:76: error: expected ';' before '{' token`

so the function was:

`UA_unpublishedNotification *not, *tmp;
    LIST_FOREACH_SAFE(not, &sub->unpublishedNotifications, listEntry, tmp)`

if I now change it to:

`UA_unpublishedNotification *not;
    LIST_FOREACH(not, &sub->unpublishedNotifications, listEntry)`

it's working fine.

I didn't change the defines:
` #define LIST_FOREACH(var, head, field)					\ `
`       for((var) = LIST_FIRST(head);					\`
`	    (var)!= LIST_END(head);					\`
`	    (var) = LIST_NEXT(var, field))`

`#define	LIST_FOREACH_SAFE(var, head, field, tvar)			\`
`	for ((var) = LIST_FIRST(head);				\`
`	    (var) && ((tvar) = LIST_NEXT(var, field), 1);		\`
`	    (var) = (tvar))``

This error is happening with every FOREACH_SAFE. 

The compiler I use is gcc on windows (cross-compiling) for an ARM-embedded device.

jpfr, 2015-12-08:
If `LIST_FOREACH_SAFE` is defined as in you post **before** its use in line 14326, then the code is correct.

That looks like a compiler bug in the cross-compile toolchain... Not that we test with three compilers (gcc, clang, msvc) for continuous integration. And none of them gives this warning.

You could follow the build instructions [1] to build the library from individual source files.
Maybe the compiler gets problems with the single-file release being too big for a single file.

[1] http://open62541.org/doc/sphinx/building.html

NetoLe, 2015-12-08:
the macros are defined in lines 330 and 335 of the file. So I tried your solution and split all the files without amalgation. the error remains the same. I used the Cflags defined in your link as well. 
Can it cause big issues if I just use the functions without SAFE?

Stasik0, 2015-12-09:
yes, it can cause segfaults, since _SAFE allows deleting items during the iteration.

jpfr, 2015-12-12:
@NetoLe Can you point us to the exact toolchain you are using?
If its a Linaro one, there should be a free download so we can test.

NetoLe, 2016-01-04:
my bad I didn't see there was a reply.

So the company I am working for uses the arm-elf-v2 for compiling for the project where the open62541-Server is located.

Stasik0, 2016-01-04:
is there a download? which gcc version is it?

NetoLe, 2016-01-05:
So I am using gcc 4.5.1. I'm not completely sure but this should be the toolchain: http://sourceforge.net/projects/netx-arm-elf-gc/files/GCC-4.5.x/4.5.1/

Stasik0, 2016-01-06:
@NetoLe just being curious, does the newest queue.h remedy your problem? https://raw.githubusercontent.com/freebsd/freebsd/master/sys/sys/queue.h ?

NetoLe, 2016-01-07:
I tried this, but it does not :( For now it's a workaround to just write the macro directly into the code at the given places and at all it really seems to be a compiler bug

jpfr, 2016-01-11:
Making changes to queue.h is nothing I am very keen on.
Since it seems to be a compiler bug that turns up for a single release of a cross-compiler, I suggest this is WONTFIX. Plus, @NetoLe has found a workaround for his project.

@NetoLe is this okay with you?

NetoLe, 2016-01-11:
of course it is. :)

closed 2016-01-11 by jpfr