open62541/open62541 #477: UA_String problem

tpiat, 2015-12-08:
In version 23980d297d Client works.
After updating to 2f4a056607 UA_String do not works, string has length equials 0xababab00.
What changes has been made in this part?

Stasik0, 2015-12-08:
The string length has changed from signed to unsigned, could you please show your code - you will probably will need to update it

tpiat, 2015-12-08:
What I must change?

Stasik0, 2015-12-08:
well, the main change is this one: https://github.com/acplt/open62541/blob/06b6687b3f25dc01ee31b0d1fa1c4e87dedaccd2/include/ua_types.h#L95

from UA_Int32 -> size_t

tpiat, 2015-12-08:
I don't use type of length.
And array of string works but scalar -- do not works.

Stasik0, 2015-12-08:
well, we need to see some code, otherway try to compare your code to the current examples that we have https://github.com/acplt/open62541/tree/master/examples

tpiat, 2015-12-08:
Thanks!
Error in length of data.

int arrayLenl = rResp.results[i].value.arrayLength;
array lenght was -1 but now it equals 0 in scalar value.

closed 2015-12-08 by tpiat