open62541/open62541 #480: processRepeatedJobs, again non-RTC-problems, but in the future

NetoLe, 2015-12-09:
Hello again, and sorry again.

So it works for now, if the server starts in the past. But if it starts in the future, like 2037, it gets problems, when it gets synchronized to RT.

the if-part:
`if(tw->nextTime > current)`
breaks the loop that could set the time back.
the return then is the `MAXTIMEOUT`, because the `timeout` has values like over 3 billion, without setting the `tw->nextTime` to a valid value with which could be worked in the next iteration.

To explain the problem a bit clearer:
I see, that the timeout is needed for `ServerNetworkLayerTCP_getJobs` and the inherent select function. But I am adding my own repeated Job (Type: UA_JOBTYPE_METHODCALL) that I use for a workaround, which updates my Variable-Values every 250 ms. So it is part of the repeated jobs but not related to the network functions. 
I don't know how to get it working correctly without breaking other parts in the code. Maybe an iteration through the list setting back the nextTimes would work.

jpfr, 2015-12-10:
We can fix that.
Most systems let us use a monotonically increasing clock to measure elapsed time.
See for example here: http://linux.die.net/man/3/clock_gettime

That means, we would use a clock to measure time between repeated jobs and not the system time.
The elapsed time can jump forward, e.g. if the system was in energy-saving mode, but never backwards.

Stasik0, 2015-12-10:
porting clock_gettime to windows with performance counters http://stackoverflow.com/questions/5404277/porting-clock-gettime-to-windows

NetoLe, 2015-12-10:
ok, thanks. The monotonic clock sound's like a good idea. 
I noticed my problem was also caused by a NTP additionally. it starts at 1900 and not at 1970. I had an offset to 1970 and subtracted it always. but the counter started at 0 as an unsigned long. Subtracting then always got me values larger then the RT, which was the root of the problem.
For now I will just subtract if the current time value is greater then the offset to 1970, which fixes the problem temporary. 
There's another simple problem with the moment, when it get's synchronized: It loses connection for about 15-20 seconds. UaExpert turns to 'ConnectionErrorApiReconnect' and creates a new session if it gets a new connection. A client I programmed with the Client-Api from the stack can't reconnect automatically. I need to restart it. This happens just, when it starts in the past and gets synched to RT.
When it starts in the future and gets synched to RT it doesn't lose connection but is not capable of working any of its jobs, besides that the ServerTime is running correct then. 
I'm not sure if creating a new Issue would be necessary, because I think this could be fixed with the monotonic clock part. 
And for my needs it's not a big issue. I appreciate your work on this whole stack.

commit b8e278d3 (2015-12-23): Merge pull request #490 from open62541/monotonic_clock

types: add a monotonic cpu timer to measure delays, fix #480
closed 2015-12-23 by jpfr