open62541/open62541 #481: Client loses connection after ~3 min

NetoLe, 2015-12-11:
This is a weird problem. 
I programmed a client with the stack. I can connect with my server I programmed with the Stack too, without problems. But after about 2:45~3 min it tries to renew the securechannel and then just stops working. I can't access any data from the server with the client. Besides UaExpert never has Problems and can be connected over hours. First I was polling a lot of data every 500 ms and thought that's the problem but then I changed to just one node every 10 seconds to check. I attached a cmd-line log + a wireshark-snippet, so that you understand what I mean. 

My dev-environment is Visual Studio 2013 for that. 

[image]
[image]

Stasik0, 2015-12-11:
could you export the wireshark traffic? you can filter the traffic with "opcua" and export only filtered stuff so we do not see other traffic. What is the server you are using? What is the client? Can you attach source code (at least the client-side one)?

NetoLe, 2015-12-11:
[open62541_client.txt](https://github.com/open62541/open62541/files/59387/open62541_client.txt)

I needed to XXXX the Client-Adapter, so don't wonder. 

The server is your open62541, latest current master branch. amalgamated.
It is placed on an embedded device running on a single core ARM.

The client is from your stack too, a normal PC that is connected with the embedded device via ethernet. On this machine is UaExpert running too. (it doesn't influence each other, the client-application was running while UaExpert wasn't activated)

Here is a source-snippet that inhabits the relevant code I am working with on the client-side. 
All other stuff is like the same, just for different values I am getting from the device. 

[github_source.txt](https://github.com/open62541/open62541/files/59408/github_source.txt)

I took out the output of the result-string to get it more compact. You see I took the polling in an extra thread. 

I can't attach server-source-code, sorry :/ But I think, when UaExpert is running without problems it can't be on the server side. 
All I can say, I deactivated Subscriptions and method calls + deactivated the write-Service, because the Server should be just read-only. 
Generate Namespace0 + Nodemanagement is also deactivated.

Stasik0, 2015-12-11:
it seems that the server does not send the response
* can you reproduce it with the default server (server.c from the repository)
* are you able to see where your client is blocking by using a debugger? is it UA_Client_Service_read?

btw I think your client is leaking memory since you malloc returnValue twice :)

NetoLe, 2015-12-14:
yes, it was the Client_Service_read, where it started to renew the channel and stopped working then. but it's not that it isn't invoking the service, it just returns: BADSERVERNOTCONNECTED

the pcap (is it ok, if I delete it, after you read it?):
[openClient_conn_problem1.zip](https://github.com/open62541/open62541/files/60922/openClient_conn_problem1.zip)

I will test with the server.c later on. But I think I can post you the simple main loop I am using. 

`server = UA_Server_new(UA_ServerConfig_standard);
      if(server)
      {
        UA_Server_addNetworkLayer(server, ServerNetworkLayerTCP_new(UA_ConnectionConfig_standard, 4840));
      
        XXXXX::initNodes();
        /* add a cyclic update to the server*/
        UA_Job job;
        job.type = UA_Job::UA_JOBTYPE_METHODCALL;
        job.job.methodCall.method = updateMethod;
        job.job.methodCall.data = NULL;
        UA_Server_addRepeatedJob(server, job, 200, NULL); // call every 200 msec
        while (1)
        {
          if (running)
            UA_Server_run(server, 1, &running); // no loop, the server itself has a built-in loop
          XXXXX::sleep(2000);
          UA_Server_addRepeatedJob(server, job, 200, NULL);
          running = UA_TRUE;
        }
       
      }`

short explanation: there are more different tcpip-components on the device, and when they are loaded it can be that the server get's kicked out if it was loaded before the other tcpip-components, so the while(1)-loop is just to get it back in. That is nothing from interest, when the device has finished its startup and is running. 
to the memory leaks: yeah it could be that I snipped parts while shortening it for demonstration. Otherwise: at leak I tried. ;)

Stasik0, 2015-12-14:
just a quick glance:
* calling UA_Server_addRepeatedJob in a loop will spam the internal scheduler, and is probably not desired
* there is a ```UA_Server_run_mainloop```function to execute exactly one loop of the server

NetoLe, 2015-12-14:
ok, my bad, I wasn't aware of this point, I thought that breaking the Server (running = FALSE), will delete all added jobs. but now I see that the jobs are stored in the server itself which won't get deleted.
Means: the loop did break 2 times, means 3*repeatedJobs. Thanks for that.

The problem still remains.

Stasik0, 2015-12-14:
okay, i confirm the problem with the example code and the server

Stasik0, 2015-12-16:
@NetoLe can we consider this fixed now, did you try 6f9514e?

NetoLe, 2015-12-17:
yep I tried both fixes. And they work :) thanks!

closed 2015-12-17 by jpfr