open62541/open62541 #482: Namespace compiler no longer initializes variables

ichrispa, 2015-12-13:
Just as an fyi, the namespace compiler does no longer initialize any values/variables from the source XML. After @jpfr changed the highlevel API to use attribute structs.

As a quick diagnostic, each nodeType had generate_early, generate and generate_late routines. With attribute structs, all functionality of the code generator was moved into the macroHelper and node specific generate routines are no longer called.

This primarily impacts (a) my current paper and (b) our ability to genrate method call parameters out of XML definitions.

I'll be on this one.

Stasik0, 2015-12-13:
We need unit tests for ns compiler...

ichrispa, 2015-12-14:
How about executing a ns0 method and reading a ns0 variable, e.g. from the ServerType primitive?

jpfr, 2015-12-14:
Sorry for the breakage. Had I been aware of it, I had not merged the PR.
If I can be of help, please tell me.

ichrispa, 2015-12-14:
Hi @jpfr,

I didn't want to imply that anyone broke this on purpose. This sort of stuff just happens ^^

But I could use some help:
Single values are apparently encoded properly. But ExtensionObjects (and in turn all user structs) appear to be buggy. I stepped through ns=0,i=114910 and 114911 (Arguments of GetMonitoredItems). The extensionsObjects are created and encoded as they should (encoding offsets match what is expected). 

During transmission, additional data is appended to the extension Object, clearly passing the bounds of the memory area. I have no idea what causes this, as the extensionObject/Variant is all nice and proper when it is passed to the variable. This causes both wireshark an UAExpert to be unable to decode the message.

Having checked readSingleAttribute and the namespace_init(), I'm running out of ideas on how to debug this :/

As a minor issue: Is addMethodNode creating Arguments on its own? There are two variableNodes from ns=1 in generated methods that my compiler is not putting there.

jpfr, 2015-12-15:
Hey @ichrispa,
Can you push your current state to a branch for debugging?
And yes, addMethodNode is creating the input and output argument nodes internally. That's why you have to provide the argument arrays when the node is added.

ichrispa, 2015-12-15:
The current state has been pushed with fd378f1.

addMethod needs to create inputArguments and outputArguments with a qualifiedName in nsIndex=0, otherwise the variables are just variables and are not considered to be arguments of the method.

I still want to opt for a way of not specifying arguments at all (or alternatively reverting all methodNodes to bootstrapping mode). Arguments are just variableNodes and the namespace compiler does not perform any semantic analysis prior to constructing them. It is significantly easier to treat variableNodes all in the same manner and attach them to the methodNode with the given references.

jpfr, 2015-12-16:
Hey @ichrispa 

can you help me setting up an example where the bug occurs?
I built two small, examples, but could not generate variables with extensionvalue values.

### First example

```xml
    <UAVariable DataType="Double" ParentNodeId="ns=1;i=1001" NodeId="ns=1;i=6001" BrowseName="1:Var1" UserAccessLevel="3" AccessLevel="3">
        <DisplayName>Var1</DisplayName>
        <Value>
          <Int32>1</Int32>
        </Value>
        <References>
            <Reference ReferenceType="HasTypeDefinition">i=63</Reference>
            <Reference ReferenceType="HasModellingRule">i=78</Reference>
            <Reference ReferenceType="HasComponent" IsForward="false">ns=1;i=1001</Reference>
        </References>
    </UAVariable>
```
produces
```c
UA_VariableAttributes attr;
UA_VariableAttributes_init(&attr);
attr.displayName = UA_LOCALIZEDTEXT("", "Var1");
attr.description = UA_LOCALIZEDTEXT("", "");
UA_Double opcua_node_variable_t_ns_1_i_6001_variant_DataContents = (UA_Double) 1.0;
UA_Variant_setScalarCopy( &attr.value, &opcua_node_variable_t_ns_1_i_6001_variant_DataContents, &UA_TYPES[UA_TYPES_DOUBLE]);
UA_Double_deleteMembers(&opcua_node_variable_t_ns_1_i_6001_variant_DataContents);
UA_NodeId nodeId = UA_NODEID_NUMERIC(1, 6001);
UA_NodeId typeDefinition = UA_NODEID_NULL;
UA_NodeId parentNodeId = UA_NODEID_NUMERIC(1, 1001);
UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, 47);
UA_QualifiedName nodeName = UA_QUALIFIEDNAME(1, "Var1");
UA_Server_addVariableNode(server, nodeId, parentNodeId, parentReferenceNodeId, nodeName
       , typeDefinition
       , attr, NULL);
// This node has the following references that can be created:
UA_Server_addReference(server, UA_NODEID_NUMERIC(1, 6001), UA_NODEID_NUMERIC(0, 40), UA_EXPANDEDNODEID_NUMERIC(0, 63), UA_TRUE);
UA_Server_addReference(server, UA_NODEID_NUMERIC(1, 6001), UA_NODEID_NUMERIC(0, 37), UA_EXPANDEDNODEID_NUMERIC(0, 78), UA_TRUE);
```

That example works perfectly fine, only int32 is converted to a double node.

### Second example
The extensionobject is copied from the huge nodeset2.xml
```xml
    <UAVariable DataType="Double" ParentNodeId="ns=1;i=5001" NodeId="ns=1;i=6002" BrowseName="1:Var1" UserAccessLevel="3" AccessLevel="3">
        <DisplayName>Var1</DisplayName>
        <Value>
          <ExtensionObject>
            <TypeId> <Identifier>i=297</Identifier> </TypeId>
            <Body>
              <Argument>
                <Name>SubscriptionId</Name>
                <DataType> <Identifier>i=7</Identifier> </DataType>
                <ValueRank>-1</ValueRank>
                <ArrayDimensions />
                <Description p5:nil="true" xmlns:p5="http://www.w3.org/2001/XMLSchema-instance" />
              </Argument>
            </Body>
          </ExtensionObject>
        </Value>
        <References>
            <Reference ReferenceType="HasTypeDefinition">i=63</Reference>
            <Reference ReferenceType="HasComponent" IsForward="false">ns=1;i=5001</Reference>
        </References>
    </UAVariable>
```

Code generation succeeds, but the variable has no value.
```c
UA_VariableAttributes attr;
UA_VariableAttributes_init(&attr);
attr.displayName = UA_LOCALIZEDTEXT("", "Var1");
attr.description = UA_LOCALIZEDTEXT("", "");
UA_NodeId nodeId = UA_NODEID_NUMERIC(1, 6002);
UA_NodeId typeDefinition = UA_NODEID_NULL;
UA_NodeId parentNodeId = UA_NODEID_NUMERIC(1, 5001);
UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, 47);
UA_QualifiedName nodeName = UA_QUALIFIEDNAME(1, "Var1");
UA_Server_addVariableNode(server, nodeId, parentNodeId, parentReferenceNodeId, nodeName
       , typeDefinition
       , attr, NULL);
// This node has the following references that can be created:
UA_Server_addReference(server, UA_NODEID_NUMERIC(1, 6002), UA_NODEID_NUMERIC(0, 40), UA_EXPANDEDNODEID_NUMERIC(0, 63), UA_TRUE);
```

ichrispa, 2015-12-17:
Hi @jpfr,

in order to generate an extensionObject, two conditions must be met:
  1. A variable or variableType must have an '''<Value></Value>''' field.
  1. The type of the variable (and its encoding) must be deducable by the namespace compiler, in particular
    * the encoding must be tracable through "hasDataType", "hasTypeDefinition" or "usSubTypeOf" references
    * All traced types of the dataType must be encodable (nothing abstract like "numerical")

Your first example is fine, but the compiler detects the float as it should. ExtensionObjects are only used for structured/hierarchic types.

In case of Arguments, the complete dataType description must be present in the namespace, otherwise the compiler does not know how to perform an encoding. If you run the compiler with LOG_LEVEL_WARN or higher, it will propably notify you that the dataType is not encodable because some part of it is unknown.

I would advise using the Opc.Ua.NodeSet2.xml as a reference. ObectTypes->ServerType holds a method called GetMonitoredItems that I have been using as a reference point (Arguments are i=11490 and i=11491). It should generate an array of extensionObjects. Here's my generated code:

```C
\\... snipped out of ua_namespace_generated.c
UA_VariableAttributes attr;
UA_VariableAttributes_init(&attr);
attr.displayName = UA_LOCALIZEDTEXT("", "InputArguments");
attr.description = UA_LOCALIZEDTEXT("", "");

\\ value is encoded by printOpen62541CCode_early()
\\ derived structure of the extensionObject "Argument"
struct {
  UA_String Name;
  UA_NodeId DataType;
  UA_Int32 ValueRank;
  UA_Int32 ArrayDimensionsSize;
  UA_UInt32 *ArrayDimensions;
  UA_LocalizedText Description;
} opcua_node_variable_t_i_11490_Argument_0_0_struct;

\\ Populating values
opcua_node_variable_t_i_11490_Argument_0_0_struct.Name = UA_STRING_ALLOC("SubscriptionId");
opcua_node_variable_t_i_11490_Argument_0_0_struct.DataType = UA_NODEID_NUMERIC(0, 7);
opcua_node_variable_t_i_11490_Argument_0_0_struct.ValueRank = (UA_Int32) -1;
opcua_node_variable_t_i_11490_Argument_0_0_struct.ArrayDimensionsSize = 1;
opcua_node_variable_t_i_11490_Argument_0_0_struct.ArrayDimensions = (UA_UInt32 *) UA_malloc(sizeof(UA_UInt32));
opcua_node_variable_t_i_11490_Argument_0_0_struct.ArrayDimensions[0]  = (UA_UInt32) 0;
opcua_node_variable_t_i_11490_Argument_0_0_struct.Description = UA_LOCALIZEDTEXT("en_US", "");
UA_ExtensionObject *opcua_node_variable_t_i_11490_Argument_0_0 =  UA_ExtensionObject_new();
opcua_node_variable_t_i_11490_Argument_0_0->encoding = UA_EXTENSIONOBJECT_ENCODED_BYTESTRING;
opcua_node_variable_t_i_11490_Argument_0_0->content.encoded.typeId = UA_NODEID_NUMERIC(0, 296+ UA_ENCODINGOFFSET_BINARY);
if(UA_ByteString_allocBuffer(&opcua_node_variable_t_i_11490_Argument_0_0->content.encoded.body, 65000) != UA_STATUSCODE_GOOD) {}

\\ Encode values as binary part of the extensionObject (as we cannot create UA_DataType members yet):
size_t opcua_node_variable_t_i_11490_Argument_0_0_encOffset = 0;
UA_String_encodeBinary(&opcua_node_variable_t_i_11490_Argument_0_0_struct.Name, &opcua_node_variable_t_i_11490_Argument_0_0->content.encoded.body, &opcua_node_variable_t_i_11490_Argument_0_0_encOffset);
UA_NodeId_encodeBinary(&opcua_node_variable_t_i_11490_Argument_0_0_struct.DataType, &opcua_node_variable_t_i_11490_Argument_0_0->content.encoded.body, &opcua_node_variable_t_i_11490_Argument_0_0_encOffset);
UA_Int32_encodeBinary(&opcua_node_variable_t_i_11490_Argument_0_0_struct.ValueRank, &opcua_node_variable_t_i_11490_Argument_0_0->content.encoded.body, &opcua_node_variable_t_i_11490_Argument_0_0_encOffset);
UA_UInt32_encodeBinary(&opcua_node_variable_t_i_11490_Argument_0_0_struct.ArrayDimensions[0], &opcua_node_variable_t_i_11490_Argument_0_0->content.encoded.body, &opcua_node_variable_t_i_11490_Argument_0_0_encOffset);
UA_LocalizedText_encodeBinary(&opcua_node_variable_t_i_11490_Argument_0_0_struct.Description, &opcua_node_variable_t_i_11490_Argument_0_0->content.encoded.body, &opcua_node_variable_t_i_11490_Argument_0_0_encOffset);
opcua_node_variable_t_i_11490_Argument_0_0->content.encoded.body.length = opcua_node_variable_t_i_11490_Argument_0_0_encOffset;

\\ Because calcSizeBinary went missing at some point, reallocate the overly long bytestring buffer to be exactly the right size and copy the encoded value
UA_Byte *opcua_node_variable_t_i_11490_Argument_0_0_newBody = (UA_Byte *) UA_malloc(opcua_node_variable_t_i_11490_Argument_0_0_encOffset );
memcpy(opcua_node_variable_t_i_11490_Argument_0_0_newBody, 
opcua_node_variable_t_i_11490_Argument_0_0->content.encoded.body.data, opcua_node_variable_t_i_11490_Argument_0_0_encOffset);

\\ Swap pointer to the data in the exstensionObject from the 65k buffer to the new one
UA_Byte *opcua_node_variable_t_i_11490_Argument_0_0_oldBody = opcua_node_variable_t_i_11490_Argument_0_0->content.encoded.body.data;
opcua_node_variable_t_i_11490_Argument_0_0->content.encoded.body.data = opcua_node_variable_t_i_11490_Argument_0_0_newBody;

\\ free overly long buffer
UA_free(opcua_node_variable_t_i_11490_Argument_0_0_oldBody);

\\ This is part of the generic value encoding from the UA_Value supertype
\\ If a value exists, it has been encoded in UA_ExtensionObject opcua_node_variable_t_i_11490_variant_DataContents by the value

attr.value.type = &UA_TYPES[UA_TYPES_EXTENSIONOBJECT];
UA_ExtensionObject opcua_node_variable_t_i_11490_variant_DataContents[1]; \\ Array strategy
[... cut ...]
UA_free(opcua_node_variable_t_i_11490_Argument_0_0);
UA_Variant_setArrayCopy( &attr.value, &opcua_node_variable_t_i_11490_variant_DataContents, (UA_Int32) 1, &UA_TYPES[UA_TYPES_EXTENSIONOBJECT]);

\\Continue in macroHelper
UA_NodeId nodeId = UA_NODEID_NUMERIC(0, 11490);
```

The problem is that I do not see an error during encoding. The bytestring is 41 bytes long and the encoding checks out. Reading the value also works as far as I can tell. But sending it results in an BadEncodingLimits exceeded.

Thanks for you help here @jpfr

jpfr, 2015-12-17:
Hey @ichrispa,

I think I got it.
- Fixed a bug on master
- There is still a bug in the generator. It does not encode the length of arrays inside the extensionobject.

Here's a working version of an encoded UA_Attribute value with some manual changes.
In addition to adding the array length, I made some simplifications that are possible only now.
http://paste2.org/ImE3VEpe

Once we can generate UA_DataTypes, we could also just make the value variant point to the struct in memory. That would be awesome. :-)

ichrispa, 2015-12-17:
Hi @jpfr,

perfect :) I would never have come up with the idea to look for a bug in master in this regard, I was fixed on having the generation wrong.

> There is still a bug in the generator. It does not encode the length of arrays inside the extensionobject.

Do you mean this part?

```C
struct {
  UA_String Name;
  UA_NodeId DataType;
  UA_Int32 ValueRank;
  UA_Int32 ArrayDimensionsSize;
  UA_UInt32 *ArrayDimensions;
  UA_LocalizedText Description;
} opcua_node_variable_t_i_11490_Argument_0_0_struct;
opcua_node_variable_t_i_11490_Argument_0_0_struct.Name = UA_STRING_ALLOC("SubscriptionId");
opcua_node_variable_t_i_11490_Argument_0_0_struct.DataType = UA_NODEID_NUMERIC(0, 7);
opcua_node_variable_t_i_11490_Argument_0_0_struct.ValueRank = (UA_Int32) -1;
opcua_node_variable_t_i_11490_Argument_0_0_struct.ArrayDimensionsSize = 1;
opcua_node_variable_t_i_11490_Argument_0_0_struct.ArrayDimensions = (UA_UInt32 *) UA_malloc(sizeof(UA_UInt32));
opcua_node_variable_t_i_11490_Argument_0_0_struct.ArrayDimensions[0]  = (UA_UInt32) 0;
```
In particular that last line?

That's because that's exactly what the XML says about the node:

```XML
           <Body>
            <Argument>
              <Name>ServerHandles</Name>
              <DataType>
                <Identifier>i=7</Identifier>
              </DataType>
              <ValueRank>1</ValueRank>
              <ArrayDimensions />
              <Description p5:nil="true" xmlns:p5="http://www.w3.org/2001/XMLSchema-instance" />
            </Argument>
          </Body>
```

I tried setting 

```
              <ArrayDimensions>15</ArrayDimensions>
```

Which shows up just fine. But I can't seem to find any examples for multi-dimensional arrays, like [3,3,1], to test that out (<ListOfArrayDimension> and <UInt32></UInt32><UInt32></UInt32>... don't appear to work). Do you have any example for those?

UAModeler also does not really support multidimensional input arguments, I can just set a single dimension here.

The namespace compiler does have provisions for the multidimensional array in the value() XML parsing subroutine and in code generation. But I haven't found an example for arguments to test it.

Regards,

ichrispa

jpfr, 2015-12-18:
> There is still a bug in the generator. It does not encode the length of arrays inside the extensionobject.

I meant that I manually had to insert the line
`UA_Int32_encodeBinary(&opcua_node_variable_t_ns_1_i_6001_Argument_0_0_struct.ArrayDimensionsSize, &opcua_node_variable_t_ns_1_i_6001_Argument_0_0->content.encoded.body, &opcua_node_variable_t_ns_1_i_6001_Argument_0_0_encOffset);`
But that only works in this particular case.

Since array lengths are encoded as `size_t` internally, but null-arrays are encoded as -1. There is a function `Array_encodeBinary` (currently not exported) in ua_types_encoding_binary.c that does the right thing automatically.

ichrispa, 2016-01-12:
I'm referring this to #485. The principle bug is solved. Thanks again @jpfr

closed 2016-01-12 by ichrispa