open62541/open62541 #485: Method argument generation from XML uses a dirty workaround

ichrispa, 2015-12-17:
@All: Please ignore this one, it's a work item that I should handle in the near feature. It just serves as a documentation of "why".

The namespace compiler treats all nodes as nodes. It does currently not know that a method has arguments that are variable nodes or that there are structs that represent these. The method node and the variable nodes for the arguments are constructed independantly from each other.

ua_nodemanagement.c contains a dirty workaround: an argumentSize < 0 inhibits the automatic creation of arguments (which is very much desired for userspace though). Hence the namespace compiler can create methods from XML definitions, but it's not a nice solution.

The namespace compiler should be patched in this regard.

jpfr, 2015-12-18:
Note that we now use `size_t` practically eveywhere for array lengths.

The documentation for internal array handling is here:
http://open62541.org/doc/current/datatypes.html#array-handling

I know, I'm a heckler for consistency. Do what fits you best.

ichrispa, 2016-04-15:
Closed and fixed since the last iteration....

closed 2016-04-15 by ichrispa