open62541/open62541 #492: Towards chunking while encoding

Stasik0, 2015-12-25:
okay, xmas is here, time to get chunking working

@jpfr @FlorianPalm @ichrispa: I need some feedback on the architectural questions, do you think its gonna be okay, since the changes affect encoding and will make it instable again

There are 2 strategies possible in general:

* encode full message and cut it in chunks -> encoder can be left unchanged, however we need a buffer for the full message which may be arbitrary large

* make encoder to be able to "skip" the already encoded chunk while it runs recursively -> this is what I have chosen, it means:
    * every encode function becomes a "skip" size_t parameter, as long as skip is greater than 0 the function does not do anything
    * we interpret the error messages ```UA_STATUSCODE_BADENCODINGERROR``` of the encoder as "buffer has ended, but there is more stuff to go"

So the strategy to make chunks in ```ua_securechannel.c``` will be to call encoder several times on the actual message in [this line](https://github.com/open62541/open62541/blob/master/src/ua_securechannel.c#L138) (I will assume that no chunking will happen in the header stuff, since a minimal chunk size is 8192), as long as we receive ```UA_STATUSCODE_BADENCODINGERROR``` from the encoder we just accumulate offsets from the previous chunks as the skip parameter. Getting the offest from the previous chunk will automatically ensure the skip is always valid e.g. we do not try to skip a half of an integer.

okay, i started to implement the stuff in a separate branch, please look into the array encoder to see how it works, __CODE NOT REALLY TESTED__

so guys, is it a valid approach to go? it is actually an additional check ```skip > 0``` per single encoder call.

HolgerJeromin, 2015-12-25:
Do you want to release version 0.2 before or after this change?

Stasik0, 2015-12-25:
Do we have to decide now? User api is not changed.

HolgerJeromin, 2015-12-26:
But stability probably :-)

jpfr, 2015-12-26:
1. What is the interplay with encryption? Is the entire message encrypted and then cut into chunks or is it encrypted separately?
2. We have a max chunk size and a max chunks count. So messages always have an upper bound on size.
3. If you activate the "readspeed" test case, you can see the effects on speed.

Imho the second approach is way more difficult to code and debug. And we lose speed since we have to "replay" all prior messages. So for 10 chunks, the first chunk is encoded 10 times...

There is one "evil" but C-conformant way to get the same speed with the second approach:
Using coroutines with setjmp/longjmp. Maybe we can avoid the encoding "replay" and jump straight back to where we left the encoding..

Stasik0, 2015-12-26:
> 
> What is the interplay with encryption? Is the entire message encrypted and then cut into chunks or is it encrypted separately?
Separately if i got it right
> We have a max chunk size and a max chunks count. So messages always have an upper bound on size.
Maxmessagesize in hellio message can be set to 0=arbitrary, i am not sure if max chunk count cannot be 0 aswell. 

Still, ns0 has pretty large certificates, so mallocing hundreds of kb rightaway suck, imho.
> If you activate the "readspeed" test case, you can see the effects on speed.
> Imho the second approach is way more difficult to code and debug. And we lose speed since we have to "replay" all prior [...]
This is genious, i will give it a try today or tomorow!

Stasik0, 2015-12-26:
the first try is segfaulting, i cannot understand valgrind messages, the problem occurs after the Array_encodeBinary returns after being jumped into... error is ```Address 0x1d is not stack'd ```... well obviously there is a "1" threaten as an address... help help help

jpfr, 2015-12-27:
Wikipedia says:

> If the function in which setjmp was called returns, it is no longer possible to safely use longjmp with the corresponding jmp_buf object. This is because the stack frame is invalidated when the function returns. Calling longjmp restores the stack pointer, which—because the function returned—would point to a non-existent and potentially overwritten or corrupted stack frame.

Stasik0, 2015-12-27:
So we need a second longjmp to simulate return?

jpfr, 2015-12-27:
Sounds good.
Let's give names to places in the code.

A loop for setting up the buffer, etc. before decoding. With a setjmp `a` within the loop.
A second setjmp `b` when decoding aborts prior to jumping to `a`. Then we do a loop iteration in `a`, send the msg, replace the buffer and jump back to `b`.

Stasik0, 2015-12-29:
okay guys, here is one working version with array encoding - look into the 'high-level' code in the test file... well the code looks like coming straight from hell, but its kinda seems to work... is something like this acceptable?

jpfr, 2015-12-30:
Sweet! Great stuff!
Now let's simplify the sh** out of it.

1. `UA_DataType* jmp_type`, `void* jmp_src`, `size_t jmp_length` don't require on-the-fly replacement imho. Only the buffer and the offset.
2. We need a way to recover from errors when builtin data is partially encoded before the buffer ends. Say we encoded half a NodeId.
  - Chunks don't need to exactly fill the available buffer
  - Redo encoding the NodeId by setting the offset to a saved value
  - Exchange the buffer and retry encoding the same NodeId
3. Imho there are two places that need to exchange the buffer: `Array_encodeBinary` and `UA_encodeBinary`. A single function `UA_StatusCode exchangeBuffer(UA_ByteString **bufPointer, size_t *offset)` could do the job for both of them.
4. We can export the jmp_buf directly instead of pointers to jmp_buf. The statuscode for the exchangeBuffer function (e.g. chunks are not supported, the message is already too long, or the buffer could not be allocated) can be passed with the longjmp back.

Stasik0, 2015-12-30:
> 
> Sweet! Great stuff!
> Now let's simplify the sh** out of it.
> 
> UA_DataType* jmp_type, void* jmp_src, size_t jmp_length don't require on-the-fly replacement imho. Only the buffer and the offset.
Yes, but imho they need to be restored and get clobered otherwise (since may be saved in a register)
> We need a way to recover from errors when builtin data is partially encoded before the buffer ends. Say we encoded half a NodeId.
> Chunks don't need to exactly fill the available buffer
> Redo encoding [...]
This is simple: i'd propose that every function except array just jumps back if there is no space - so jmp instead of returning error.
> 
> Imho there are two places that need to exchange the buffer: Array_encodeBinary and UA_encodeBinary. A single function UA_StatusCode exchangeBuffer(UA_ByteString **bufPointer, size_t *offset) could do the job for both of them.
Yea, the function is there (reinit buffer),
> We can export the jmp_buf directly instead of pointers to jmp_buf. The statuscode for the exchangeBuffer function (e.g. chunks are not supported, the message is already too long, or the buffer could not be allocated) can be passed with the longjmp back.
Pointer scan be tested for null while buffers are opaque a cannot be tested for "being initialized" directly. In this way we have a defined way to enable and disable coroutine behaviour (enable/disableResume) of the encoder. The second pointer is used to indicate that encoder successfully finished and the while loop can be canceled.

jpfr, 2015-12-30:
> Yes, but imho they need to be restored and get clobered otherwise (since may be saved in a register)

Register-values are stored in the jmp_buf and are restored from there. So clobbering happens because an updated value is overwritten when restoring the registry dump.

> Pointer scan be tested for null while buffers are opaque a cannot be tested for "being initialized" directly.

I meant *always jump and determine the action at the target location*.
That removes some control-flow complexity. But other solutions are also ok.

Stasik0, 2015-12-30:
Hmm, can you make a suggestion? Simpler code would be great!

Stasik0, 2015-12-30:
okay, @jpfr made a quantum-leap and got something working, however the solution is too fragile (there is a magic offset that needs to be adjusted for every platform/compiler... we should probably built in some different options like 1) chunking with a buffer that is stable but requires an upper bound of message size and buffer of that length and 2) chunking with longjmp which is quick but fragile

post 0.2 issue?

jpfr, 2015-12-31:
After discussion with @Stasik0 it seems the best way to go is to use a callback pointer instead of longjmp. It's just too messy.

ichrispa, 2016-01-05:
Far too late of course, but I would have picked the first approach (because it's simpler). 

Create a max_chunk_size buffer, then encode into it as long as it is not full. When the buffer is full, stop the encoding, pack it into a secureChannel layer, send it as a chunk (A-Type), then continue.

@jpfr: Signing/Encryption affects the entire message (before chunking). Segmentation, security and transport header are appended unencrypted. Decryption can only take place after all chunks are gathered.

But again, too late. Nice job though :)

FlorianPalm, 2016-01-08:
I was thinking what would be if we introduce an iterative version of UA_encodeBinary. This would give us the the possibility to interrupt the encoding when the send buffer is full. However, it would be necessary to keep track of the encoding state to be able to jump back into the encoding after the packet is sent. Basically the changes would mostly concern UA_encodeBinary and UA_SecureChannel_sendBinaryMessage.
I am not sure if I got it right with the callback function but this implies that we couple the encoding functions with the connection object in order to get a new send buffer, doesn't it?

Stasik0, 2016-01-08:
A callback function does not mean coupling - a callback  is just a pointer, it can be an arbitrtry one or even empty.

The coupling will only happen before sendBinaryMessage where a callback  to send the full buffer and get a new lne is formed and passed to the encoder.

So going iterative can be okay, but avoiding coupling is not an argument, imho.

FlorianPalm, 2016-01-12:
After some discussions with @Stasik0 (thanks!) I have tried the callback approach. It seems to work so far (chunking_callback branch). There is only chunking on server side atm which can be tested with a new binary server_chunking. Feel free to try it.

closed 2016-06-06 by jpfr