open62541/open62541 #493: QNX and ARM compilation

tpiat, 2015-12-30:
Latest version of open62541 do not copmpiles in QNX and ARM processors.
1. No CLOCK_MONOTONIC_RAW constant in QNX.
Can I use  CLOCK_MONOTONIC instead?

I insereted to open62541.c
```
UA_DateTime UA_DateTime_nowMonotonic(void) {
#ifdef _WIN32
    LARGE_INTEGER freq, ticks;
    QueryPerformanceFrequency(&freq);
    QueryPerformanceCounter(&ticks);
    UA_Double ticks2dt = UA_SEC_TO_DATETIME;
    ticks2dt /= freq.QuadPart;
    return (UA_DateTime)(ticks.QuadPart * ticks2dt);
#else
    struct timespec ts;
#ifdef __QNX__
#define CLOCK_MONOTONIC	2
    extern int clock_gettime( clockid_t clock_id, struct timespec *tp);
    clock_gettime(CLOCK_MONOTONIC, &ts);
#else
    clock_gettime(CLOCK_MONOTONIC_RAW, &ts);
#endif
    return (ts.tv_sec * UA_SEC_TO_DATETIME) + (ts.tv_nsec / 100);
#endif
}
```

2. No endian macros in ARM.
I insereted to open62541.h
```
/* Mixed Endian */
#ifdef __ARM_ARCH_4T__
# define UA_MIXED_ENDIAN
# define UA_NON_LITTLEENDIAN_ARCHITECTURE
# if ( __BYTE_ORDER == __LITTLE_ENDIAN )
# define htole16(x) (x)
# define htole32(x) (x)
# define htole64(x) (x)
# define le16toh(x) (x)
# define le32toh(x) (x)
# define le64toh(x) (x)
#endif
#endif
```

Stasik0, 2015-12-30:
Is it working?

uleon, 2015-12-30:
Instead of inserting all the ifdefs in UA_DateTime_nowMonotonic and make it almost unreadable  I would like to suggest that you added somewhere in open62541.h
```
#ifdef __QNX__
#include <time.h> // prototype for clock_gettime
#ifndef CLOCK_MONOTONIC_RAW
#define CLOCK_MONOTONIC_RAW CLOCK_MONOTONIC
#endif
#endif 
```

tpiat, 2015-12-30:
I'm testing.
Will continue tomorrow.

tpiat, 2015-12-31:
Probably works.

jpfr, 2016-01-05:
Hey @tpiat,
can we close the issue?

closed 2016-01-05 by tpiat