open62541/open62541 #496: MonitoredItem marked as changed in MonitoredItem_QueuePushDataValue but value remains the same

NetoLe, 2016-01-04:
this is a bit weird. I hope I got it right. 

The function MonitoredItem_QueuePushDataValue calls a part:
```C
if(UA_String_equal(&newValueAsByteString, &monitoredItem->lastSampledValue) == UA_TRUE) {
            UA_DataValue_deleteMembers(&newvalue->value);
            UA_free(newvalue);
            UA_String_deleteMembers(&newValueAsByteString);
            return;
```
but UA_String_equal always returns false and not even once true. 
so my Server always returns a positive response even if nothing changed. The value at all always remains the same. 

I tried to track the problem and following occured:
[image]
[image]

short description: string1 and string2 are in two locations (which you see at the address), which means that the pointer is just changed from string1 to string2 and other way round all the time when MonitoredItem_QueuePushDataValue is called.
Where I see the problem: there is a hard coded 512 length and it seems that the 512 Bytes are not for the string itself completely and get overwritten in parts as you see in the memory dump. the distance between both strings is a bit more than 512 Bytes.
could a temporary strlen() be a workaround?

- I am using the newest master branch (which caused problems because aligned memory access and endian-issues appeared because there is no more check for architecture like in the version before christmas, and not every framework or compiler-toolpackage has implemented htole16 and stuff-functions, I needed to comment those functions out)
- compiled for an ARM device with gcc arm-elf-v2
- I am not using xx_SAFE- or LIST_REPLACE, because that causes compiler errors, like in https://github.com/open62541/open62541/issues/476 so I use workarounds from older commits without temp-variables.

Stasik0, 2016-01-04:
okay, one by one:
1) endinannes - please open a different issue so we can discuss it there, we can revert the commit if there are problems (it was surely not intended)
2) xx_SAFE in #476 you were asked for the toolchain so we can analyze it a bit more 

now to the issue:
is the changed value i.e. its serialization longer than 512? What is returned by ´´´newValueAsByteString´´´?

Stasik0, 2016-01-04:
the problem begins somewhere here:
https://github.com/open62541/open62541/blob/19fb205713f9c092e3ec695c970c7bcb76406dec/src/server/ua_subscription.c#L470

there is a hard-coded 512 hack, however the UA_encodeBinary is buffer-safe i.e. it does not exceed the buffer-length

Stasik0, 2016-01-04:
@NetoLe is it possible to get a minimal-reproducible bug as c-code?

Stasik0, 2016-01-04:
As a quick fix - try to crank up 512 to some higher value to see if it helps. Is the issue happening on all variables or only on particular - big ones?

NetoLe, 2016-01-04:
ok, sorry so far.
the value is not longer than 512. It's a "84" decimal UA_Int64.
here are the values: 
[image]
The PC is at line 488 ua_subscription.c at this point. (how do I link to lines in the code?)
they are the same. 
I made a bad workaround which is so bad I am ashamed to show but for now this works:
```C
UA_Boolean UA_String_equal(const UA_String *string1, const UA_String *string2) {
    size_t length;
    if(string1->length != string2->length)
        return UA_FALSE;
    if(string1->length == 512) 
    {
      length = strlen(string1->data);
    }
    else
    {
      length = string1->length;
    }
    UA_Int32 is = memcmp((char const*)string1->data, (char const*)string2->data, length);
    return (is == 0) ? UA_TRUE : UA_FALSE;
}
```

hm I don't know if I can reproduce this whith just code. I tried it with your examples on my local machine and there it is working fine. 

but I will try.
my main looks like this:
```C
    server = UA_Server_new(UA_ServerConfig_standard);
      if(server)
      {
        UA_Server_addNetworkLayer(server, ServerNetworkLayerTCP_new(UA_ConnectionConfig_standard, 4840));
      
        ServerManager::initNodes();
        while (server) // device interna can kick the server out of the tcpip-ressource, the loop get's it back in
        {
          if (running)
            UA_Server_run(server, 1, &running); // no loop, the server itself has a built-in loop
          XXX::XXX::XXX::sleep(2000);
          running = UA_TRUE;
        }
       
      }
```

the initNodes() initializes a lot of nodes( ~80), here is just one, but all variable-nodes are initalized the same just with UInt32 and Int16. I am working with valuecallbacks (great idea btw.), which are called with every publishrequest and then update the value for the variable (this did work with a polling-functionality with just read-services from client-side every 500 ms as well). I initialize only numeric node-identifiers.
```C
              attr.description = UA_LOCALIZEDTEXT_ALLOC("en_EU","OK");
              attr.displayName = UA_LOCALIZEDTEXT_ALLOC("en_EU","OK");
              UA_Int64 xxxOKValue = 1000;
              UA_Variant_setScalarCopy(&attr.value, &xxxOKValue, &UA_TYPES[UA_TYPES_INT64]);
              statuscode = UA_Server_addVariableNode(server, UA_NODEID_NUMERIC(namespaceIndex, XXXDEFINE_OK_SR), UA_NODEID_NUMERIC(namespaceIndex, XXXDEFINE_SR), UA_NODEID_NUMERIC(0,UA_NS0ID_HASCOMPONENT), UA_QUALIFIEDNAME_ALLOC(namespaceIndex, "OK"), UA_NODEID_NULL, attr, NULL);       
              UA_Variant_deleteMembers(&attr.value);
              UA_Server_setVariableNode_valueCallback(getUAServer(), UA_NODEID_NUMERIC(namespaceIndex, XXXDEFINE_OK_SR),
                                        valuecallback);
```

this are the both parts where I use the open62541-functions directly on the server.

the client looks like this:

```C
static void handler_TheAnswerChanged(UA_UInt32 handle, UA_DataValue *value) {
	printf("The Answer has changed!\n");
	return;
}
int main(void)
{
        char *ServerURL = "opc.tcp://192.168.60.8:4840"; // here you put in your ServerURL.
	char *result;
	printf("Try to connect to client at %s\n", ServerURL);
	client = UA_Client_new(UA_ClientConfig_standard, Logger_Stdout); // a new client
	UA_StatusCode retval = UA_Client_connect(client, ClientNetworkLayerTCP_connect, ServerURL); // connect to the client
	if (retval != UA_STATUSCODE_GOOD)
	{
		UA_Client_delete(client);
		printf("No connection possible\n");
		return retval;
	}
	UA_UInt32 subId;
	UA_Client_Subscriptions_new(client, UA_SubscriptionSettings_standard, &subId);
	if(subId)
		printf("Create subscription succeeded, id %u\n", subId);

	// .. and monitor TheAnswer
	UA_NodeId monitorThis = UA_NODEID_NUMERIC(2, XXXDEFINE_OK_SR);
	UA_UInt32 monId;
	UA_Client_Subscriptions_addMonitoredItem(client, subId, monitorThis,
		UA_ATTRIBUTEID_VALUE, &handler_TheAnswerChanged, NULL, &monId);
	if (monId)
		printf("Monitoring 'the.answer', id %u\n", subId);
	// First Publish always generates data (current value) and call out handler.
	UA_Client_Subscriptions_manuallySendPublishRequest(client);

	// This should not generate anything
	UA_Client_Subscriptions_manuallySendPublishRequest(client);
			while (1)
			{
				UA_Client_Subscriptions_manuallySendPublishRequest(client);
				Sleep(1000);
			}
return 0;
}
```

you see I just took your example and changed it a bit for my purposes to check the functionality.

Stasik0, 2016-01-04:
@NetoLe okay, the buffer is just too long. Try replacing 512 by ```UA_calcSizeBinary(&newvalue->value, &UA_TYPES[UA_TYPES_DATAVALUE]);``` this should fix it for the moment, however a clean solution would use some mechanism of hashing

Stasik0, 2016-01-04:
@NetoLe please try https://github.com/open62541/open62541/tree/issue496, there i have a workaround @jpfr we need to fix it properly, the buffer size has to be known

jpfr, 2016-01-04:
Imho, a clean mechanism would let the user set a flag when something changed.

1. Commit 4523086d7095f990884049f375d09d3ff7e77264 hopefully ends the endianness-thing. I tried a lot of stuff. And the standard endianness macros memcpy is by far the fastest. Especially on native little-endian.
2. We need to get a definitive solution for MonitoredItems.

@NetoLe We'll get back to you when we have it.

ichrispa, 2016-01-04:
Just acknowledging the ticket, I'll give more detailed feedback tomorrow.

jpfr, 2016-01-04:
Hey @ichrispa,
welcome back. :-)

The easiest solution probably is to bring calcSizeBinary back (again...).
However, that binds our server core to the binary protocol. Not good if one wants to play with different encodings. I hear a JSON encoding is coming up in Part 14 with the Pub/Sub extension of UA. So supporting different encodings is definitely relevant.

As an alternative, I quickly coded up a function [1]
```c
hash_t hash_type(const void *p, const UA_DataType *type, hash_t seed); 
```

That takes any types and produces a hash. If anything changes (including the pointers to member-arrays), the hash will be different. That way, we can store a deepcopy of the value + the hash in the publishing queue. And only keep the last hash once the deepcopy was delivered.

So. Where am I thinking wrong?
I'm sure there are edge cases where this won't work...

[1] In the hash_type branch:
 `https://github.com/open62541/open62541/commit/a8d36d8da6177ce29a54200da6c2e6f0c3edba66

NetoLe, 2016-01-05:
thanks for the workaround! for now it does its job.

ichrispa, 2016-01-05:
Hi @NetoLe, @jpfr, @Stasik0,

monitoredItems need a way to see if the value changed. Right now, they encode the any value as binary, then use it as a comparison reference. That allows any type to be sampled and changes detected. IMHO there is no need to change this part - the binary encoding is not propagated to the client. Our stack will continue to support binary and I really see no reason to pick another mode of change detection since this one works.

A flag can be implemented if an event system would detect any writes - which it does not at this time. This would however again increase the memory needed to represent data. With regard to embedded use of our stack, I don't feel to warm about that (or JSON for that matter).

The hard coded string length is a result of @jpfr's short-time change to the stack. calcSizeBinary vanished back then, so both the namespace compiler and subscriptions desist from using it. 

About the bug: We had that before, and as far as I remember I fixed it back then.
Do not use strlen, strcmp or similar in UA_String_equal. C function terminate when a 0-Byte is read. I had changed that part from strcmp to memcmp back then for exactly that reason. When comparing encoded types (such as ints), it is quite likely they start with a 0 and hence the "are equal" to any C string function.

The solution here is to get the right binary string length. I would route for just uncommenting calcsizeBinary again. Alternatively, we could create calcSizeEncoded proxies to access the respective transport encoding of the connection?

Best regards,

ichrispa

Stasik0, 2016-01-06:
@jpfr hash looks good, @ichrispa i do not think that collisions should be a great issue, I mean we typically rely on a collision-free hash for our nodestore, do not we? Should we stick on hashing?

jpfr, 2016-01-06:
The nodestore uses hashes to find potential buckets but still compares the nodeids. Because the are only (hash % buckets) possibilities, collisions are way more common.

It's also a question of speed. Imagine thousands of variables with many subscriptions.
Hashing reduces the load on RAM but is still computationally intensive.

Imho we should let the user choose between options.
- Userland calls a `hasChanged` function on the nodeid after updating the value. That sets a "sample-me" flag in the relevant monitoreditems (or makes the sample immediately if the timeout is zero). **I'd encourage users to use that option.**
- Only if the variable is mapped to volatile RAM, we have no way of detecting changes without comparing the actual value. Remember, if the sampling rate is zero, we do that once in every mainloop iteration. We can compare via
   - encoding + memcmp
   - hashing
   - storing the last sample and comparing the content

   Given that the value is mapped to volatile RAM, we can treat it as a blob of size datatype->memSize. So storing the last sample and comparing is rather easy to do on a per-monitoreditem basis.

Stasik0, 2016-01-06:
does it make sense to add a data field instead of the flag? i think the spec has something like "samplingTime" for variables?

ichrispa, 2016-01-08:
I think the binaryzation of the data value is a more definite way of identifying changes (though hashes are cool in principle).

I think we should put the sample flag/datafield on a temporary hold. My next project will need chunking and events, so I see myself implementing events in the coming months. Once we have that, we can cause DataChangeEvents to be triggered for nodes, which should by design also trigger monitoredItems to resample their value. That solution would not only be spec-conformant (instead of a work around), it would also remove a ton of todo-items from the code and bring the subscription service forward. What do you think?

Stasik0, 2016-01-08:
+1 for hold of the flag. Still events do not solve the problem of identifying changes in an in-RAM variable that is tackled with hash or binary-memcmp.

jpfr, 2016-01-11:
Ok. So we resurrect calcSizeBinary to have a 100% solution for the 0.2 release.
Then @ichrispa can think about events without a looming release date.

ichrispa, 2016-01-12:
Give me a heads up once calcSizeBinary crops back up. This would also be nice-to-have for the namespace generator...

ichrispa, 2016-01-20:
Is this issue solved?

NetoLe, 2016-01-21:
so with the FIXME-part it is working, yes :)

Stasik0, 2016-01-21:
It is solved iff calcsizebinary is resurrected and plugged back into the FIXME part.

jpfr, 2016-01-26:
calcSizeBinary landed in 5fbf79ca39f1b3901acee602e13d315734707221.

Can we close this issue then?

Stasik0, 2016-01-26:
> calcSizeBinary landed in 5fbf79c.
>
> Can we close this issue then?

I will fix it asap

Stasik0, 2016-01-26:
oh, it is already by @jpfr, @NetoLe: please verify the fix and reopen if needed

closed 2016-01-26 by Stasik0