open62541/open62541 #497: current master branch 80eded cause problems in endiannes

NetoLe, 2016-01-04:
After I updated my server to the current master branch there is no more checking for endiannes which was implemented in the master before christmas. Now the endian-functions like le16toh are used which surely is an code improvement but this caused a bad behaviour for the toolchain and framework I am using, because there are no implemented endian-functions. (arm-elf-v2 and a self implemented framework as OS-Abstraction-layer)

So I needed to comment those endian-parts out and then my server was working again correctly. 
In my opinion the check for endiannes and aligned memory access like before was a good choice, because beeing more portable at first sight.

Stasik0, 2016-01-04:
this is probably this commit 06774a36ac864ee8c00afc5a2d6e2267d8512443

Stasik0, 2016-01-04:
@NetoLe please look if this works 4523086d7095f990884049f375d09d3ff7e77264

jpfr, 2016-01-04:
The current approach with memcpy + htole-macros is not only cleaner, but also significantly faster.
In order to support embedded platforms, we will rather provide byte-switching macros.
But in the absence of operating-system or compiler headers with endianess information, these need to be set manually.

The commit will follow in the coming days.

jpfr, 2016-01-05:
Hey @NetoLe,
defining the endianness of embedded platforms is now easy:
https://github.com/open62541/open62541/blob/master/include/ua_config.h.in#L103

Thank you for confirming if that works for you.
Than when we can close the issue.

NetoLe, 2016-01-05:
ah! yeah, I don't know why but it seems I didn't notice this yesterday. This now works fine and seems to be a nice solution. thank you!

closed 2016-01-05 by jpfr