open62541/open62541 #499: How to stop UaExpert Send peridoic ReadRequest to open62541Server?

mentospcg, 2016-01-06:
When I use UaExpert connect to the open62541Server, the UaExpert peridoic send ReadRequest to the server, about 5 seconds per one. How to stop it ? 

thanks a lot！

Stasik0, 2016-01-06:
It is probably reading server status to check if the server is alive. 

Is this behavior seen with other servers or only with open62541?

mentospcg, 2016-01-07:
@Stasik0 only open62541 got this phenomenon. when use UaExpert connect to the OPC foundation UA Sample server does not have this phenomenon.

The periodic ReadRequest wiresharp capture as below:
[image]

Stasik0, 2016-01-07:
It is reading node 2259. Maybe it is trying to simulate a subscription by polling? Do you have subscriptions activated in open62541 server?

You can also try to raise a question on uaexpert forum.

jpfr, 2016-01-07:
ns=0,i=2259 ist the "State" variable. That is frequently read by UAExpert.

> when use UaExpert connect to the OPC foundation UA Sample server does not have this phenomenon.

I cannot confirm this. On my box, connecting to the official C Server example from UnifiedAutomation results in the same behavior. Here's a screenshot of the network dump:
[image]

@mentospcg Can you tell us your exact configuration so we can reproduce?

destogl, 2016-01-07:
Just today was talk about that in the [python-opcua repository](https://github.com/FreeOpcUa/python-opcua/issues/106). There was cited the specification:

> The preferred mechanism for a Client to monitor the connection status is through the keep-alive of the Subscription. A Client should subscribe for the State Variable in the ServerStatus to detect shutdown or other failure states. If no Subscription is created or the Server does not support Subscriptions, the connection can be monitored by periodically reading the State Variable.

jpfr, 2016-01-08:
Hey Denis, nice find. :-)
I guess we can close the issue then.

closed 2016-01-08 by jpfr