open62541/open62541 #503: Initializing object instances is missing the instantiation callback

ichrispa, 2016-01-12:
Hi @jpfr,

I have a minor problem: I am initializing a complex object type which contains variables. These need to be initialized for each instance. Currently, creating objects just copies all hasComponents/hasProperties, but there is no way to change them as they are created.

My old addInstance method used to pass a handle to an instiantiation callback along which was called for each node being initialized (containing its ID, type ID and a handle). That way, specific nodes could be "customized" when they where being instantiated. The method was really handy when creating complex object hierarchies. As far as I see, that callback didn't make it to the revised HL API.

Is there any specific reason it was left out? Are there any objections to me putting it back in? 

Regards,

ichrispa

Wtango, 2016-01-12:
I use UA_Server_forEachChildNodeCall to  custumize a complex object :).

ichrispa, 2016-01-12:
@Wtango: You can use the node iterator for that, however I wrote that with search operations in mind. The old instantiation callback provided you with the nodeID of the template (name independant), which makes it far easier than string comparisons to ID the source type. Here's the old instantiation callback from my hl-branch:

```C
typedef UA_StatusCode (*UA_InstantiationCallback)(UA_NodeId objectId, UA_NodeId definitionId, void *handle);
```

The old instantiation only referred to objects and objectTypes
```C
UA_StatusCode UA_Server_addInstanceOf(UA_Server *server, UA_NodeId nodeId, const UA_QualifiedName browseName,
                        UA_LocalizedText displayName, UA_LocalizedText description, const UA_NodeId parentNodeId, 
                        const UA_NodeId referenceTypeId, UA_UInt32 userWriteMask, UA_UInt32 writeMask, 
                        const UA_ExpandedNodeId typeDefinition, UA_InstantiationCallback callback, void *handle, 
                        UA_NodeId *createdNodeId) 
```

Wtango, 2016-01-12:
Cool,this stuff is both handy and useful.

https://github.com/open62541/open62541/issues/503#issuecomment-170938093

jpfr, 2016-01-13:
Have you seen the object lifecyclemanagement?
https://github.com/open62541/open62541/blob/master/include/ua_server.h#L222

Basically, you can set a "constructor" of an object type. Then, whenever such an object is instantiated (by yourself or over the network), the constructor is called with the nodeid of the new object.
The constructor returns a void pointer "object handle" that is attached to the object.
If a destructor is defined, you are again called with the nodeid and the handle.

Does that cover your use cases?

ichrispa, 2016-01-13:
Hi @jpfr,

I ran across the olm system when I looked into it - and I applaud the idea. The use case here however is ininitializing variable sources for the objects contents. That is an object hasComponents in the type definition, but I don't want them to be initialized with the objects proper value (sources). 

The only way this can be done right now is to have the object constructor in place. We then need to attach the constructors procedurally to each type definition, since it is not an opc ua attribute and cannot be modelled with say ua modeller. I then have to call the node iterator in the constructor and examine each typeDefinition node for the variables to determine which one it is.

I really just need a function to be called saying "I initialized this object from that template". The only way I can think of implementing that is passing a callback to addObject and addVariable that is called in initializeInstance. Is there a better way?

As far as the olm goes, I like destructors... to detach or clean up the data sources.

ichrispa, 2016-01-13:
Hi @jpfr,

I noticed the olm when looking into my problem. They do not cover the use case (though I do like the idea).

The use case is initializing object contents (like variables). Currently, a constructor needs to be attached procedurally to each type, since ua modeller cannot handle this non-standard property. The constructor is then called for the object only... which is mute, since I call addNode for that type anyway and can determine it's nodeId that way. But I cannot determine the variables contained in the object.

So now, the contructor needs to browse/iterate all the objects variable nodes, determine the hasTypeDefinition of each and then call the proper init procedure. I think it is much easier just having initializeObject|Variable call a callback ("I have copied this template node to form that instance"). However this would require to hand down the callback through addObject to addNode to addSingle to initializeInstance... 

Can you think of a  better way to achieve what my old callback did?

jpfr, 2016-01-13:
Okay. So you want to initialize a variable inside an object.
I see three possibilities.

**a) Loop through a callback for variables inside the object to the constructor**
To me that is difficult, because the pattern works only for the exact use case you have.

What if there are more than one variable that needs initialization?
What about objects containing objects containing variables containing variables?
That seems a bit absurd, but somebody could have that use case.

**b) Define a lifecycle with constructor / destructor for variabletypes also**
Then you can simply add variables of a special variabletype with a constructor to the object.
What about variables instantiated inside an objecttype?
Maybe we don't want to call the variable constructor then.

**b) Change nothing and use 'UA_Server_browse' in the object constructor**
All variables in an object are instantiated before the constructor is called.
Imho using 'UA_Server_browse' inside the constructor to get the variables won't take more than 10 LoC.

ichrispa, 2016-01-13:
(a) is we already have and why I am disgruntled.
(b) unnecessarily complicates things; we are not writing an object oriented C++ stack, so I see no requirement to adher to complex OO principles when a simple procedural solution will do the trick. Further more, that would not cover an ObjectType containing variables, only types.
(c) is inacceptable.

b has an additional drawback: I know how I want to initialize specific objects and there are multiple of the same type. So the constructor is not static, but changes with every instance I want to initialize.  That would mean that I need to change the types constructor before every instantiation. 

The OO approach generates more overhead then benefit here. All I want (and this is not a complicated or implausable demand for a dynamic server) is that when a node is created from a template I have a way to determine (i) which nodeId it has and (ii) what template it was derived from.

Since there is no better solution at hand, the problem was solved (without influencing the lifecycle management in place).

jpfr, 2016-03-05:
We are now adding a lot of NULLs to the API call as only a fraction of the users needs the instantiation callbacks.
```c
UA_Server_addVariableNode(server, myIntegerNodeId, parentNodeId,
                          parentReferenceNodeId, myIntegerName,
                          UA_NODEID_NULL, attr, NULL, NULL);
```

@ichrispa What about if we the instantiation callback to the `UA_ServerConfig` with a default NULL?
This should cover most use cases. And we keep the addNodes API to a minimum of arguments.

ichrispa, 2016-03-08:
First of, I am not at all bothered by a trailing NULL - adding or removing nodes has not become more or less complex since the hl api was started. Some stuff was offloaded into structs, but in terms of usability not a lot changed. So what we could quite simply do if you are bothered by the number of arguments would be to offload the callback into the attr struct?

Can you explain how you reached the "most of our users don't user it" conclusion? My use case is quite definetely only settled by the callback.

jpfr, 2016-03-09:
> My use case is quite definetely only settled by the callback

Exactly, lets make sure your use case remains covered. Hence the reopened issue.

My current use case is to **track all new nodes** on the server. That's why I need the instantiation-callback also when the **node is added via the addNodes service over the network**. The only place to put that callback is the server config. And that's where my duplicate-functionality-radar went on.

So let me rephrase the question: If the callback was in the server config, would your use case be covered, albeit more complicated? I acknowledge that switching callbacks in the config before adding a node is kinda ugly. Do you have different instantiation callbacks or is it always the same?

In any case, that change would land via a PR that is only merged when and once you're ok with it.

PS: The attr argument of `UA_Server_add{Variable/Object/...}Node` cannot be changed easily. The datatype is autogenerated and is used by the standard UA service.

ichrispa, 2016-03-09:
Hi jpfr,

I'll let you in on the use case so you can better gauge the problem. Maybe you can see a solution that I am missing: 

I am using the stack in C++. The classes get|setSomething functions are hooked to the namespace's variable instance with dataSources. When addNodes instantiates the object in UA, I only need to check the nodeId of the template and attach a DataSource to the variables to call the specific class's getSet function. That's why I need a callback to be active on each variable and I need that callback to be passed the template nodeId used during instantiation. I hence have a different callback not only for each class type, but also per instance (the class instance is passed as handle).

The deciding question in my use case is how the server config can be accessed and modified (a) from a multithreaded  (b) userspace after the server is instantiated? afaik we do not have a get|setServerConfig HL interface at  the moment.

Both of our usecases are basically covered by modelChange Events - but I really do not have the time to implement event handling at the moment.

jpfr, 2016-03-30:
Hey @ichrispa,

I'm not quite sure what you mean by "template".
Maybe we can draw up an example together.

I was thinking of
- putting a newNodeCallback & deleteNodeCallback into the server config
- using a single `nodeHandle` void pointer for everything. Currently, we many different void handle pointers in the nodes. For value callbacks, method calls, constructor / destructor, ...

Here's how the API could work. In the order of use.
```c
/* returns the nodeid of the type of the node or UA_NODEID_NULL */
UA_NodeId getNodeType(const UA_NodeId *nodeId) {
    UA_BrowseDescription d;
    UA_BrowseDirection(&d);
    d.nodeId = *nodeId;
    d.browseDirection = UA_BROWSEDIRECTION_INVERSE;
    d.includeSubtype = true;
    UA_BrowseResult browseResult = UA_Server_browse(server, 0, const &d);
    UA_NodeId typeId = UA_NODEID_NULL;
    if(browseResult.referencesSize == 0)
        UA_NodeId_copy(&browseResult.references[0].nodeId.nodeId, &typeId);
    UA_BrowseResult_deleteMembers(&browseResult);
    return typeId;
}

/* 1) start adding the node */

/* 2) this callback is optionally defined in the server config called for every new node */
void * newNodeCallback(const UA_NodeId *nodeId) {
    /* return the nodeHandle that is then attached to the node. can be null. */
    
    /* once in the server config, I cannot replace the callback function.
     * But I can toggle switches from the outside.*/
    if(*myGlobalVariable) {
        
    }
    /* or use it as a trampoline */
    myGlobalFunctionPointer(nodeId);
}

/* 3) now all the children are instantiated if the node has an objecttype or variabletype */

/* 4) the constructor is optionally attached to the objecttype or variabletype */
void * myConstructor(const UA_NodeId *nodeId, void *nodeHandle) {
    /* return the nodeHandle or replace  */
}

/* 5) now we work with the node ... the nodeHandle pointer is forwarded to
 *    method calls, datasources and value callbacks */

/* 6) start deleting the node */

/* 7) the destructor is optionally attached to the objecttype or variabletype */
void * myDestructor(const UA_NodeId instance, void *nodeHandle) {
    /* the returned pointer replaces the nodeHandle (can be NULL) */
}

/* 8) all the children nodes are removed recursively here */

/* 9) this callback is optionally defined in the server config and
 *    called for every node that is deleted */
void removeNodeCallback(const UA_NodeId *nodeId, void *nodeHandle) {
    
}
```

ichrispa, 2018-02-01:
Old; we have callbacks.

closed 2018-02-01 by ichrispa