open62541/open62541 #505: Client loses connection to Codesys server after ca. 3 mins.

Aleskey78, 2016-01-14:
Hello,

after around 3 minutes a client loses a connection to server, see the picture below.
Testet on 476af29985.

[image]

Stasik0, 2016-01-14:
Can be a token renewal problem, can you get us a wireshark dump?

Aleskey78, 2016-01-14:
here is a dump:
[connectionlost.pcapng.zip](https://github.com/open62541/open62541/files/90122/connectionlost.pcapng.zip)

NetoLe, 2016-01-14:
so your open62541 client sends a correct renew token but the codesys Server (is that on Port 4840, right?) response is an Error; maybe the Codesys server has no renewal implementation for SecureChannels?
I would suggest that, because the open62541-client gets into an Error-State and initializes a new SecureChannel following with a new Hello-Message, which it only does if the response is not the response which was expected
Can you get an log output for debug purposes, when `#define UA_LOGLEVEL 200` is set. this is placed in the generated ua_config.h. 
I think you will get a `"Reply answers the wrong request. Expected OpenSecureChannelResponse."` if I'm not totally wrong.

Aleskey78, 2016-01-14:
Hi, thanks for your prompt response.
I set UA_LOGLEVEL to 200, as you suggested, now a compiler brings me an error: 'UA_Server' has no member named 'logger'.
I use an amalgamated version.
It looks like "server->logger" should be changed to "server->config.logger".

Aleskey78, 2016-01-14:
I've got these two messages:
[01/14/2016 11:09:49.597] debug/channel Requesting to renew the SecureChannel
[01/14/2016 11:09:49.610] debug/channel Receiving OpenSecureChannelResponse failed

Stasik0, 2016-01-14:
Alexey, could you please also dump the three minutes of communication with uaexpert? I would like to see how it renewes the token, it should be the OpenSecureChannel request.

Please open a new issue for the logger problem - we probably need to include the debug level into the CI, you can also open a new pull request if you can fix it ad hoc.

Aleskey78, 2016-01-14:
@Stasik0, uaexpert dump probably won't help:[uaexpert_3min.pcapng.zip](https://github.com/open62541/open62541/files/90354/uaexpert_3min.pcapng.zip)

OpenSecureChannel message: OpenSecureChannelRequest
    UaExpert: RequestedLifetime: 3600000
    Open62541-client: RequestedLifetime: 30000

Stasik0, 2016-01-14:
Is not it possible to change the value in uaexpert config?

NetoLe, 2016-01-14:
you can configure the lifetime in UaExpert with Settings-> Configure UaExpert 
than search for General.SecureChannelLifetime

Aleskey78, 2016-01-14:
the connection is stable, no drops after 3 mins: [uaexpert_lifetime_30000.zip](https://github.com/open62541/open62541/files/90378/uaexpert_lifetime_30000.zip)

Stasik0, 2016-01-14:
please camputre a bit more - there is no second OpenSecureChannelRequest
for renewal. Currently there is only one (packet No. 10). You can
alternatively crank the lifetime even more down to capture some
OpenSecureChannelRequests for renewal.

Stasik0, 2016-01-14:
from what I see is that, we set no client nonce, but i cannot test atm

NetoLe, 2016-01-14:
ok, but there is no renewal of the Secure Channel. It is a bit weird. UaExpert first seems to open a secure channel Request with a requested lifetime of 600000, and gets a response with a revised lifetime of 600000. then it closes the secure channel after getting the endpoints and opens one again. Then it wants a requested lifetime of 30000 but gets a revised lifetime of 600000 back. Which means that the lifetime will not be 30000, what causes that the problem does not occure because in three minutes there is no renewal. 
Maybe the Codesys-Server does not close the channel at all? Or reuses the old lifetime? 
what is weird too: I tried to re-enact the problem with my open62541-Server and UaExpert for understanding what is different. but If I change the lifetime to a value less then default I get a 'BadInvalidArgument on UaExpert' besides the open62541-client works with a 30000 lifetime. 
Besides that: the open62541-server returns a correct revisedLifetime of 30000, different than the Codesys does.

Stasik0, 2016-01-14:
we need to look that lifetime issues in the spec, does the server has to follow the request, or does the client has to follow the rewised lieftime. I suspect, that the server has priority. So much said, i cannot test atm

Aleskey78, 2016-01-14:
UaExpert ignores a  General.SecureChannelLifetime after reconnecting to PLC...
[uaexpert_lifetime_1000.zip](https://github.com/open62541/open62541/files/90435/uaexpert_lifetime_1000.zip)

Aleskey78, 2016-01-14:
setting secureChannelLifeTime to 600000 in a open62541-client didn't take any effect. After ~3minutes a connection was lost. It's all weird.
[client62541_lifetime_600000.zip](https://github.com/open62541/open62541/files/90524/client62541_lifetime_600000.zip)

NetoLe, 2016-01-14:
I had a look in the spec.
From Part 6 Mappings 6.7.4 Establishing a SecureChannel

"The Server shall return an error response as described in Part 4 if there are any errors with the parameters specified by the Client." 
-> Maybe this is what the Codesys Server did.

"The RevisedLifetime tells the Client when it shall renew the SecurityToken by sending another OpenSecureChannel request." 
-> I can't find a point where it tells whether the Client or the server tells the right lifetime, that should be used. But regarding the sentence up there I would suggest that the server RevisedLifetime has priority. 

"The OpenSecureChannel Messages are not signed or encrypted if the SecurityMode is None. The Nonces are ignored and should be set to null." 
-> The Nonces should be ok then with 00.

I've taken a look into Part 4 Services too. 6.1.4 Creating a SecureChannel
Figure 21 on p. 89
describes the token lifetime as a response from the server as well, so this should be the priority. Even when Part 6 tells in 6.7.4: "For this reason, the OpenSecureChannel Service is not the same as the one specified in Part 4."

At all I would say it is not really defined who decides which lifetime is valid or how the server handles a requested lifetime. I searched through all documents and "RequestedLifetime" is just mentioned as a parameter but never explained how to handle. Maybe this is just unspecified?

But please take a look for yourself, not that I'm telling wrong stuff.

Stasik0, 2016-01-14:
@NetoLe imho this is correct, we had already implemented the priority of revisedLifetime

The problem was in a wrong signed-unsigned conversion, now the lifetime seems to be applied correctly. @Aleskey78 could you please test?

The codesys server seemed to drop the connection if the token is renewed too early. I could not found the exact description of renewal timing in the spec (somehow i believed one should do it from 75% of revisedLifetime).. 

@jpfr @ichrispa @FlorianPalm any idea of this? Now we renew 2 seconds before expirations which may be not enough, e.g., if node is read with a 5 seconds period. Should we use a value that depends on revisedLifeTime and not an absolute one?

NetoLe, 2016-01-15:
Your believing is right.
depending on Part 4 1.03 p.19:
"a) Clients should request a new SecurityToken after 75 % of its lifetime has elapsed. This should ensure that Clients will receive the new SecurityToken before the old one actually expires." 
So I would suggest what you think in your last sentence is right.

Aleskey78, 2016-01-15:
@Stasik0, after chnages in cd5cec7 PLC doesn't pay any attention now to RequestedLifetime in OpenSecureChannelRequest and answers me with RevisedLifetime: 600000. Restarting the PLC has no effect.

Aleskey78, 2016-01-15:
@Stasik0, here is dump, take a look please to the packet #1884.
[client62541_lifetime_600000_9817.zip](https://github.com/open62541/open62541/files/91495/client62541_lifetime_600000_9817.zip)

Stasik0, 2016-01-15:
This is spec-conformal behavior - the server decides about the lifetime. But there is also no 3-minute drop, right?

Aleskey78, 2016-01-15:
There is a drop after ~10 minutes. (RevisedLifetime: 600000, packet #1884)
UaExpert causes no connection drops at all (I have captured communication for more then 15 minutes).

Stasik0, 2016-01-15:
@Aleskey78 okay, but the timeout is correct, could you please capture one channel renewal with UaExpert, i need to see thats the difference between the renewal messages

Stasik0, 2016-01-15:
okay, I beleive, I have found the problem: the channelId was missing while renewing, 
I have also added 75% renewal lifetime
could you test please @Aleskey78 and others?

Aleskey78, 2016-01-15:
@Stasik0, the issue is solved! ~40 minuts without drops. I see only OpenSecureChannelRequest/Respons messages each 450 seconds. What exactly corresponds to 75% from RevisedLifetime: 600000.

Stasik0, 2016-01-15:
okay, nice to hear that

closed 2016-01-15 by Stasik0