open62541/open62541 #507: Compilation for Android is broken

Aleskey78, 2016-01-14:
environment:
 - Qt 5.5
 - android-ndk-r10d
- arm-linux-androideabi-gcc.exe (GCC) 4.9 20140827 (prerelease)
compilation errors:
[errors_6fe903.txt](https://github.com/open62541/open62541/files/90739/errors_6fe903.txt)

Aleskey78, 2016-01-15:
commit b1826a4 is still ok, the next one 06774a3 brings errors:
open62541.c:3780: error: undefined reference to 'le16toh'
open62541.c:3820: error: undefined reference to 'le32toh'
open62541.c:3866: error: undefined reference to 'le64toh'

Stasik0, 2016-01-15:
Apparently the macros are named differently in ndk, could you please rename and see if it works before we patch?

http://stackoverflow.com/a/11702786

Aleskey78, 2016-01-15:
@Stasik0, thanks for the link. Now  06774a3 compiles without any errors. But I still can not compile 6fe903 or switch to the latest master, see attached log above.

Aleskey78, 2016-01-15:
I think the problem starts with a commit "improve checking for endianness" 4523086d7095f990884049f375d09d3ff7e77264.
The previous version can be easily corrected, as states in link you provided.

Stasik0, 2016-01-15:
well as the logfile states, the problem starts with the inclusion of endian.h
"endian.h:171:12: error: '__uint64_t' redeclared" the previous declaration was in "inttypes.h"
@jpfr any idea? @Aleskey78 can some Android-specialist help us?

Aleskey78, 2016-01-15:
I've managed to compile a master version with a "ua_config.h.in" file from the commit 80eded3 and renaming le16toh to letoh16, le32toh to letoh32 and le64toh to letoh64. Some kind of git revert :)

Stasik0, 2016-01-15:
okay now we need to get it working somehow... we need @jpfr 

we need a special case for android

@Aleskey78 can you try to comment out the swap* definitions? https://github.com/open62541/open62541/blob/master/include/ua_config.h.in#L97 to see if it brings some progress? 

you can also add 

#  define le16toh(x) letoh16(x)
#  define le32toh(x) letoh32(x)
#  define le64toh(x) letoh64(x)

Aleskey78, 2016-01-15:
no effect:
error: '__uint64_t' redeclared as different kind of symbol __uint64_t htole64(__uint64_t);

and so on....

Stasik0, 2016-01-15:
in this case, try to ```#undef __uint64_t``` somewhere at the end of config.h.in

also try to find out which of it - else if cases fires... does android define ```__linux__``` i'd try to add 

```
# if defined(__ANDROID__)
#  include <endian.h>
# define le16toh(x) letoh16(x)
# define le32toh(x) letoh32(x)
# define le64toh(x) letoh64(x)
//maybe #undef __uint64_t or something
# elif defined(__linux__)
#  include <endian.h>
# elif defined(__OpenBSD__)
```

Aleskey78, 2016-01-15:
i came up to this version:
```c++
...
#if defined(_WIN32)
# define htole16(x) (x)
# define htole32(x) (x)
# define htole64(x) (x)
# define le16toh(x) (x)
# define le32toh(x) (x)
# define le64toh(x) (x)
#else
# if defined(__linux__)
#  ifndef __QNX__
#   if defined(__APPLE__) || defined(__MACH__)
#    include <machine/endian.h>
#   else
#    include <endian.h>
#    define le16toh(x) letoh16(x)
#    define le32toh(x) letoh32(x)
#    define le64toh(x) letoh64(x)
#   endif
#  endif
# elif defined(__OpenBSD__)
...
```
it compiles for both w32 and andriod, but i'm not sure it is all correct.

Stasik0, 2016-01-15:
I am also not sure, since @jpfr improved it, I mean it compiles, but does it work? Can you add the whole file somewhere? gist or just a branch? We need to wait for @jpfr 

doesn't my suggestion work?
```
# if defined(__ANDROID__)
#  include <endian.h>
# define le16toh(x) letoh16(x)
# define le32toh(x) letoh32(x)
# define le64toh(x) letoh64(x)
//maybe #undef __uint64_t or something
# elif defined(__linux__)
#  include <endian.h>
# elif defined(__OpenBSD__)
.....
```

Aleskey78, 2016-01-15:
i'm out now, it is too late for today...
https://gist.github.com/Aleskey78/414d2c65971be0b9f45e

have a nice weekend!

Aleskey78, 2016-01-15:
your solution works:
```
//#if defined(__LITTLE_ENDIAN__) || defined(_WIN32) || (__BYTE_ORDER__ == __ORDER_LITTLE_ENDIAN__)
#if defined(_WIN32)
# define htole16(x) (x)
# define htole32(x) (x)
# define htole64(x) (x)
# define le16toh(x) (x)
# define le32toh(x) (x)
# define le64toh(x) (x)
#else
# if defined(__ANDROID__)
#  include <endian.h>
# define le16toh(x) letoh16(x)
# define le32toh(x) letoh32(x)
# define le64toh(x) letoh64(x)
# elif defined(__linux__)
#  include <endian.h>
# elif defined(__OpenBSD__)
```

Stasik0, 2016-01-15:
okay, i have added this, could you verify once again next week, @Aleskey78. Then we can close the issue

jpfr, 2016-01-16:
Looks good. I'm surprised that Android (Linux-based) contains BSD system header files.
I added a condition to check for endianness on such systems.
921dce5b99b8dff6cb42b6839d40d68da7e164c1
But that's only to speed up the handling of arrays, nothing functional.

Aleskey78, 2016-01-20:
sorry for the late response, but i am ill this week... so... the change didn't help.

it compiles only if I change this line https://github.com/open62541/open62541/blob/921dce5b99b8dff6cb42b6839d40d68da7e164c1/include/ua_config.h.in#L50
```c++
#if defined(__LITTLE_ENDIAN__) || defined(_WIN32) || (__BYTE_ORDER__ == __ORDER_LITTLE_ENDIAN__)
```
to this one
```c++
#if defined(__LITTLE_ENDIAN__) || defined(_WIN32) 
```

Aleskey78, 2016-01-21:
small update: it works now also without the -mno-unaligned-access compiler flag, see #469

Stasik0, 2016-01-21:
Do you get undefined errors?

Aleskey78, 2016-01-21:
some kinde of:
[image]

The problem is that ```"# if defined(__ANDROID__)"``` line would not even be reached during preprocessing because of this condition:
```#if defined(__LITTLE_ENDIAN__) || defined(_WIN32) ```**|| (__BYTE_ORDER__ == __ORDER_LITTLE_ENDIAN__)**

Stasik0, 2016-01-31:
@jpfr @Aleskey78 can we fix it for real? should we workaround "|| (BYTE_ORDER == ORDER_LITTLE_ENDIAN)" to smth like "|| (!defined(__ANDROID__) && BYTE_ORDER == ORDER_LITTLE_ENDIAN)"

>>

commit 8400be77 (2016-02-04): workaround for androud compilation due to no activity, fixes #507
closed 2016-02-04 by Stasik0