open62541/open62541 #508: Compilation of myServer.c from section "Creaing your first server" of First steps failed

Kuichen, 2016-01-14:
OS: Ubuntu 14.04

I followed the instructions of the tutorial "First steps with open62541-server" in order to build a server. 
All steps before compilation of "myServer.c" have been realized without problem. By compilation with "> gcc -Wl,-rpath,`pwd` -I ./include -L ./ ./myServer.c -o myServer  -lopen62541" error and warning appeared (pasted at bottom), which mainly suggest that too many arguments for  "UA_Server_run". I looked it up in file "ua_server.h" , it seems "UA_Server_run" are only expecting 2 arguments but in the tutorial 3 arguments are passed to it by "UA_Server_run(server, 1, &running)" (see http://open62541.org/doc/current/tutorial_server_firstSteps.html). 
Could it be the reason?  How can I fix it? Thanks.

Error msg:

In file included from ./include/ua_types.h:28:0,
                 from ./myServer.c:3:
./include/ua_config.h:6:0: warning: "_BSD_SOURCE" redefined [enabled by default]
 # define _BSD_SOURCE
 ^
In file included from /usr/include/stdio.h:27:0,
                 from ./myServer.c:1:
/usr/include/features.h:188:0: note: this is the location of the previous definition
 # define _BSD_SOURCE 1
 ^
./myServer.c: In function ‘main’:
./myServer.c:13:3: warning: passing argument 2 of ‘UA_Server_run’ makes pointer from integer without a cast [enabled by default]
   UA_Server_run(server, 1, &running);
   ^
In file included from ./myServer.c:4:0:
./include/ua_server.h:73:25: note: expected ‘volatile UA_Boolean *’ but argument is of type ‘int’
 UA_StatusCode UA_EXPORT UA_Server_run(UA_Server *server, volatile UA_Boolean *running);
                         ^
./myServer.c:13:3: error: too many arguments to function ‘UA_Server_run’
   UA_Server_run(server, 1, &running);
   ^
In file included from ./myServer.c:4:0:
./include/ua_server.h:73:25: note: declared here
 UA_StatusCode UA_EXPORT UA_Server_run(UA_Server *server, volatile UA_Boolean *running);

Stasik0, 2016-01-14:
oh no, we are currently changing the API a bit, please have a look into current examples that are compiled automatically, e.g., https://github.com/open62541/open62541/blob/master/examples/server_firstSteps.c. The examples are build automatically so it should work. Which release are you using?

Kuichen, 2016-01-14:
sorry I'm quite new to github and Linux, I can't find the release info of my installed version. I just followed the tutorial using ":> git clone https://github.com/acplt/open62541" (see http://open62541.org/doc/current/tutorial_server_firstSteps.html) and got it cloned on my machine.
I tried the example you mentioned (which also has some differences with the second example server from the tutorial First steps although it claims they are same), another error appears saying that missing file "open62541.h". But I checked its feather options using ccmake and it shows that the AMALGAMATION option is OFF, so the main file should not include "open62541.h". I have to delete the "ifdef UA_NO_AMALGAMATION"  statement to make it work. The generated server runs on the port 16664, I haven't test it with UaExpert yet.

closed 2016-01-15 by Kuichen