open62541/open62541 #509: Verify tutorials (relates #508)

Stasik0, 2016-01-14:
Before freezing 0.2 we need to make sure tutorials are aligned with current api

ichrispa, 2016-01-15:
Hmmm... I have two "volunteers" that can be assigned this task (with ETA mid february). Would that be sufficient?

Stasik0, 2016-01-15:
Yea,  but first we need to be sure that the api is stable

Pro, 2016-01-21:
See also: PR https://github.com/open62541/open62541/pull/516
I'll go through the tutorials and try to fix any errors when I see them using the current version.

Stasik0, 2016-01-21:
Can we actually include .c files in sphinx? Kinda, "include xyz.c:3-15"? Would be much better than double maintenence.

jpfr, 2016-01-22:
Totally.
Maybe we can write large parts of the documentation as comments in tutorial c-files and in the public header files. You know, take the code, detect comments in `/** */` double-star comments and wrap everything else in code blocks.

Otherwise we will run into impedence mismatches between code and docs. At least between releases.

Stasik0, 2016-01-26:
isn't there some sphinx solution?

FlorianPalm, 2016-01-27:
We would also need Doxygen, wouldn't we?  [Look here](http://stackoverflow.com/questions/11246637/using-sphinx-apidoc-to-generate-documentation-from-c-code)

Stasik0, 2016-01-27:
I thought about "including" parts of source code into tutorial articles which is, imho, not doxygen anymore.

memaikey, 2016-06-17:
First steps with open62541-client tutorial doesn't compile.
(trying to upload fix, but new to git)

memaikey, 2016-06-20:
Tutorial "Adding nodes to a server and connecting nodes to user-defined values" mentions 3 cases under  "Variants", but only lists 2.

Stasik0, 2016-06-20:
@memaikey its even worth: the listed items do not match the values in the struct: the truth is in the code: https://github.com/open62541/open62541/blob/38d4e7fbad8d11826521cecda630046ffa20035d/include/ua_types.h#L535

Stasik0, 2016-06-23:
@memaikey did you check all the tutorials?

memaikey, 2016-06-23:
@Stasik0 I finished toturial 1&2 and just made a pull request for my latest updates. I've read tutorial 3&4, but they basically say "go look at the code". I am currently testing these two. I don't think I'll check the final tutorial in a while.

schmidmaxa, 2016-06-30:
I've checked up to "Creating Object Instances".. I can't get that one to compile, but I'm sure the issue is on my end getting the correct headers. Still new to much of this, so the documentation might be a little light on that one for a newbie like myself.

markusgraube, 2016-07-14:
@Stasik0 Sphinx support including parts of the source files:

```
.. literalinclude:: ../../examples/server_firstSteps.c
   :language: c
   :lines: 1,3,5-10,20-
```

However, current files in examples are not the same as in the tutorials.

jpfr, 2017-04-24:
Tutorials were refactored and are now included in example .c files.

closed 2017-04-24 by jpfr