open62541/open62541 #510: Memory leak.

Wtango, 2016-01-15:
It seems to be a memory leak problem in UA_Array_copy.That what valgrind says:
```
==14235== 
==14235== HEAP SUMMARY:
==14235==     in use at exit: 28 bytes in 1 blocks
==14235==   total heap usage: 5,579 allocs, 5,578 frees, 92,478,473 bytes allocated
==14235== 
==14235== 28 bytes in 1 blocks are definitely lost in loss record 1 of 1
==14235==    at 0x4C2CC70: calloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==14235==    by 0x4E4CD12: UA_Array_copy (ua_types.c:814)
==14235==    by 0x4E4C7D7: copyNoInit (ua_types.c:709)
==14235==    by 0x4E4C75B: copyNoInit (ua_types.c:699)
==14235==    by 0x4E4C8F6: UA_copy (ua_types.c:724)
==14235==    by 0x4E640C2: UA_EndpointDescription_copy (ua_types_generated.h:440)
==14235==    by 0x4E645BD: Service_GetEndpoints (ua_services_discovery.c:97)
==14235==    by 0x4E5F90B: processMSG (ua_server_binary.c:427)
==14235==    by 0x4E5FD37: UA_Server_processBinaryMessage (ua_server_binary.c:483)
==14235==    by 0x4E61C83: processJobs (ua_server_worker.c:55)
==14235==    by 0x4E62A50: UA_Server_run_iterate (ua_server_worker.c:603)
==14235==    by 0x4E62C25: UA_Server_run (ua_server_worker.c:653)
==14235== 
==14235== LEAK SUMMARY:
==14235==    definitely lost: 28 bytes in 1 blocks
==14235==    indirectly lost: 0 bytes in 0 blocks
==14235==      possibly lost: 0 bytes in 0 blocks
==14235==    still reachable: 0 bytes in 0 blocks
==14235==         suppressed: 0 bytes in 0 blocks
==14235== 
==14235== For counts of detected and suppressed errors, rerun with: -v
==14235== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
```
another case :
```
==14136== 962 (432 direct, 530 indirect) bytes in 1 blocks are definitely lost in loss record 692 of 696
==14136==    at 0x4C2CC70: calloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==14136==    by 0x4E4175C: UA_Array_copy (open62541.c:3724)
==14136==    by 0x4E40E10: UA_Variant_setArrayCopy (open62541.c:3461)
==14136==    by 0x4F4920E: ua_namespaceinit_generated (open62541.c:38796)
==14136==    by 0x4E4693F: UA_Server_new (open62541.c:7263)
==14136==    by 0x401E0B: main (server.c:221)
==14136== 
==14136== 2,882 (1,104 direct, 1,778 indirect) bytes in 1 blocks are definitely lost in loss record 695 of 696
==14136==    at 0x4C2CC70: calloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==14136==    by 0x4E4175C: UA_Array_copy (open62541.c:3724)
==14136==    by 0x4E40E10: UA_Variant_setArrayCopy (open62541.c:3461)
==14136==    by 0x4F572FE: ua_namespaceinit_generated (open62541.c:40771)
==14136==    by 0x4E4693F: UA_Server_new (open62541.c:7263)
==14136==    by 0x401E0B: main (server.c:221)
==14136== 
==14136== 3,957 (1,536 direct, 2,421 indirect) bytes in 1 blocks are definitely lost in loss record 696 of 696
==14136==    at 0x4C2CC70: calloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==14136==    by 0x4E4175C: UA_Array_copy (open62541.c:3724)
==14136==    by 0x4E40E10: UA_Variant_setArrayCopy (open62541.c:3461)
==14136==    by 0x4F52AB9: ua_namespaceinit_generated (open62541.c:40077)
==14136==    by 0x4E4693F: UA_Server_new (open62541.c:7263)
==14136==    by 0x401E0B: main (server.c:221)
==14136== 
==14136== LEAK SUMMARY:
==14136==    definitely lost: 20,355 bytes in 523 blocks
==14136==    indirectly lost: 11,948 bytes in 312 blocks
==14136==      possibly lost: 0 bytes in 0 blocks
==14136==    still reachable: 0 bytes in 0 blocks
==14136==         suppressed: 0 bytes in 0 blocks
==14136== 
==14136== For counts of detected and suppressed errors, rerun with: -v
==14136== ERROR SUMMARY: 523 errors from 523 contexts (suppressed: 0 from 0)
```

Stasik0, 2016-01-15:
Copy allocates memory at dst that seems not to be freed. @jpfr, is it a regression?

Stasik0, 2016-01-15:
@Wtango I fixed the leak in the getEndpoints, no time to look in the second one atm

Stasik0, 2016-01-15:
@ichrispa there is a leak with array values i cannot resolve, could you please help me? btw, could you also look into the previous commit?

jpfr, 2016-01-18:
The new commit c54348ab9ddbb880ef7746fa5dc15cc9b58ce74f is relevant to this discussion.
Now, only the nodestore can create / delete / copy UA_Node and friends.
That is, the nodestore creates a struct with the node at the end and returns a pointer inside the struct.

That way, we can ensure that nodes are replaced only with an edited copy of the current node version.

Also, the bug where @ichrispa had to make an intermediate copy should no longer apply.
- Without multithreading, nodes are deleted only after an explicit call to UA_NodeStore_replace or UA_NodeStore_delete.
- With multithreading, nodes are deleted only after an explicit call to UA_NodeStore_replace or UA_NodeStore_delete **and** only after every worker thread has completed his current job. The delayed deletion is done via the RCU kernel mechanism.

However, if you are single-threaded and for example change the references-array, you will have to re-load the reference!!

Wtango, 2016-01-18:
@jpfr After c54348a the namespace compiler is broken,it does not fit the code currently.

jpfr, 2016-01-18:
I'm on it.
Still no unit tests.
Otherwise the commit would not have happened without checking this.

jpfr, 2016-01-18:
Hey @Wtango can you be a bit more specific about the issue with c54348a?
The server_nodeset example works just fine on my box.

Wtango, 2016-01-18:
I'm in bed.
You can build the project with UA_ENABLE_GENERATENAMESPACE0,will see the error.

https://github.com/open62541/open62541/issues/510#issuecomment-172571883

Wtango, 2016-01-18:
BTW,the memory leak problem also happens in the generated code.

https://github.com/open62541/open62541/issues/510#issuecomment-172571883

jpfr, 2016-01-18:
Generation is back in 4d553b9c1fc40561cb3e74db02e96f1f1a1ca168.
Nearly all memleaks in the generated ns0 are gone in fa20920d3fb07545ed2b919fc3bd25d7fb3b5c7b.

@ichrispa There are still some UA_Byte* malloced but never freed in the generator. But that is the scary binary encoding part. Can u check?

jpfr, 2016-03-05:
This was good last time I checked.
Please reopen if this is still an issue.

closed 2016-03-05 by jpfr