open62541/open62541 #511: Publish request to "automatic" [issue from mailinglist]

Stasik0, 2016-01-15:
We got the following:

Hello, 

I compiled and implemented the server and client from the tutorials and almost everythings works fine.

1. However when I change the value by writing a new Value, the value gets changed but when I call out UA_Client_Subscriptions_manuallySendPublishRequest(), the handler ("The Answer has changed") isn't call out. 

   What could be the reason for this ?
   

2. Also I realized that in order to find out whether the value on my MonitoredItem  has changed, I continuosly need to call UA_Client_Subscriptions_manuallySendPublishRequest().

Is there a way to set  the subscription, so that it automatically calls the PublishRequest whenever the value changes?

@ichrispa could you please look into it, do you monitor the mailing list?

ichrispa, 2016-01-15:
Hi @Stasik0,

we are getting a lot of tutorial related stuff lately... please forward to the mailing list:

(1) Please provide a test case client/server and the version of the stack to test against.
(2) No; there where attempts by @FlorianPalm was working on a clientside job/multithreading case? It can be attempted to actually multithread this by hand, but I last used that 9 months ago and I have no clue if that still works... it did back then.

Best regards,

ichrispa

NetoLe, 2016-01-18:
for no 2. If I get this right you mean kind of a push-notification? When I read over the spec I think this is not meant to be implemented.
I looked up in the spec and it says: "c) NotificationMessages are sent to the Client in response to Publish requests." (Part 4 p.69)
and "d) At the beginning of a cycle, if there are Notifications to send but there are no Publish requests queued, the Server enters a wait state for a Publish request to be received."
So there is always a need fo a request-call from the client side, where you can't know if the value changed. 
Especially: "requestHandle IntegerId The requestHandle given by the Client to the request."
which is located in the responseHeader, so the client needs to call the request. 

Or did I get this wrong and you mean that you send a request and the response will be hold until a real value changed?

jpfr, 2016-01-18:
> Or did I get this wrong and you mean that you send a request and the response will be hold until a real value changed?

Something like that.
There is an internal timer that samples values with a recurring interval -> MonitoredItem
A second timer sends out all the samples received until then for several variables -> Subscription

OPC UA strictly adheres to the request/response pattern. But responses can be sent asynchronously and do not need to respect the ordering of the requests. Therefore, we can just "freeze" publish requests for the subscription and reply only when the timer runs out.

ichrispa, 2016-01-20:
I understand the question to be more of the sort "is there a background task that can handle the publish requests in periodic intervals?". Sort of a clientside UA_Job. Since there is no main run loop for clients, the answer is a straightforward "no, unless you implement such a loop". In my eyes it would not make sense to implement a main loop for the client, since it drives the interaction with the datasources with a very application-specific context.

HolgerJeromin, 2016-01-20:
`UA_Client_manuallyRenewSecureChannel` is called cyclic from OV client. Perhaps we should have a better named function which ensure the securechannel and can do other stuff as well.

ichrispa, 2016-01-20:
@HolgerJeromin: Isn't that function only called when sendBinary is invoked? If I leave my client idling in a loop for a minute, it will still time out... 

Can we generalize the mechanism to some sort of UA_Client_Job that will become available if multithreading is enabled?

HolgerJeromin, 2016-01-20:
I am not 100% sure if this works as intented. OV does not need/use a multithreading code

Stasik0, 2016-01-21:
You are correct, long idling can cause channel timeout

jpfr, 2016-01-21:
@ichrispa 
Yes. Ideally, we can refactor ua_server_worker.c into a general work dispatch mechanism.
Client and server would have nearly no custom dispatch code.

Imho the client will definitely need either a separate thread or the possibility to plug into an existing main-loop.
- Keep the SecureChannel alive without user interaction
- Message responses are async. For fast reaction times we better have a separate thread listening. And we need to demux subscription responses from regular requests.

ichrispa, 2016-04-15:
I am handling this sort of thing by simply threading the client for over a year now, which works great. A state machine does periodic stuff (channel renewal, subs), mutexes are used to yield client usage back to userspace. I do however consider this an userspace implementation aspect, not a stack functionlity.

Should we embed this functionality as default /w multithreading enabled?

ichrispa, 2016-04-15:
edit: better idea: include this as an example client!? @Stasik0 what do you thing?

Stasik0, 2016-04-15:
@ichrispa why dont we just use the scheduler from the server? which seems to be kinda the same things and can be threaded out for the periodic stuff?

jpfr, 2016-04-19:
@ichrispa It is a good idea to have an *outside main loop* such as you describe it.
But this will become quite involved as more and more things need to be handled manually that could be done automatic.
- Keep the securechannel open
- Keep the session open
- Reconnect if the connection was lost (before the session times out)
- Feed publish requests to the server
- Do internal cleanups, e.g. timed-out async callbacks

This is all post 0.2, imho.
But we could have a single `UA_Client_run_iterate` that is called periodically (or after a returned timeout duration) instead of `UA_Client_manuallyRenewSecureChannel` and `UA_Client_Subscriptions_manuallySendPublishRequest`.

With this facade, we also have more freedom to adjust the internal API due to the growing number of features. And the public API doesn't change since everything is trigger within `UA_Client_run_iterate`.

HolgerJeromin, 2016-04-19:
`UA_Client_run_iterate` would fit nicely in the OV context.

ichrispa, 2016-04-19:
There is a reason I do not want the client to be inside the stack and use threading: Since jpfr introduced userspace rcu's, it is impossible to compile the stack on windows when multithreading is enabled. I considered refactoring liburcu out of the stack, but that would be rude. I do however need this feature to run under windows - currently, using userpace coordination, it does.

jpfr, 2016-04-19:
I was **not** proposing to give the client its own thread.

Concurrency != parallelism

It's the same with the server. We have this worker loop and everything. But we can just run single-threaded where you call `UA_Server_run_iterate` cyclically from you own mainloop in userland.

With regards to liburcu. Yes, let's get rid of that dependency!
Once @FlorianPalm has a concept for external nodestores, we can demote nodestore_concurrent to a plugin.

In general, there are lots of parts in the stack that access shared memory more or less unprotected. I made sure to use atomic compare-and-switches to protect from races, but its all very hand-wavy with no formal verification or testing of significant coverage.
I'm inclined to give the multithreading option an *experimental* flag for the next release.
It works. But we haven't done enough "engineering" yet to recommend it for productive use.

uleon, 2016-04-19:
:+1: for getting rid of liburcu as this unfortunately restricts multithreading to linux. Personally I believe it's a pitty that we neglected multithreading with vanilla posix-threads. In my opinion this used to be an important feature in the early days of the stack development...

ichrispa, 2016-04-20:
@jpfr: Just to make sure: We are talking about an internal loop akin to this one
https://github.com/plt-tud/PLT_MRT_ARM-RPi2/blob/master/UA_TrafficLight/src/ua_mapped_class/ua_remoteobject.cpp#L253
But not the threading. We would essentially call manuallySendPublish in the IDLE state and maybe hook UA_Jobs in there?

ichrispa, 2016-04-21:
After tampering about with clientside jobs, I ended up with an idea... claiming the client_iterate and client_job issue.

If anyone else is attempting to detach jobs from the serverside or creating a mainloop'able client, feedback would be appreciated.

jpfr, 2016-04-24:
Hey @ichrispa,
cool. Go for it!

PR #571 is also relevant to this imho.
In a nutshell, in standard-compliant subscriptions, the server can decide to answer a publish request at a later time. That makes our (synchronous) client block because he waits for an publish response that may come minutes later.

That's kinda related to the clientside jobs. We could have async requests sending that doesn't wait for a response. Instead we register a callback that is triggered when the response finally arrives. But somebody must be listening on the socket from time to time.

I'm at Hannover Fair for a couple of days. But maybe we can find some minutes to talk...

jpfr, 2016-07-21:
Food for realizing a multi-threaded dispatch queue for event processing:
http://chaoran.me/assets/pdf/wfq-ppopp16.pdf

StalderT, 2018-02-18:
Can be closed (#1497) ?

Pro, 2018-03-22:
Closed, because it was implemented in #1497 
Thanks @StalderT!

closed 2018-03-22 by Pro