open62541/open62541 #512: "UA_NO_AMALGAMATION" is always undeined

Kuichen, 2016-01-16:
By compilation of "server_firstSteps.c" and "client_firstSteps.c" from open62541/examples, "UA_NO_AMALGAMATION" is always undefined even the AMALGAMATION feather is OFF, which leads to error: missing  "open62541.h".

Stasik0, 2016-01-16:
Do you compile manually? If you compile with amalgamation, the client_firstSteps.c and open62541.c, open62541.h have to be in the same dir:
```gcc -std=c99 server_firstSteps.c open62541.c -o server```

Otherwise, I'd suggest using cmake to generate makefiles and use make, the cmake will do it for you.
as you see from
```VERBOSE=1 make```
the compiler call is
```/usr/bin/cc  -DBENCHMARK -DUA_BUILD_UNIT_TESTS -DUA_NO_AMALGAMATION ..... and so on```

The stuff is woking, as you can see from our build server e.g. here: https://travis-ci.org/open62541/open62541/builds/102761169#L956

Kuichen, 2016-01-16:
Yes, I compile manually. They work fine with static library open62541.c, I have trouble when compiling with dynamic library libopen62541.so, which was generated from cmake when AMALGAMATION option is OFF. Isn't here with following statements from the client_firstSteps.c:

#ifdef UA_NO_AMALGAMATION 
# include "ua_types.h"
# include "ua_server.h"
# include "logger_stdout.h"
# include "networklayer_tcp.h"
#else
# include "open62541.h"
#endif

"open62541.h" should not be included but those other 3 header files?

Stasik0, 2016-01-16:
well, as a rule of thumb, I'd advice any end-user to use open62541.h and not the separate files
* option 1 (nice): ```gcc -std=c99 server_firstSteps.c libopen62541.so -o server``` - you need open62541.h
if the absence of the header is a problem, we provide precompiled headers for last 50 builds: http://open62541.org/releases/
* option 2 (hard): if you really want to fiddle with single headers, you can define UA_NO_AMALGAMATION manually ```gcc -std=c99 -DUA_NO_AMALGAMATION server_firstSteps.c libopen62541.so -o server``` (you will probably need to set include path with -I accordingly, as a tip: ```-I/home/user/open62541/src_generated -I/home/user/open62541/include -I/home/user/open62541/src -I/home/user/open62541/src_extra -I/home/user/open62541/deps```)

Kuichen, 2016-01-17:
Alles klar! Thanks a lot for the explanations.

closed 2016-01-17 by Kuichen