open62541/open62541 #514: nodes.h not included in open62541.h but in .c by amalgamation

NetoLe, 2016-01-19:
I got a problem generating the nodes for "OPC UA for Devices" out of the given XML-File: https://opcfoundation.org/UADevices/1.1/Opc.Ua.Di.NodeSet2.xml
I used the generate_open62541CCode.py-script which generates this in the header-file:
```C
#ifdef UA_NO_AMALGAMATION
...
#else
#include "openOPC\open62541.h"
#define NULL ((void *)0)
#endif
```

My compiler then throws errors like 
`
src/nodesetfordevices.c:51:1: error: 'UA_ObjectNode' undeclared (first use in this function)
src/nodesetfordevices.c:51:16: error: 'opcua_node_object_t_ns_1_i_5001' undeclared (first use in this function)
`
and so on for every Node and reference etc.
But I found these Structures in the .c but not in the referenced .h-file
Maybe this needs to be changed somehow.

jpfr, 2016-01-19:
Hey @NetoLe,

the user should not come into contact with ua_nodes.h. It is not part of the public API. Save access is provided via the UA services.

The nodeset generator however uses it to allow "bootstrapping" of the namespace 0.
So far, @ichrispa insisted not to assume an existing standard-ns0 from ua_server.c in the nodeset generator. That's why the generated code relies on "internal" APIs.

It would be best if you build against the normal source tree and not with the single-file release.
Or, you copy the ua_nodes.h out of the /src/server/ directory and put it next to open62541.h.
But that is a quite hackish solution...

ichrispa, 2016-01-20:
Hi @NetoLe, Hi @jpfr,

I do not recommend building monolithic namespaces for applications. The namespace generator will assume that no nodes exist and attempt to bootstrap some nodes, which results in exactly this problem.

Instead, I would suggest that you build an "add-on" userspace namespace. To do that, generate an XML file that includes only your nodes, not namespace 0. This is the standard behavior for tools like ua modeller. Then run the namespace compiler and suppress the creation of any namespace 0 node like this:

```
python ./generate_open62541CCode.py -i  NodeID_Blacklist_FullNS0.txt your.nodeset.xml my_namespace
```

This will generate a my_namespace.c and my_namespace.h file; Since NS0 exists, no bootstrapping should take place. You can safely remove the ua_nodes.h or any other internal header from the generated includes. Calling the generated function will load your namespace. 

Best regards,

ichrispa

NetoLe, 2016-01-21:
ok thanks, so I tried it with your command but it seems, that this does not work as expected.
in every generated node I find this line: "// Parent node does not exist yet. This node will be bootstrapped and linked later." I used the script in windows, that's my exact cmd-line: 
`D:\GitHub\open62541\tools\pyUANamespace>generate_open62541CCode.py -i NodeID_Blacklist_FullNS0.txt Opc.Ua.Di.NodeSet2.xml nodeset1`

Anyways I did this with generating the nodeset xml of ns0 + the nodeset xml of the "OPC UA for Devices"-spec. this did work and I just deleted all the irrelevant stuff that had to do with generated ns0. Then I implemented the generated and corrected function at the end of the amalgamated open62541.c-file. UAExpert proves that it works. But two more things: the script does not look for "isAbstract". I needed to implement those parts for myself. And just curious: could it be that there is no kind of InstanceDeclaration or use of ModellingRules in the Server at all?

Stasik0, 2016-01-21:
Colud you show us these code improvements?

NetoLe, 2016-01-22:
I think "code improvements" is a bit too much ;)

but yeah of course. So I generated the file with 
`D:\GitHub\open62541\tools\pyUANamespace>generate_open62541CCode.py Opc.Ua.NodeSet2.xml Opc.Ua.Di.NodeSet2.xml nodeset`
at first this caused a 2,6 MB big file with 36425 loc.
Both nodesets are directly from the OPC UA specs. The second one is from "OPC UA for Devices".
The Namespace I was interested was the namespace 1. so namespace 0-nodes were deleted. But now I had the right references. 
After deleting all the irrelevant stuff, also the generated mandatoryPlaceholders etc. I had what I needed. 
This is the header: I needed to uncomment some defines because they were redefined, which is caused by the nodeset.xml of the spec. 

[nodesetfordevices.h.txt](https://github.com/open62541/open62541/files/100481/nodesetfordevices.h.txt)

and this is the c-stuff I added at the bottom of the open62541.c. Note: I deleted all the madatory-modellingrules-references since it seems this has no effect. 

[nodesetfordevices.c.txt](https://github.com/open62541/open62541/files/100485/nodesetfordevices.c.txt)

jpfr, 2016-01-28:
> could it be that there is no kind of InstanceDeclaration or use of ModellingRules in the Server at all?

Jup. We have not implemented ModellingRules or any sophisticated consistency checking so far. But hey, we are working towards a 0.2 release. We will get to it eventually.

@ichrispa 
Are there plans to integrate the improvements described by @NetoLe into the mainline code?

closed 2016-06-06 by jpfr