open62541/open62541 #517: open62541-server does not close session with UAExpert.

NetoLe, 2016-01-22:
Ok maybe the title is not totally correct because I can't locate the problem for that.

What happens:
If I want to connect to an open62541-Server with UaExpert 1.4.1 it takes 2 min till the connection is established (while that time UaExpert just freezes). once this happened I can close and open my server as often as I want, it does immediately reconnect. If I want to disconnect it takes 2 min too!
This problem didn't occure with UaExpert 1.3.1, where it did immediately connect to the server and disconnect.
So I took a wireshark trace and saw that it took exactly 2 min till the connection worked. 

[uaexpert2minprob.zip](https://github.com/open62541/open62541/files/100763/uaexpert2minprob.zip)
(192.168.60.111 is client, ....60.8 is Server)
So I looked in the UaExpert configuration for a number with "120" in it. I found the session timeout which had 1200000,00 as value.
But changing this to a lower number didn't change the problem. it are still exactly 2 minutes and I can't locate which member is the problem here. Is this an UaExpert-problem?

jpfr, 2016-01-25:
Hey @NetoLe,

I can't reproduce the behavior with UAExpoer 1.4.1 on my box.
The messages on wireshark look clean too.
However there are some artifacts on the TCP level. Reordered packets, duplicate ACK messages, and so on.

Is this behavior reproducible?

NetoLe, 2016-01-25:
Hey @jpfr 
so I am using the open62541-Server on an embedded device and PC is linked over normal ethernet with it, there it is always reproducible.
I tried to run a PC-Application, so it's just localhost to localhost. What works is the Port 16664 which is given with the tutorial without any problems. What doesn't work is the Port 4840, which I am using on the embedded device too. The server immediately crashes with "Error during socket binding".
At that point UaExpert is not connected to the embedded device anymore.
So I looked up in cmd with "netstat -o" but the port isn't used at all. 
that is really weird. any ideas what I could try?

NetoLe, 2016-01-25:
oh, sorry, I should've used "netstat -ao"

There is a Service using this Port opcualds.exe "UA Local Discovery Server Windows Service".
Ok, what is this?

NetoLe, 2016-01-27:
so I installed 1.3.1 again and made a capture. It seems like the problem is caused by UaExpert itself?
because with 1.3.1 there is an immediately SYN, even before the ACK for the FIN+ACK comes. 
I guess UaExpert 1.4.1 expects a different answer for its FIN/ACK than an ACK or now waits at all for a correct answer and this strange behaviour is caused by the embedded device's TCP/IP-controller. 
what do you think?

[uaexpert131.zip](https://github.com/open62541/open62541/files/106634/uaexpert131.zip)

Stasik0, 2016-01-27:
Maybe you can ask ua support explicitely? The have a uaexpert support forum.

Stasik0, 2016-02-09:
closed due expiration @NetoLe: feel free to reopen

closed 2016-02-09 by Stasik0