open62541/open62541 #521: How to browse recursively in open62541?

PriyaVeryx, 2016-01-26:
I am trying to browse nodes from root directory in a recursive manner. But at one point it loops because parent node contains child in its array and child node contains parent in its array.

Could u please help me?can we recursively browse nodes?if so could u please attach a sample?

jpfr, 2016-01-26:
Hello @PriyaVeryx,

OPC UA has hierarchical and non-hierarchical references. Hierarchical references shall not form directed cycles.

There is no way to request a recursive browse in a single request.
Just take the result of the first browse to "dig deeper" with subsequent browse requests. 

I suggest you
- Connect to an UA Server with a GUI-Client, such as UAExpert
- Take a look at the reference hierarchy at /Types/ReferenceTypes/References
- Read the description of the "Browse" Service in Part 4 of the OPC UA standard. The download is free on the website of the OPC Foundation

billran, 2016-01-29:
I have some examples of recursive browsing both in C and in java on the jOpen62541 project.  For C, see dumpBrowseNode() method:
https://github.com/billran/jOpen62541/blob/master/c/CallDll/src/CallDll.c

There is a java example in the "java" folder that uses the Bridj library.

Stasik0, 2016-01-31:
@bilach wow! jOpen62541 is impressive, what are your plans with that?

billran, 2016-02-01:
Thank you.  I had started working with CERN guy, Brice Copy but I have not heard recently from him.  They wanted a 64 bit Java version and I only did 32 bits.  Last I heard he had not solved all the issues.  When you have an official release with subscriptions I will update the library.

ichrispa, 2016-02-02:
Hi @bilach ,

for recursive browsing I would recommend calling UA_[Server|Client]_forEachChildNodeDo recursively. The callback will give you sufficient filter capabilities to screen out undesired reference types.

Beste regards,

ichrispa

billran, 2016-02-02:
Great thanks.  I will try to get it to work in Java.

Stasik0, 2016-02-03:
can I close? the isssue, @PriyaVeryx

closed 2016-02-09 by Stasik0

akshirasagar, 2016-02-29:
@PriyaVeryx : One approach we used which seems to work so far with out setup is as follows:
1) Call UA_Browse_client()
2) In the referenceDescription section of the result, if nodeClass == Variable, this is normally a leaf node where you can stop browsing.
3) If nodeClass == object, usually these are folders and you need to browse them further.