open62541/open62541 #526: networklayer_tcp.c problem

StalderT, 2016-01-30:
Hello,

When I try to compile under cygwin I have the error : 

```
`[  3%] Building C object CMakeFiles/open62541-object.dir/src_extra/networklayer_tcp.c.o
/home/btc/open62541-master/src_extra/networklayer_tcp.c:43:0: error: "fd_set" redefined [-Werror]
 #  define fd_set(fd, fds) FD_SET((unsigned int)fd, fds)
 ^
In file included from /home/btc/open62541-master/src_extra/networklayer_tcp.c:21:0:
/usr/include/sys/select.h:58:0: note: this is the location of the previous definition
 #define fd_set _types_fd_set
 ^
/home/btc/open62541-master/src_extra/networklayer_tcp.c:238:40: error: unknown type name 'fd_set'
 setFDSet(ServerNetworkLayerTCP *layer, fd_set *fdset) {
                                        ^
/home/btc/open62541-master/src_extra/networklayer_tcp.c: In function 'ServerNetworkLayerTCP_getJobs':
/home/btc/open62541-master/src_extra/networklayer_tcp.c:363:5: error: unknown type name 'fd_set'
     fd_set fdset;
     ^
/home/btc/open62541-master/src_extra/networklayer_tcp.c:364:5: error: implicit declaration of function 'setFDSet' [-Werror=implicit-function-declaration]
     UA_Int32 highestfd = setFDSet(layer, &fdset);
     ^
/home/btc/open62541-master/src_extra/networklayer_tcp.c:367:38: error: passing argument 2 of 'select' from incompatible pointer type [-Werror]
     resultsize = select(highestfd+1, &fdset, NULL, NULL, &tmptv);
                                      ^
In file included from /home/btc/open62541-master/src_extra/networklayer_tcp.c:21:0:
/usr/include/sys/select.h:74:5: note: expected 'struct _types_fd_set *' but argument is of type 'int *'
 int select __P ((int __n, fd_set *__readfds, fd_set *__writefds,
     ^
/home/btc/open62541-master/src_extra/networklayer_tcp.c:44:29: error: request for member 'fds_bits' in something not a structure or union
 #  define fd_isset(fd, fds) FD_ISSET((unsigned int)fd, fds)
                             ^
/home/btc/open62541-master/src_extra/networklayer_tcp.c:374:8: note: in expansion of macro 'fd_isset'
     if(fd_isset(layer->serversockfd, &fdset)) {
        ^
/home/btc/open62541-master/src_extra/networklayer_tcp.c:44:29: error: request for member 'fds_bits' in something not a structure or union
 #  define fd_isset(fd, fds) FD_ISSET((unsigned int)fd, fds)
                             ^
/home/btc/open62541-master/src_extra/networklayer_tcp.c:398:13: note: in expansion of macro 'fd_isset'
         if(!fd_isset(layer->mappings[i].sockfd, &fdset))
             ^
cc1: all warnings being treated as errors
CMakeFiles/open62541-object.dir/build.make:594: recipe for target 'CMakeFiles/open62541-object.dir/src_extra/networklayer_tcp.c.o' failed
make[2]: *** [CMakeFiles/open62541-object.dir/src_extra/networklayer_tcp.c.o] Error 1
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/open62541-object.dir/all' failed
make[1]: *** [CMakeFiles/open62541-object.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

```

fd_set is already define in /usr/include/sys/select.h

I have try to rename all : fd_set to UA_fd_set but after I have the error :

```
/home/btc/open62541-master/src_extra/networklayer_tcp.c:238:40: error: unknown type name 'UA_fd_set'
 setFDSet(ServerNetworkLayerTCP *layer, UA_fd_set *fdset) {
                                        ^
/home/btc/open62541-master/src_extra/networklayer_tcp.c: In function 'ServerNetworkLayerTCP_getJobs':
/home/btc/open62541-master/src_extra/networklayer_tcp.c:363:5: error: unknown type name 'UA_fd_set'
     UA_fd_set fdset;
     ^
/home/btc/open62541-master/src_extra/networklayer_tcp.c:364:5: error: implicit declaration of function 'setFDSet' [-Werror=implicit-function-declaration]
     UA_Int32 highestfd = setFDSet(layer, &fdset);
     ^
/home/btc/open62541-master/src_extra/networklayer_tcp.c:367:38: error: passing argument 2 of 'select' from incompatible pointer type [-Werror]
     resultsize = select(highestfd+1, &fdset, NULL, NULL, &tmptv);
                                      ^
In file included from /home/btc/open62541-master/src_extra/networklayer_tcp.c:21:0:
/usr/include/sys/select.h:74:5: note: expected 'struct _types_fd_set *' but argument is of type 'int *'
 int select __P ((int __n, fd_set *__readfds, fd_set *__writefds,
     ^
/home/btc/open62541-master/src_extra/networklayer_tcp.c:44:32: error: request for member 'fds_bits' in something not a structure or union
 #  define UA_fd_isset(fd, fds) FD_ISSET((unsigned int)fd, fds)
                                ^
/home/btc/open62541-master/src_extra/networklayer_tcp.c:374:8: note: in expansion of macro 'UA_fd_isset'
     if(UA_fd_isset(layer->serversockfd, &fdset)) {
        ^
/home/btc/open62541-master/src_extra/networklayer_tcp.c:44:32: error: request for member 'fds_bits' in something not a structure or union
 #  define UA_fd_isset(fd, fds) FD_ISSET((unsigned int)fd, fds)
                                ^
/home/btc/open62541-master/src_extra/networklayer_tcp.c:398:13: note: in expansion of macro 'UA_fd_isset'
         if(!UA_fd_isset(layer->mappings[i].sockfd, &fdset))
             ^
cc1: all warnings being treated as errors
CMakeFiles/open62541-object.dir/build.make:594: recipe for target 'CMakeFiles/open62541-object.dir/src_extra/networklayer_tcp.c.o' failed
make[2]: *** [CMakeFiles/open62541-object.dir/src_extra/networklayer_tcp.c.o] Error 1
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/open62541-object.dir/all' failed
make[1]: *** [CMakeFiles/open62541-object.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```

I have rename UA_fd_set to fd_set at line 238 and 363.

Now, compilation works great!

You can see the patch here : https://github.com/open62541/open62541/commit/6086d72fc103622a801a2ee0da3c952c589ec3de

Stasik0, 2016-01-31:
@StalderT hey, I try to bootstrap cygwin on our build server, can you tell me how you install python-lxml in the cygwin environment?

Stasik0, 2016-01-31:
@StalderT I am a bit puzzeled, where is the commit you linked?

StalderT, 2016-01-31:
@Stasik0 
[image]

I have made a PR : https://github.com/open62541/open62541/pull/527

Stasik0, 2016-01-31:
@StalderT okay, this is funny, it did not work out... I am running in other problems now (some c99 support) are you on IRC atm?
you can trace my progress here https://ci.appveyor.com/project/Stasik0/open62541/history branch appveyor_cygwin

StalderT, 2016-01-31:
No sorry I'm not on IRC

Stasik0, 2016-01-31:
Skype?

StalderT, 2016-01-31:
No  :-(

I have send to you my email by email

closed 2016-01-31 by StalderT

Stasik0, 2016-02-01:
okay, i added cygwin and mingw to CI so we dont miss such issues in the future 2efc343c1a877b5c7e130077faf18303856fa6df

StalderT, 2016-02-01:
Great Job !

Thanks for all