open62541/open62541 #533: Invalid read of size 4 on server

StalderT, 2016-02-02:
Hello,

When I start the server (in example directory) and the client (in example directory), I have this warning under valgrind (in server) : 

```
[02/02/2016 16:05:09.992] info/network	New Connection 5 over TCP from 127.0.0.1:16664
==3726== Invalid read of size 4
==3726==    at 0x403A814: NodeId_calcSizeBinary (open62541.c:4879)
==3726==    by 0x403ABF0: ExtensionObject_calcSizeBinary (open62541.c:4936)
==3726==    by 0x403A6FA: UA_calcSizeBinary (open62541.c:5073)
==3726==    by 0x4050E73: MonitoredItem_QueuePushDataValue (open62541.c:15191)
==3726==    by 0x40511F1: Service_Publish (open62541.c:14554)
==3726==    by 0x4051F26: processMSG (open62541.c:8326)
==3726==    by 0x40522B7: UA_Server_processBinaryMessage (open62541.c:8392)
==3726==    by 0x40525E2: processJobs (open62541.c:8642)
==3726==    by 0x40526FC: UA_Server_run_iterate (open62541.c:9198)
==3726==    by 0x40528D8: UA_Server_run (open62541.c:9241)
==3726==    by 0x804A76F: main (server.c:431)
==3726==  Address 0x43efad0 is 0 bytes after a block of size 64 alloc'd
==3726==    at 0x402A11C: malloc (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==3726==    by 0x4050D89: MonitoredItem_QueuePushDataValue (open62541.c:15152)
==3726==    by 0x40511F1: Service_Publish (open62541.c:14554)
==3726==    by 0x4051F26: processMSG (open62541.c:8326)
==3726==    by 0x40522B7: UA_Server_processBinaryMessage (open62541.c:8392)
==3726==    by 0x40525E2: processJobs (open62541.c:8642)
==3726==    by 0x40526FC: UA_Server_run_iterate (open62541.c:9198)
==3726==    by 0x40528D8: UA_Server_run (open62541.c:9241)
==3726==    by 0x804A76F: main (server.c:431)
==3726== 
==3726== Invalid read of size 4
==3726==    at 0x403A84C: NodeId_calcSizeBinary (open62541.c:4881)
==3726==    by 0x403ABF0: ExtensionObject_calcSizeBinary (open62541.c:4936)
==3726==    by 0x403A6FA: UA_calcSizeBinary (open62541.c:5073)
==3726==    by 0x4050E73: MonitoredItem_QueuePushDataValue (open62541.c:15191)
==3726==    by 0x40511F1: Service_Publish (open62541.c:14554)
==3726==    by 0x4051F26: processMSG (open62541.c:8326)
==3726==    by 0x40522B7: UA_Server_processBinaryMessage (open62541.c:8392)
==3726==    by 0x40525E2: processJobs (open62541.c:8642)
==3726==    by 0x40526FC: UA_Server_run_iterate (open62541.c:9198)
==3726==    by 0x40528D8: UA_Server_run (open62541.c:9241)
==3726==    by 0x804A76F: main (server.c:431)
==3726==  Address 0x43efad4 is 4 bytes after a block of size 64 alloc'd
==3726==    at 0x402A11C: malloc (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==3726==    by 0x4050D89: MonitoredItem_QueuePushDataValue (open62541.c:15152)
==3726==    by 0x40511F1: Service_Publish (open62541.c:14554)
==3726==    by 0x4051F26: processMSG (open62541.c:8326)
==3726==    by 0x40522B7: UA_Server_processBinaryMessage (open62541.c:8392)
==3726==    by 0x40525E2: processJobs (open62541.c:8642)
==3726==    by 0x40526FC: UA_Server_run_iterate (open62541.c:9198)
==3726==    by 0x40528D8: UA_Server_run (open62541.c:9241)
==3726==    by 0x804A76F: main (server.c:431)
==3726== 
==3726== Conditional jump or move depends on uninitialised value(s)
==3726==    at 0x4030104: bcmp (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==3726==    by 0x403AF6E: UA_String_equal (open62541.c:2927)
==3726==    by 0x4050EFA: MonitoredItem_QueuePushDataValue (open62541.c:15214)
==3726==    by 0x40511F1: Service_Publish (open62541.c:14554)
==3726==    by 0x4051F26: processMSG (open62541.c:8326)
==3726==    by 0x40522B7: UA_Server_processBinaryMessage (open62541.c:8392)
==3726==    by 0x40525E2: processJobs (open62541.c:8642)
==3726==    by 0x40526FC: UA_Server_run_iterate (open62541.c:9198)
==3726==    by 0x40528D8: UA_Server_run (open62541.c:9241)
==3726==    by 0x804A76F: main (server.c:431)
==3726== 
==3726== Conditional jump or move depends on uninitialised value(s)
==3726==    at 0x4030124: bcmp (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==3726==    by 0x403AF6E: UA_String_equal (open62541.c:2927)
==3726==    by 0x4050EFA: MonitoredItem_QueuePushDataValue (open62541.c:15214)
==3726==    by 0x40511F1: Service_Publish (open62541.c:14554)
==3726==    by 0x4051F26: processMSG (open62541.c:8326)
==3726==    by 0x40522B7: UA_Server_processBinaryMessage (open62541.c:8392)
==3726==    by 0x40525E2: processJobs (open62541.c:8642)
==3726==    by 0x40526FC: UA_Server_run_iterate (open62541.c:9198)
==3726==    by 0x40528D8: UA_Server_run (open62541.c:9241)
==3726==    by 0x804A76F: main (server.c:431)
==3726== 
getMonitoredItems was called
```

When the client call "UA_Client_Subscriptions_manuallySendPublishRequest(client)" for the first time, I have the 2 "Invalid read of size 4".

When the client call "UA_Client_Subscriptions_manuallySendPublishRequest(client)" for the second time, I have the "Conditional jump or move depends on uninitialised value(s)".

After nothing.

No issue with UaExpert

Best Regards,

Stasik0, 2016-02-03:
I could not reproduce it on my machine, do you use vanilla server and client? how exactly can i reproduce the error?

StalderT, 2016-02-03:
I use current master branch (for server and client).

I run server (with valgrind) and after I run client. Valgrind warnings occures on server application.

Stasik0, 2016-02-03:
Which palatform do you use?

ichrispa, 2016-02-03:
Could you please reproduce&paste without amalgamation? I suspect that MonitoredItem_QueuePushDataValue (ua_subscriptions.c#467) might attempt to get an empty dataValue at some point, but I'd like to verify.

A reproducable testcase would be nice. Is this the example big server?

ichrispa, 2016-02-03:
Also can't reproduce the leak with the big example server ( and subscription to currenttime) :/

Stasik0, 2016-02-03:
@ichrispa its not a leak, its some illegal access / no initialization

StalderT, 2016-02-03:
@ichrispa 
without amalgamation no issue
with amalgamation issue

my platform build from scratch : linux 32 bits / gcc 5.2.0 / glibc 2.22 / kernel 4.1.15

StalderT, 2016-02-03:
testcase : 

build with :

 cmake .. \
        -DUA_ENABLE_AMALGAMATION=ON \
        -DUA_ENABLE_SUBSCRIPTIONS=ON \
        -DUA_ENABLE_METHODCALLS=ON \
        -DUA_BUILD_EXAMPLESERVER=ON \
        -DUA_BUILD_EXAMPLECLIENT=ON \
        -DUA_ENABLE_NODEMANAGEMENT=OFF

then start : 

valgrind ./server

and ./client

ichrispa, 2016-02-03:
Sry, no errors, no supressed warnings (gcc4.9 and clang, linux x86_64, debug build, current master):

```
ichrispa@Cassandra:~/work/svn/working_copies/open62541/build> valgrind --leak-check=yes ./server
==23034== Memcheck, a memory error detector
==23034== Copyright (C) 2002-2013, and GNU GPL'd, by Julian Seward et al.
==23034== Using Valgrind-3.10.1 and LibVEX; rerun with -h for copyright info
==23034== Command: ./server
==23034== 
[02/03/2016 21:09:19.742] info/network  TCP network layer listening on opc.tcp://Cassandra:16664
[02/03/2016 21:09:21.859] info/network  New Connection 4 over TCP from 127.0.0.1:16664
[02/03/2016 21:09:22.051] info/network  Closing the Connection 4
[02/03/2016 21:09:22.090] info/network  New Connection 4 over TCP from 127.0.0.1:16664
getMonitoredItems was called
[02/03/2016 21:09:22.525] info/network  Closing the Connection 4
[02/03/2016 21:09:22.709] info/network  New Connection 4 over TCP from 127.0.0.1:16664
^C[02/03/2016 21:09:25.101] info/server Received Ctrl-C
[02/03/2016 21:09:25.104] info/network  Shutting down the TCP network layer with 1 open connection(s)
==23034== 
==23034== HEAP SUMMARY:
==23034==     in use at exit: 0 bytes in 0 blocks
==23034==   total heap usage: 4,123 allocs, 4,123 frees, 52,132,331 bytes allocated
==23034== 
==23034== All heap blocks were freed -- no leaks are possible
==23034== 
==23034== For counts of detected and suppressed errors, rerun with: -v
==23034== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
```

ichrispa, 2016-02-03:
Also tried `valgrind -v --track-origins=yes --leak-check=yes`.

Is this an 32Bit ARM plattform? I had something similar (though SIGSEVing) months ago, but I fixed it... maybe I was sloppy.

StalderT, 2016-02-03:
No intel platform

Tomorow I will make some more tests (64 vs 32 and gcc 5.2 vs 4.x)

Stasik0, 2016-02-03:
also no issue here: 
'''
$ valgrind --leak-check=full ./server
==12519== Memcheck, a memory error detector
==12519== Copyright (C) 2002-2013, and GNU GPL'd, by Julian Seward et al.
==12519== Using Valgrind-3.10.1 and LibVEX; rerun with -h for copyright info
==12519== Command: ./server
==12519== 
[02/03/2016 21:35:25.069] info/network	TCP network layer listening on opc.tcp://gemini:16664
[02/03/2016 21:35:26.809] info/network	New Connection 5 over TCP from 127.0.0.1:16664
[02/03/2016 21:35:26.883] info/network	Closing the Connection 5
[02/03/2016 21:35:26.884] info/network	New Connection 6 over TCP from 127.0.0.1:16664
getMonitoredItems was called
[02/03/2016 21:35:27.005] info/network	Closing the Connection 6
^C[02/03/2016 21:35:28.729] info/server	Received Ctrl-C
[02/03/2016 21:35:28.734] info/network	Shutting down the TCP network layer with 0 open connection(s)
==12519== 
==12519== HEAP SUMMARY:
==12519==     in use at exit: 0 bytes in 0 blocks
==12519==   total heap usage: 1,948 allocs, 1,948 frees, 3,644,959 bytes allocated
==12519== 
==12519== All heap blocks were freed -- no leaks are possible
==12519== 
==12519== For counts of detected and suppressed errors, rerun with: -v
==12519== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
'''

x86 amalgated, 32 bit, gcc 4.8.4, linux 3.13 kernel

StalderT, 2016-02-03:
Tomorow I will compile gcc 4.x on my platform and test It.

StalderT, 2016-02-04:
Ok I have found : 

```
export CFLAGS="-O2"
  
cmake ../open62541-master \
        -DUA_ENABLE_AMALGAMATION=ON \
        -DUA_ENABLE_SUBSCRIPTIONS=ON \
        -DUA_ENABLE_METHODCALLS=ON \
        -DUA_BUILD_EXAMPLESERVER=ON \
        -DUA_BUILD_EXAMPLECLIENT=ON \
        -DUA_ENABLE_NODEMANAGEMENT=OFF
```

The problem come with -O2 compilation flag (gcc 4.9.3 x86 and gcc 5.2 x86), with our without amalgation!

Stasik0, 2016-02-04:
okay i confirm, it comes from -O2, O2 is also activated by -DCMAKE_BUILD_TYPE=Release or RelWithDebInfo

StalderT, 2016-02-04:
Little test : 

```
1332 #include <stdio.h>
1333 
1334 size_t UA_calcSizeBinary(void *p, const UA_DataType *contenttype) {
1335     size_t s = 0;
1336     uintptr_t ptr = (uintptr_t)p;
1337     UA_Byte membersSize = type->membersSize;
1338     const UA_DataType *typelists[2] = { UA_TYPES, &contenttype[-contenttype->typeIndex] };
1339     for(size_t i = 0; i < membersSize; i++) {
1340         const UA_DataTypeMember *member = &contenttype->members[i];
1341         const UA_DataType *membertype = &typelists[!member->namespaceZero][member->memberTypeIndex];
1342         if(!member->isArray) {
1343 printf("HERE1 %d %d\n", member->padding, membertype->memSize);
1344             ptr += member->padding;
1345             size_t encode_index = membertype->builtin ? membertype->typeIndex : UA_BUILTIN_TYPES_COUNT;
1346             s += calcSizeBinaryJumpTable[encode_index]((const void*)ptr, membertype);
1347             ptr += membertype->memSize;
1348         } else {
1349 printf("HERE2 %d %d\n", member->padding, membertype->memSize);
1350             ptr += member->padding;
1351             const size_t length = *((const size_t*)ptr);
1352             ptr += sizeof(size_t);
1353             s += Array_calcSizeBinary(*(void *UA_RESTRICT const *)ptr, length, membertype);
1354             ptr += sizeof(void*);
1355         }
1356     }
1357     return s;
1358 }

```

With O2 : 

HERE1 0 56
HERE1 0 36
HERE1 0 56
HERE1 0 36
HERE1 0 56
HERE1 0 36

Without O2 : 

HERE1 0 56
HERE1 0 24
HERE1 0 56
HERE1 0 24
HERE1 0 56
HERE1 0 24

Stasik0, 2016-02-04:
@jpfr too much magic during last improvement-iteration? ;)

StalderT, 2016-02-04:
```
1332 #include <stdio.h>
1333 
1334 size_t UA_calcSizeBinary(void *p, const UA_DataType *contenttype) {
1335     size_t s = 0;
1336     uintptr_t ptr = (uintptr_t)p;
1337     UA_Byte membersSize = type->membersSize;
1338     const UA_DataType *typelists[2] = { UA_TYPES, &contenttype[-contenttype->typeIndex] };
1339     for(size_t i = 0; i < membersSize; i++) {
1340         const UA_DataTypeMember *member = &contenttype->members[i];
1341         const UA_DataType *membertype = &typelists[!member->namespaceZero][member->memberTypeIndex];
1342         if(!member->isArray) {
1343 printf("HERE1 %d %d %d %d\n", i, member->padding, membertype->memSize, member->memberTypeIndex);
1344             ptr += member->padding;
1345             size_t encode_index = membertype->builtin ? membertype->typeIndex : UA_BUILTIN_TYPES_COUNT;
1346             s += calcSizeBinaryJumpTable[encode_index]((const void*)ptr, membertype);
1347             ptr += membertype->memSize;
1348         } else {
1349 printf("HERE2 %d %d %d %d\n", i, member->padding, membertype->memSize, member->memberTypeIndex);
1350             ptr += member->padding;
1351             const size_t length = *((const size_t*)ptr);
1352             ptr += sizeof(size_t);
1353             s += Array_calcSizeBinary(*(void *UA_RESTRICT const *)ptr, length, membertype);
1354             ptr += sizeof(void*);
1355         }
1356     }
1357     return s;
1358 }
```

With O2 : 
HERE1 0 0 56 22
HERE1 1 0 24 23
HERE1 0 0 56 22
HERE1 1 0 24 23
HERE1 0 0 56 22
HERE1 1 0 24 23

Without O2 :
HERE1 0 0 56 22
HERE1 1 0 36 21
HERE1 0 0 56 22
HERE1 1 0 36 21
HERE1 0 0 56 22
HERE1 1 0 36 21

Stasik0, 2016-02-04:
@StalderT yes this is sizeof(UA_Variant) and it is, imho, okay that it varies. We just have to make sure it is accessed correctly

StalderT, 2016-02-04:
@Stasik0 : sizeof(UA_Variant) is constant : 

```
1332 #include <stdio.h>
1333 
1334 size_t UA_calcSizeBinary(void *p, const UA_DataType *contenttype) {
1335     size_t s = 0;
1336     uintptr_t ptr = (uintptr_t)p;
1337     UA_Byte membersSize = type->membersSize;
1338     const UA_DataType *typelists[2] = { UA_TYPES, &contenttype[-contenttype->typeIndex] };
1339     for(size_t i = 0; i < membersSize; i++) {
1340         const UA_DataTypeMember *member = &contenttype->members[i];
1341         const UA_DataType *membertype = &typelists[!member->namespaceZero][member->memberTypeIndex];
1342         if(!member->isArray) {
1343 printf("HERE1 %d %d %d %d %d\n", i, member->padding, membertype->memSize, member->memberTypeIndex, sizeof(UA_Variant));
1344             ptr += member->padding;
1345             size_t encode_index = membertype->builtin ? membertype->typeIndex : UA_BUILTIN_TYPES_COUNT;
1346             s += calcSizeBinaryJumpTable[encode_index]((const void*)ptr, membertype);
1347             ptr += membertype->memSize;
1348         } else {
1349 printf("HERE2 %d %d %d %d\n", i, member->padding, membertype->memSize, member->memberTypeIndex);
1350             ptr += member->padding;
1351             const size_t length = *((const size_t*)ptr);
1352             ptr += sizeof(size_t);
1353             s += Array_calcSizeBinary(*(void *UA_RESTRICT const *)ptr, length, membertype);
1354             ptr += sizeof(void*);
1355         }
1356     }
1357     return s;
1358 }

```
With O2:
HERE1 0 0 56 22 24
HERE1 1 0 36 21 24
HERE1 0 0 56 22 24
HERE1 1 0 36 21 24
HERE1 0 0 56 22 24
HERE1 1 0 36 21 24

Without O2:
HERE1 0 0 56 22 24
HERE1 1 0 24 23 24
HERE1 0 0 56 22 24
HERE1 1 0 24 23 24
HERE1 0 0 56 22 24
HERE1 1 0 24 23 24

But member->memberTypeIndex change

Stasik0, 2016-02-04:
@StalderT could you test the last commit?

StalderT, 2016-02-04:
Don't work for me.

client never end : 

[root@devel build2]$ ./server
[02/04/2016 21:54:07.133] info/network	TCP network layer listening on opc.tcp://devel:16664
[02/04/2016 21:54:11.247] info/network	New Connection 4 over TCP from 127.0.0.1:16664

[root@devel build2]$ ./client

StalderT, 2016-02-04:
client block in UA_Client_getEndpoints. Probably an another problem. I inspect.

Stasik0, 2016-02-04:
@StalderT i fixed too much :) try now please

StalderT, 2016-02-04:
Yes !!!!

It works.

If someone can explain me negative index?

closed 2016-02-04 by StalderT

Stasik0, 2016-02-04:
negative array index works just like a pointer subtraction, since a[offset] equals to *(a+offset) -> a[-offset] equals to *(a-offset). In that case this is a jump to the beginning of the types array of the other namespace

StalderT, 2016-02-04:
With this problem, I have seen that the client, loop infinitely in SecureChannelHandshak.

```
249     do {
250         retval = c->recv(c, &reply, client->config.timeout);
251         retval |= UA_Connection_completeMessages(c, &reply, &realloced);
252         if(retval != UA_STATUSCODE_GOOD) {
253             UA_LOG_DEBUG(client->logger, UA_LOGCATEGORY_SECURECHANNEL,
254                          "Receiving OpenSecureChannelResponse failed");
255             return retval;
256         }
257     } while(reply.length == 0);
```

It would be great to return an error after n retry.

Stasik0, 2016-02-04:
we actualy do not need a while loop at all, since we have a timeout in recv

Stasik0, 2016-02-04:
@StalderT could you open a new issue? ) a good blocking-server commit that can be tested is: d6935e65fdfe397b456bdd969e1f22eb9b92d5aa

jpfr, 2016-02-04:
> we actualy do not need a while loop at all, since we have a timeout in recv

We do need the loop. TCP has a streaming API. So we can receive "half packets" from the OS. So we loop until we have pieced together at least one chunk. That's what UA_Connection_completeMessages does.

StalderT, 2016-02-04:
Ok. But, do we really need TEST_RETRY in networklayer_tcp.c#161 for EAGAIN and EWOULDBLOCK?

Stasik0, 2016-02-04:
@StalderT thats the point, it depends on whether the socket is blocking or non-blocking, in the blocking case (client) it means a timeout and an error, in the non-blocking case (server) it means everything is good, we will need to add a flag marking the socket-mode

StalderT, 2016-02-04:
getsockopt?

StalderT, 2016-02-04:
Or test timeout variable?

Tomorow I will open a new issue.