open62541/open62541 #535: gethostname problem.

Wtango, 2016-02-03:
In a "FindServers" service,the open62541 server responds a endpointUrl like that:

> opc.tcp://tango-Lenovo:16664
In my network environment，the client can't connect my open62541 server through that domain.
Any solution？

Stasik0, 2016-02-03:
Did you try the ip address directly?

Wtango, 2016-02-03:
@Stasik0 yes,i use ip address in a uaexpert mobile client,but open62541 server responds a hostname i got it in wireshark.

Stasik0, 2016-02-03:
Yes, we have to respond a hostname, since certificates have to sign one.

Other ua dem servers will peobably do the same. 

You can set hostname on your server machine, s.t. it gets resolved by dns.

jpfr, 2016-02-03:
I added a "feature tag" to this issue.
Currently we use the gethostname result from the network layer.
Imho the cleanest solution is adding custom endpointurls in the server config.
Then we can duplicate the endpoints for all endpointurls at startup.

This feature is currently up for grabs. Please write a short comment if you (plan to) work on it.

Wtango, 2016-02-04:
@jpfr Sorry for that i will be AFK tomorrow for half a month.

StalderT, 2017-11-06:
I will work on it. Can I add a parameter to 'UA_ServerConfig_new_minimal' function?

Pro, 2017-11-19:
Or maybe even better adding a new method like `UA_ServerConfig_custom` which supports more parameters (if necessary). Like this we do not break the current api.

For the discovery URL we should already prepare to support the path in the future, so when we introduce that feature we don't need to change the api. e.g. `opc.tcp://host:4840/some_path/value` is also a valid OPC UA endpoint url.

HolgerJeromin, 2017-11-23:
Close this? @pro

Pro, 2017-11-23:
Thanks, yes, fixed in #1382
There is now a method `UA_ServerConfig_set_customHostname(config, UA_STRING("custom"));`

closed 2017-11-23 by Pro