open62541/open62541 #539: client can loop infinitely

StalderT, 2016-02-05:
Hello,

If the client connect to the server and that the server reply nothing, client can loop infinitely in : 

UA_StatusCode HelAckHandshake
```
249     do {
250         retval = c->recv(c, &reply, client->config.timeout);
251         retval |= UA_Connection_completeMessages(c, &reply, &realloced);
252         if(retval != UA_STATUSCODE_GOOD) {
253             UA_LOG_DEBUG(client->logger, UA_LOGCATEGORY_SECURECHANNEL,
254                          "Receiving OpenSecureChannelResponse failed");
255             return retval;
256         }
257     } while(reply.length == 0);
```
a good blocking-server commit that can be tested is: https://github.com/open62541/open62541/commit/d6935e65fdfe397b456bdd969e1f22eb9b92d5aa

To correct this problem, socket_recv function must return an error, if a timeout occures (only for blocking socket).

Currently on windows, there is no easy way to obtain the socket's current blocking mode since WSAIsBlocking was deprecated (http://stackoverflow.com/questions/5489562/in-win32-is-there-a-way-to-test-if-a-socket-is-non-blocking).

I suggest to use timeout variable (PR https://github.com/open62541/open62541/pull/538) : 
- socket is blocking
    - timeout = 0 : we should test EAGAIN and EWOULDBLOCK (OK)
    - timeout > 0 : we should ignore EAGAIN and EWOULDBLOCK (OK)
- socket is non-blocking
    - timeout = 0 : we should test EAGAIN and EWOULDBLOCK (OK)
    - timeout > 0 : we should test EAGAIN and EWOULDBLOCK (NOK). This case is problematic, because the PR ignore EAGAIN and EWOULDBLOCK. Normally only the server is non-blocking and timeout is set to 0. This case should not come.

If you have a better idea?

Stasik0, 2016-02-05:
looks good for me

jpfr, 2016-02-06:
To me, we only have two use cases for recv. If the socket blocks does not matter.
- Timout: Wait (and loop) until a complete message has arrived or return an error after a timeout. 
- No timeout: Receive a message immediately or return nothing (and no error).
We could reduce code duplication by moving `UA_Connection_completeMessages` into the network layer's socket_recv.
For the blocking case, run `UA_Connection_completeMessages` in a loop until we have at least one complete message. And adapt the timeout for the 2nd, 3rd, ... recv accordingly.
But all inside `socket_recv`.

Stasik0, 2016-02-06:
Will calling completeMessage inside of the NL break the layer architecture? Imho the PR is good for now.

Stasik0, 2016-02-09:
okay, we merged the PR #538, but will postpone the issue for a later "clean" solution

closed 2016-06-06 by jpfr