open62541/open62541 #541: Stress testing the server causes a block (subscription)

StalderT, 2016-02-05:
Hello,

I have add to server.c :
```
static UA_StatusCode
readTest(void *handle, const UA_NodeId nodeId, UA_Boolean sourceTimeStamp,
             const UA_NumericRange *range, UA_DataValue *value) {
    if(range) {
        value->hasStatus = UA_TRUE;
        value->status = UA_STATUSCODE_BADINDEXRANGEINVALID;
        return UA_STATUSCODE_GOOD;
    }   
    UA_UInt32 val = (UA_UInt32)handle;
    UA_DateTime currentTime = UA_DateTime_now();
    UA_Variant_setScalarCopy(&value->value, &val, &UA_TYPES[UA_TYPES_UINT32]);
    value->hasValue = UA_TRUE;
    if(sourceTimeStamp) {
        value->hasSourceTimestamp = UA_TRUE;
        value->sourceTimestamp = currentTime;
    }   
    return UA_STATUSCODE_GOOD;
}
```

and  

```
    for(UA_Uint32 i=0; i<10000; i++)
    {
            UA_DataSource dateDataSource_test = (UA_DataSource) {.handle = (void*)i, .read = readTest, .write = NULL};
            UA_VariableAttributes v_attr_test;
            UA_VariableAttributes_init(&v_attr_test);
            char name[100];
            snprintf(name, 100, "TEST%04d", i);
            v_attr_test.description = UA_LOCALIZEDTEXT("en_US",name);
            v_attr_test.displayName = UA_LOCALIZEDTEXT("en_US",name);
            const UA_QualifiedName dateName_test = UA_QUALIFIEDNAME(1, name);
            UA_NodeId dataSourceId_test;
            UA_Server_addDataSourceVariableNode(server, UA_NODEID_NULL,
                                                UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
                                                UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES), dateName_test,
                                                UA_NODEID_NULL, v_attr_test, dateDataSource_test, &dataSourceId_test);
    }
```

If I add 1000 test variables to Data Access View in UaExpert, all works fine : 

[image]

If I try to add 1500 test variables to Data Access View in UaExpert server block with 100% CPU usage and UaExpert display : 
[image]

Wireshark : [opccapt.zip](https://github.com/open62541/open62541/files/118996/opccapt.zip)

jpfr, 2016-02-09:
bump @ichrispa ?

ichrispa, 2016-03-09:
hmphh... good question. Is there any trace from the stack side to find out which loop causes the glitch?

jpfr, 2016-05-09:
@StalderT 
Can you test with master after commit e4ff940?
The subscriptions part was refactored and should be fine.

StalderT, 2016-05-11:
If I try to monitor 1000 items I have the error : 
CretaeMonitoredItems failed (BadDisconnected)

[image]

Sometimes I have the error : 
CreateMonitoredItems failed (BadTcpMessageTooLarge)

[image]

jpfr, 2016-05-13:
Probably the message size got bigger than what the client can handle.
This will be fixed in the next days when PR #534 lands.

I'll test it myself and come back to you.

ichrispa, 2016-05-20:
@StalderT: afaik @jpfr has refactored the server-side subscriptions; is this issue still persisting?

StalderT, 2016-05-20:
If I try to monitor 3500 nodes I have the error : 

[image]

sometimes I have :

[image]

with this case, the server block with 100% CPU usage

ichrispa, 2016-05-20:
When the server generates the 100% CPU load, can you ask the debugger to pause and find out where the hanging loop is?

Also: is this a single-threaded server?

I suspect there might be a list iterator somewhere and a node gets unlinked uncleanly, causing an the iterator to recurse indefinetely.

StalderT, 2016-05-20:
Yes single-threaded server

```
^C
Program received signal SIGINT, Interrupt.
0xb7e923a7 in ?? () from /lib/libc.so.6
(gdb) where
#0  0xb7e923a7 in ?? () from /lib/libc.so.6
#1  0xb7e93c45 in malloc () from /lib/libc.so.6
#2  0xb7fa1df3 in UA_ByteString_allocBuffer (bs=0xbffff0c4, length=5) at /data/linux/open62541/src/build/open62541.c:3157
#3  0xb7fbb8fa in SampleCallback (server=0x804e038, monitoredItem=0x8849820) at /data/linux/open62541/src/build/open62541.c:20411
#4  0xb7fafe91 in processJobs (server=0x804e038, jobs=0xb75d6b80, jobsSize=1) at /data/linux/open62541/src/build/open62541.c:14881
#5  0xb7fb020d in processRepeatedJobs (server=0x804e038, current=48467300116) at /data/linux/open62541/src/build/open62541.c:15132
#6  0xb7fb0710 in UA_Server_run_iterate (server=0x804e038, waitInternal=true) at /data/linux/open62541/src/build/open62541.c:15443
#7  0xb7fb0943 in UA_Server_run (server=0x804e038, running=0x804c1b0 <running>) at /data/linux/open62541/src/build/open62541.c:15527
#8  0x0804b5d2 in main ()
(gdb) continue
Continuing.
^C
Program received signal SIGINT, Interrupt.
0xb7fdbad0 in __kernel_vsyscall ()
(gdb) where
#0  0xb7fdbad0 in __kernel_vsyscall ()
#1  0xb7fdb908 in __vdso_clock_gettime ()
#2  0xb7f01bb2 in clock_gettime () from /lib/libc.so.6
#3  0xb7fa18e2 in UA_DateTime_nowMonotonic () at /data/linux/open62541/src/build/open62541.c:3068
#4  0xb7fb06f9 in UA_Server_run_iterate (server=0x804e038, waitInternal=true) at /data/linux/open62541/src/build/open62541.c:15442
#5  0xb7fb0943 in UA_Server_run (server=0x804e038, running=0x804c1b0 <running>) at /data/linux/open62541/src/build/open62541.c:15527
#6  0x0804b5d2 in main ()
(gdb) continue
Continuing.
^C
Program received signal SIGINT, Interrupt.
0xb7fdbad0 in __kernel_vsyscall ()
(gdb) where
#0  0xb7fdbad0 in __kernel_vsyscall ()
#1  0xb7eee80d in select () from /lib/libc.so.6
#2  0xb7fbe836 in ServerNetworkLayerTCP_getJobs (nl=0xbffff708, jobs=0xbffff1d8, timeout=17) at /data/linux/open62541/src/build/open62541.c:21804
#3  0xb7fb078e in UA_Server_run_iterate (server=0x804e038, waitInternal=true) at /data/linux/open62541/src/build/open62541.c:15456
#4  0xb7fb0943 in UA_Server_run (server=0x804e038, running=0x804c1b0 <running>) at /data/linux/open62541/src/build/open62541.c:15527
#5  0x0804b5d2 in main ()
(gdb) continue
Continuing.
^C
Program received signal SIGINT, Interrupt.
0xb7fbbe90 in UA_Subscription_publishCallback (server=0x804e038, sub=0x84b7580) at /data/linux/open62541/src/build/open62541.c:20546
20546	        LIST_FOREACH(mon, &sub->MonitoredItems, listEntry) {
(gdb) where
#0  0xb7fbbe90 in UA_Subscription_publishCallback (server=0x804e038, sub=0x84b7580) at /data/linux/open62541/src/build/open62541.c:20546
#1  0xb7fafe91 in processJobs (server=0x804e038, jobs=0x84b73e8, jobsSize=1) at /data/linux/open62541/src/build/open62541.c:14881
#2  0xb7fb020d in processRepeatedJobs (server=0x804e038, current=49364515576) at /data/linux/open62541/src/build/open62541.c:15132
#3  0xb7fb0710 in UA_Server_run_iterate (server=0x804e038, waitInternal=true) at /data/linux/open62541/src/build/open62541.c:15443
#4  0xb7fb0943 in UA_Server_run (server=0x804e038, running=0x804c1b0 <running>) at /data/linux/open62541/src/build/open62541.c:15527
#5  0x0804b5d2 in main ()
(gdb) continue
Continuing.
```

[image]

ichrispa, 2016-05-20:
When trying to reproduce your code, there are several things I noticed:

You are passing (as a handle) the iterator from the for loop:
```
for(UA_UInt32 i=0; i<10000; i++)
    {
      UA_DataSource dateDataSource_test = (UA_DataSource) {.handle = (void*) &i, .read = readBenchmarkCallback, .write = NULL};
```

This is completely invalid - the variable is no longer allocated/in scope once you leave the loop. Your value returned by the Node (val) is a pointer to a non allocated object - any attempt of derefencing will propably cause all kinds of memory violations. But you do not use the handle at all...

```
UA_UInt32 val = (UA_UInt32)handle;
UA_Variant_setScalarCopy(&value->value, &val, &UA_TYPES[UA_TYPES_UINT32]);
```
In essence, your variables return the address of a temp variable allocated inside the scope of your callback. This is only valid if your archtecture uses 32 bit addresses.

After doing some cleanup on your testbench side, I can compile and run the test with all 10000 variable nodes subscribed to in UA Expert.

ichrispa, 2016-05-20:
Addendum: Tested on Linux x86, x86_64, Win7 x86_64

StalderT, 2016-05-20:
OK.

Can you give me your sample code ?

I have try with : 

```
static UA_StatusCode
readTest(void *handle, const UA_NodeId nodeId, UA_Boolean sourceTimeStamp,
             const UA_NumericRange *range, UA_DataValue *value) {
    if(range) {
        value->hasStatus = UA_TRUE;
        value->status = UA_STATUSCODE_BADINDEXRANGEINVALID;
        return UA_STATUSCODE_GOOD;
    }   
    UA_UInt32 val = (UA_UInt32)100;
    UA_DateTime currentTime = UA_DateTime_now();
    UA_Variant_setScalarCopy(&value->value, &val, &UA_TYPES[UA_TYPES_UINT32]);
    value->hasValue = UA_TRUE;
    if(sourceTimeStamp) {
        value->hasSourceTimestamp = UA_TRUE;
        value->sourceTimestamp = currentTime;
    }   
    return UA_STATUSCODE_GOOD;
}

```

```
    for(UA_UInt32 i=0; i<10000; i++)
    {
            UA_DataSource dateDataSource_test = (UA_DataSource) {.handle = NULL, .read = readTest, .write = NULL};
            UA_VariableAttributes v_attr_test;
            UA_VariableAttributes_init(&v_attr_test);
            char name[100];
            snprintf(name, 100, "TEST%04d", i);
            v_attr_test.description = UA_LOCALIZEDTEXT("en_US",name);
            v_attr_test.displayName = UA_LOCALIZEDTEXT("en_US",name);
            const UA_QualifiedName dateName_test = UA_QUALIFIEDNAME(1, name);
            UA_NodeId dataSourceId_test;
            UA_Server_addDataSourceVariableNode(server, UA_NODEID_NULL,
                                                UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
                                                UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES), dateName_test,
                                                UA_NODEID_NULL, v_attr_test, dateDataSource_test, &dataSourceId_test);
    }

```

And I have this error : 

[image]

ichrispa, 2016-05-20:
The datasource creation part is nearly identical (naming was bit different). My read uses a global

```
UA_UInt32 val=0;

static UA_StatusCode
readTest(void *handle, const UA_NodeId nodeId, UA_Boolean sourceTimeStamp,
             const UA_NumericRange *range, UA_DataValue *value) {
    if(range) {
        value->hasStatus = UA_TRUE;
        value->status = UA_STATUSCODE_BADINDEXRANGEINVALID;
        return UA_STATUSCODE_GOOD;
    }   
    val++;
    UA_Variant_setScalarCopy(&value->value, &val, &UA_TYPES[UA_TYPES_UINT32]);
    value->hasValue = UA_TRUE;
    if(sourceTimeStamp) {
        value->hasSourceTimestamp = UA_TRUE;
        value->sourceTimestamp =  UA_DateTime_now();
    }   
    return UA_STATUSCODE_GOOD;
}
```

Can you give some plattform specifics (Server arch, Client arch, OS on either side)? What network connection is between the sides? Could you have a real timeout because of the bandwidth?

StalderT, 2016-05-20:
My server arch :  linux 4.1.15 on VMWARE / gcc 5.3.0 
My client arch : Windows 10

The server and the client run on the same machine (no network connection timeout)

If it helps you I made a video:   https://www.dropbox.com/s/cerovj7umxodlw5/linux.WMV?dl=0

I have test it under cygwin and I have the same result : 

https://www.dropbox.com/s/h0w99lwlp4uvq81/cygwin.WMV?dl=0

StalderT, 2016-06-21:
https://github.com/open62541/open62541/commit/87393df325f944e05edd5f0f5bf00c26d537e73b fix the problem

Thanks!

StalderT, 2016-06-21:
Oops. No, the problem still exists.

StalderT, 2016-06-23:
With current master : 

On UA_Expert : 

[image]

On open62541 server : 

```
[06/23/2016 09:37:24.941] debug/session Connection 4 | SecureChannel 2 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | Read the attribute 13
[06/23/2016 09:37:24.941] debug/session Connection 4 | SecureChannel 2 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | Read the attribute 13
[06/23/2016 09:37:24.941] debug/session Connection 4 | SecureChannel 2 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | Read the attribute 13
[06/23/2016 09:37:24.941] debug/session Connection 4 | SecureChannel 2 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | Read the attribute 13
[06/23/2016 09:37:24.941] debug/session Connection 4 | SecureChannel 2 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | Read the attribute 13
[06/23/2016 09:37:24.941] debug/session Connection 4 | SecureChannel 2 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | Read the attribute 13
[06/23/2016 09:37:24.941] debug/session Connection 4 | SecureChannel 2 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | Read the attribute 13
send chunk of length 65534
[06/23/2016 09:37:24.947] debug/channel Connection 4 | SecureChannel 2 | Could not send the message over the SecureChannel with error code 0x80060000
send chunk of length 52
[06/23/2016 09:37:24.953] debug/session Connection 4 | SecureChannel 2 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | Read the attribute 13
[06/23/2016 09:37:24.953] debug/session Connection 4 | SecureChannel 2 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | Subscription 2 | MonitoredItem 1 | Sampling the value
[06/23/2016 09:37:24.953] debug/session Connection 4 | SecureChannel 2 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | Read the attribute 13
[06/23/2016 09:37:24.953] debug/session Connection 4 | SecureChannel 2 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | Subscription 2 | MonitoredItem 2 | Sampling the value
[06/23/2016 09:37:24.953] debug/session Connection 4 | SecureChannel 2 | Session {152CA78D-6003-027C-F3BF-BB7BEEFEEFBE} | Read the attribute 13

```

jpfr, 2016-06-23:
Fixed. On my box, I can do 10,000 MonitoredItems now.
With 50ms sampling interval, this gives a 15% load on 1 CPU. :-)

closed 2016-06-23 by jpfr

StalderT, 2016-06-23:
@jpfr  Good job. Works for me. Thanks. 

For me with 50ms sampling interval, this gives a 5% load on 1 CPU on a I7

commit 94c1070f (2016-06-23): fix #541