open62541/open62541 #542: Advanced Examples

coussej, 2016-02-05:
Hi everyone, first of all: great job on this project. Looks very promising.

I was wondering: can anyone share or link to some more advanced examples or projects that use this implementation?

I want to write something for collecting OPC data and log it to InfluxDb, but I don't have enough experience in C (or in general :smile:) to start from scratch.

Thanks in advance.

jpfr, 2016-02-06:
Hey @coussej,

that sounds like a great idea to use InfluxDB. Looks like the HackerNews crowd is slowly dicovering OPC UA. :-)

You will have to make two major decisions for your design:

**Server or Client**
Usually, the data source is the server. Clients connect and read the data. However, if you have the data source on the same machine, you can be the server and use OPC UA to expose historical data to interested clients.

**Polling or Subscriptions**
Usually, people use the subscriptions mechanism to define fixed sample intervals for data. But when subscriptions are not supported, you can also poll the data manually with the Read-Service.

Maybe you can start by playing around with the basics.
If you have an existing server, I suggest you use the free UAExpert client to explore the content.
Then you can look at our client examples on how to access values and call methods.

When you have specific questions, we are gladly helping you out. But there is no general support from us, as this would quickly consume all of our time. If you need interactive help, most of the time there are people idling on IRC.

Best, Julius

coussej, 2016-02-06:
Hi @jpfr , 

thanks for your answer! I'm indeed looking to write an OPC client that installs a subscription for a number of defined 'tags' in the proprietary OPC servers from Siemens, Rockwell, ... 

However, with few examples combined with my limited knowledge on this topic, it is really hard to get a grasp of how people handle monitoring a larger set nodes for longer periods, that's why I'm looking for some projects outside of this repo that already use this implementation to achieve similar goal, I suppose there must be some?

Best, Jeroen.

Stasik0, 2016-02-06:
Hey, are we talking about OPC (classic) or OPC UA? Do your data sources have an OPC UA server?

coussej, 2016-02-07:
Hi @Stasik0, Although classic OPC is still the most common, most vendors also provide OPC UA access in their server software (Simatic NET for example), so I'd prefer implementing an UA collector.

jpfr, 2016-02-08:
We use open62541 in an every increasing number of research projects.
I know some industrial users. But only one who is making it public / explicit on the homepage: http://www.steinhoff-automation.com/?OPC_UA_Client

Please bear in mind that we are still a zero-dot release phase. Most commercial users want to pay for an SDK with support through a company. We don't provide that.

coussej, 2016-02-09:
I'm a heavy supporter for open source projects like this. You don't provide tech support but I have quicker answers to my questions than with most SLA's :smile: 

Anyway, I started digging through your tutorials and code, while learning a bit of C, and I am now at the point where I have a very basic client wrapped in a Go package using [cgo](https://golang.org/cmd/cgo/). I do this because I like Go, and I find it very easy to write this kind of applications in it. I expose only high level functions, and do the low level stuff in C:

```go
type Client struct {
	c_client *C.UA_Client
}

func newClient() Client {
	c := C.gopcua_client_new()
	return Client{c}
}

func (c *Client) Connect(ConnInfo string) (err error) {
	ci := C.CString(ConnInfo)
	defer C.free(unsafe.Pointer(ci))
	_, err = C.gopcua_client_connect(c.c_client, ci)
	return
}

func (c *Client) ReadInt32(NodeId string) (i int32, err error) {...}
```

We'll see where it gets me. Once I have a working version I will most certainly upload it to Github.

Stasik0, 2016-02-09:
@coussej nice one, @jpfr has a similar project by using LUA as a host https://github.com/jpfr/uascript

coussej, 2016-02-09:
Here it is! [gopcua](https://github.com/coussej/gopcua). Please be aware that it's not really a wrapper like the LUA wrapper from @jpfr, but just some high lvl funcs to get access to OPC UA values.

I don't know if my approach makes any sense, feel free to suggest anything.

jpfr, 2016-02-09:
That's looking already quite useful!

To me, the architecture of such a language mapping depends on how far you want to go.
If you want to a quick solution to read some integer data, then your current code is perfectly nice.

If you want to support "all of OPC UA" with structured data in requests and responses (such as a method argument or a browse response that contains nodeids), then you need to wrap
- the 25 builtin types from `ua_types.h`
- structures built up from builtin types
- arrays

I don't know Go well enough and can only tell from my experience with Lua. There, it was possible to manually wrap only the builtin types and take everything else at runtime from the datatype descriptions in the global `UA_TYPES` array.

Then we can instantiate structures, such as a `WriteValue,` and interact with similarly to the standard Lua syntax.

```lua
client = ua.Client()
client:connect("opc.tcp://127.0.0.1:51202")

w1 = ua.types.WriteValue()
w1.nodeId = ua.types.NodeId(1, "z-axis")
w1.attributeId = ua.attributeIds.Value
w1.value.value = ua.types.Float(-0.2)
client:write(w1)
```

Stasik0, 2016-02-10:
@coussej just a small hint: for nodeId, you need at least the namespace indenifier (int32) and a string (nodeid) so you can also read nodes in other namespaces. Furthermore nodeid can be not only a string but also a numeric type (and some more). You can find examples of nodes in namespace=0 in our demo server (e.g. below "Server" node)

coussej, 2016-02-10:
Just found out cgo has a tool to generate Go types that can be casted to and from the source C types. Might be easy to just map all the default types this way. Also the nodeId issue you mention could be solved

jpfr, 2016-03-05:
Closing this for now.
Please reopen if you have a remaining question that is on topic for this issue.

closed 2016-03-05 by jpfr