open62541/open62541 #543: Client crahsed in UA_Client_connect

tpiat, 2016-02-10:
Version f68f7db744 works.

I try connect to simatic server

version fed0c9a6da

I call UA_Client_connect. and receive SIGILL,

Call stack
UA_Client_connect 
UA_Client_reset 
UA_Client_deleteMembers
UA_SecureChannel_deleteMembersCleanup

Latest version b1e36f73cc crashes too in the same function.

Are there hints what to do?

Stasik0, 2016-02-10:
The call to _reset from _connect function indicates an error (probably a
jump to
https://github.com/open62541/open62541/blob/master/src/client/ua_client.c#L584),
can you step through to see which error occurs?

tpiat, 2016-02-10:
In line 12559 
retval = EndpointsHandshake(client);
retval =UA_STATUSCODE_BADINTERNALERROR

Stasik0, 2016-02-10:
hi @tpiat, i was not able to reproduce the exact bug, but i found some other problems. The client works now (again) with my simatic opc ua server (V07.01. SP02), could you test?

tpiat, 2016-02-10:
Do not works, the same error.

Stasik0, 2016-02-10:
can you tell me the line of which file errors? (which tag do you use)? 
is it next to?         
```UA_LOG_ERROR(client->logger, UA_LOGCATEGORY_CLIENT, "No suitable endpoint found");```
or?
```UA_LOG_ERROR(client->logger, UA_LOGCATEGORY_CLIENT, "No anonymous token found");```

can you get a wireshark trace?

tpiat, 2016-02-10:
Download new verion. Do not crahed.
Thank you!

Stasik0, 2016-02-10:
okay

closed 2016-02-10 by Stasik0