open62541/open62541 #545: MSVC 2015 error execution

ipueyo, 2016-02-11:
Hello,

I'm quite new at OPC UA world, but i have followed the official tutorial (http://open62541.org/doc/current/building.html) from "Building the Library" tag to "First steps with open62541-server" with the next software installation on Microsoft Windows Server 2012 R2 Standard :
- cmake.3.5.0
- lxml 3.5.0
- python-2.7.10
- Microsoft Visual Studio 2015 Community Edition 14.0.24720.00 Update 1

The main open62541 project used is the "open62541-0.1.1" release and I have used the next cMake command to generate the MSVC solution:
`"C:\Users\Administrador\Desktop\Open6241(OPC UA Client)\cmake.3.5.0\bin\cmake.exe" -G "Visual Studio 14 2015"`
[image]

The problem appears after building all project and proceed to execute "server_simple.c" where I found the next error:
[image]

What could be the problem? Is there another way to execute it?

Stasik0, 2016-02-11:
@ipueyo it seems to be some issue with the environment (maybe some 32- vs. 64-bit issue?)

Try to pull the current master (the pre-0.2 version), it is compiled automatically with msvs2013 here: https://ci.appveyor.com/project/Stasik0/open62541/build/391#L273. You can find the artifacts of the build here: https://ci.appveyor.com/project/Stasik0/open62541/build/391/artifacts, try to compare or try to run these files.

you will probably need to set in order to get server_simple.c you will need to activate -DUA_BUILD_EXAMPLES:BOOL=ON to get the examples compiled.

ipueyo, 2016-02-12:
@Stasik0 First of all, thank you for you quick response. I have tried to use the cmake both "Visual Studio 14 2015" and "Visual Studio 14 2015 Win64" with the -DUA_BUILD_EXAMPLES:BOOL=ON set for examples compiled. However I get the same error execution and the Exe files aren't generated into Debug folder as  https://ci.appveyor.com/project/Stasik0/open62541/build/391/artifacts. These are the only files generated:
[image]

P.S.: About the environment, the architecture of my Microsoft Windows Server 2012 R2 Standard is 64-bits and I have executed both Exe versions of https://ci.appveyor.com/project/Stasik0/open62541/build/391/artifacts without any problem,

Stasik0, 2016-02-12:
You do not compile the last version. 

Try to build master https://github.com/open62541/open62541/archive/master.zip

With exact the same cmake config line as the buld server.

ipueyo, 2016-02-12:
@Stasik0 Perfect! Wtih this new project all the exe files are generated during the building and can be executed correctly.

Thank you very much for your help.

Kind regards.

closed 2016-02-12 by jpfr