open62541/open62541 #548: Channel renewal failure

maurhub, 2016-02-15:
Guys,

Sorry for the minimum info I can provide on this one;

Platform: Windows
Lib Version: Trunk
Server endpoint: KEPServerEx 5 Non secure endpoint

As soon as the channel needs to be renewed the renewal process fails. It doesn't matter how long you set the channel life. I suspect this might be a setting on the server to not allow channel renewal?

As a workaround I added a disconnect - connect call at UA_Client_manuallyRenewSecureChannel with a flag to block the recursion but sometimes it failed anyways, so I had to manage this same behaviour at my top level API.

Stasik0, 2016-02-15:
Can you set loglevel down to see debug messages? Can you capture a
wireshark traffic for us?

jpfr, 2016-02-19:
Hey @maurhub,

the client currently does not have a "life on its own", i.e. does not start a background thread.
So in order to keep a securechannel alive, you either have the regularly call UA_Client_manuallyRenewSecureChannel or do a keepalive read. For example on the server's status variable.

The client **will get a background thread**. But nobody got around to implementing it so far.
Our focus has been on the server.

Did you get your application to run with the workaround you described?

maurhub, 2016-02-19:
Hmm weird.
Since i was doing a read over several tags in a 2 secs interval and it
still failed.

As far as i could debug, the fail was when the renewal was intended. The
server socket was returning connection closed.

Stasik0, 2016-02-19:
@maurhub i tried to get a demo for windows, but I have a problem: it does only have signed or encrypted endpoints. So once again: could you please capture the traffic with wireshark? A tutorial is here: https://github.com/open62541/open62541/wiki/Capturing-traffic-with-Wireshark

maurhub, 2016-02-19:
Hmm, that is weird, I'm using a non secured endpoint for Kepware server 5
on windows (maybe because mine is not demo).

I'll try to capture the traffic with wireshark.

In the mean, check this snapshot, maybe you can create or disable
encription on the endpoints:

[image: Imágenes integradas 1]

Just tick "none" and untick the other encription options.

Stasik0, 2016-02-19:
@maurhub cant see the image

maurhub, 2016-02-19:
[image]

Stasik0, 2016-02-19:
I cannot find this window in demo, gotta go now, will be afk for a week :(

maurhub, 2016-02-19:
Hmm ok, I'll try to capture the wireshark traffic and upload it to the
issue.
Maybe when you're back you can take a look.

HolgerJeromin, 2016-02-19:
FYI I just updated the documentation for capturing on localhost. Installing Npcap works with no further changes. Tested this a few days ago.

maurhub, 2016-02-20:
Here is the capture, in pcap and pcapng formats just in case. Also here is the log. As you can see the server (or the libe itself?) sends the FIN flag to close the connection, don't know why. After that you can see a new connection being made by the workaround I added to reconnect at my client code.

https://www.dropbox.com/s/amhiexlrxcmbgdq/renewal_failed.pcap?dl=0
https://www.dropbox.com/s/re7fuae5pzg1k4v/renewal_failed.pcapng?dl=0

Stasik0, 2016-02-20:
Does is also fail with uaexpert? Can you also capture the traffic? It would be good to see a successful renewal.

maurhub, 2016-02-20:
What do you mean?  The traffic is captured and attached (dropbox links)

jpfr, 2016-02-22:
Thanks for uploading the traffic capture. I'll get to debugging it this evening.

jpfr, 2016-02-23:
Hey @maurhub,
in the network dump, the following happens:

[image]

After the timeout, the client sends an OpenSecureChannel request to renew the channel.
However, the requested security mode is `invalid`. It should be `none` to request clear-text communication. The server rejects the faulty renewal and closes the connection.

Our client implementation always chooses `none`: https://github.com/open62541/open62541/blob/master/src/client/ua_client.c#L211
So that issue can not happen with our current master branch.
Just updating to the current master should fix the issue.

maurhub, 2016-02-23:
Hmm ok I see. This is my current code state:

    i*f(renew) {*
*        opnSecRq.requestType = UA_SECURITYTOKENREQUESTTYPE_RENEW;*
*        UA_LOG_DEBUG(client->logger, UA_LOGCATEGORY_SECURECHANNEL,
"Requesting to renew the SecureChannel");*
*    }* else {
        opnSecRq.requestType = UA_SECURITYTOKENREQUESTTYPE_ISSUE;
        UA_ByteString_init(&client->channel.clientNonce);
        UA_ByteString_copy(&client->channel.clientNonce,
&opnSecRq.clientNonce);
        opnSecRq.securityMode = UA_MESSAGESECURITYMODE_NONE;
        UA_LOG_DEBUG(client->logger, UA_LOGCATEGORY_SECURECHANNEL,
"Requesting to open a SecureChannel");
    }

Since the renew branch doesn't set the securityMode I think it will get
fixed with the new code as you said.

Thxs!

Stasik0, 2016-02-23:
yea, it was fixed couple of weeks ago

jpfr, 2016-02-23:
Great!
I'll close the issue then for now.
Just reopen it if you need a follow up.

closed 2016-02-23 by jpfr