open62541/open62541 #549: Compare/Benchmark OPC UA Binary Encoding with ProtoBuf, ASN1, Thrift, ...

jpfr, 2016-02-27:
There is a nice Wikipedia page that compares binary data encoding formats.
https://en.wikipedia.org/wiki/Comparison_of_data_serialization_formats

OPC UA is currently missing from the list. It would be nice to add the OPC UA binary format to the table. Also, with our implementation, the OPC UA binary encoding might compare favorably in terms of raw speed.

This is up for grabs. Just say in a comment if you want to
- add the OPC UA binary encoding to the comparison table on the Wikipedia page
- do a feature comparison with another binary encoding format that you are interested in (this can be added e.g. to our github wiki and later as an annex to the open62541 docs themselves)
- write short example benchmarks where the same datatype is encoded in different formats

retzbeats, 2016-09-14:
Hi I found really interesting the open62541 project. I am thinking in doing research in semantic interoperability with OPC-UA and would like to know what open issues do you have in this topic.

Thanks and regards

jpfr, 2016-09-14:
That depends on what you mean by semantic interoperability.

If you want to implement systems, by using OPC UA with variabletypes / objecttypes in the information model, you can already infer more "meaning" than from a list of standard variables only.

If you are interested in the "Query" service of OPC UA (a bit like SQL and SPARQL), we would appreciate any ideas (also conceptional) on this topic. We currently still lack the deep understanding of Query to do an implementation.

If you mean "semantics" in an academic sense, you can have a look at @markusgraube 's work [1].

[1] Graube, M.; Hladik, J.; Urbas, L. (accepted). Integrating Industrial Middleware in Linked Data Collaboration Networks. IEEE Conf. Emerging Technologies & Factory Automation (ETFA)

jpfr, 2016-09-14:
And welcome to this forum @retzbeats !

retzbeats, 2016-09-19:
Thanks a lot Julius, I am checking everything you said I will let you know
what I found and maybe botter you again with questions. ;)

closed 2021-09-23 by jpfr