open62541/open62541 #550: manuallySendPublishRequest variable duration time

Aleskey78, 2016-02-29:
Hello,

I've written a small console application that calls a UA_Client_Subscriptions_manuallySendPublishRequest function each 100ms with the following parameters:
        UA_SubscriptionSettings settings = UA_SubscriptionSettings_standard;
        settings.requestedPublishingInterval = 0.0;
        settings.requestedMaxKeepAliveCount = 10;
        settings.requestedLifetimeCount  = settings.requestedMaxKeepAliveCount * 3;
        settings.maxNotificationsPerPublish = 1024;

At the beginning the duration of a call is about  900ms.
And now the gag: if I hold for a while (3 secs) a scrollbar of the console with a debug output -> the function takes only ca. 5ms for a PublishRequest/Response!

A Wireshark dump is attached.

Thanks
[publishReqResTime.zip](https://github.com/open62541/open62541/files/151281/publishReqResTime.zip)

Aleskey78, 2016-03-01:
Actually it’s the same if you’ll try to step through the UA_Client_Subscriptions_manuallySendPublishRequest in debug mode.

jpfr, 2016-03-04:
bump @ichrispa 
That looks like a blocker for 0.2 in the subscriptions.

Stasik0, 2016-03-04:
can it be some timing issue, since the program is paused during debug or "scrolling"?

Aleskey78, 2016-03-04:
This “scrolling” has however no effect on Read/Write operations that I had to use as workaround for polling.

ichrispa, 2016-04-05:
Is there a reproducable test case? I still use subscriptions daily and have not noticed this behavior before.

jpfr, 2016-05-09:
Hey @Aleskey78 ,

the subscriptions part was refactored. It should work fine now.

For subscriptions, the server may delay an answer depending on whether data changes and the maximum keepalive frequency. In the standard settings (UA_SubscriptionSettings_standard) for our client, the server always tries to send a publish response right away with either a datachangenotification or a keepalive.

Can you check with master after commit e4ff940 and tell us if we can close the issue?

Aleskey78, 2016-05-13:
Hi @jpfr,

I've just tested 768b261 and it's showing same issue with my settings:
```
        UA_SubscriptionSettings settings = UA_SubscriptionSettings_standard;
        settings.requestedPublishingInterval = 0.0;
        settings.requestedMaxKeepAliveCount = 10;
        settings.requestedLifetimeCount  = settings.requestedMaxKeepAliveCount * 3;
        settings.maxNotificationsPerPublish = 1024;
```

Aleskey78, 2016-05-13:
@jpfr,

just tested a little bit more :) and it seems that the issue is solved.
So after holding a scroolbar for a 3 - 5 seconds 'UA_Client_Subscriptions_manuallySendPublishRequest' function takes less than a 3ms (it was about 900ms at the beginning). BUT it still receives notifications and after the first one the duration of a call goes back to 900ms.
It looks a bit strange for me, but it works :)

closed 2016-06-06 by jpfr