open62541/open62541 #552: Asynchronous  PublishRequest/Response operations

Aleskey78, 2016-03-01:
Hello,

I would really like to use SendPublishRequest asynchronously, non-blocking to Read/Write calls.
Is there any way to do it? Some kind of message loop / dispatcher?

Thanks!

jpfr, 2016-03-04:
Hey @Aleskey78,

not yet. Eventually, I'd like to have the same event-based dispatch in client and server. But that is post 0.2 if nobody steps up.

Best, Julius

Stasik0, 2016-03-04:
the first step would be this one: https://github.com/open62541/open62541/issues/279 making a client to have its own thread

jpfr, 2016-05-09:
This feature was implemented server-side. Client-side async will follow after 0.2.

jpfr, 2017-07-05:
Implemented in bffefa4cd0331d75a8ed46698ab1a959217c14f9.

closed 2017-07-05 by jpfr