open62541/open62541 #556: Wrong debug message?

ronnydierckx, 2016-03-11:
In UA_SessionManager_getSession, the message prints token->identifier.numeric even if the token identifierType is guid.

Stasik0, 2016-03-11:
yes, this is common to the most debug messages - we somehow assume that most ids are numerical in debug outputs. @open62541/open62541-core this seems to be a minor problem, 0.2. or 0.3.? We need some generic print function for node ids that can be plugged into debug messages

jpfr, 2016-03-13:
The nodeid type is always numeric when we refer to a standard-defined data type. That's the case for 90% of the debut messages in question.

Let's just fix this one instance where it can be guid. Are there any cases where the nodeid is unknown in advance?

closed 2016-05-09 by jpfr