open62541/open62541 #559: UA_MIXED_ENDIAN does not compile

ronnydierckx, 2016-03-15:
Hi,

In Float_decodeBinary()  *offset += 4  should be *dst += 4

Also

```
Double_decodeBinary(UA_ByteString const *src, bufpos pos, buflen len) {
    if(*offset + sizeof(UA_Double) > src->length)
```
should be:
```
Double_decodeBinary(bufpos pos, bufend end, UA_UInt64 *dst) {
    if(*pos + sizeof(UA_Double) > end)
```
Best regards,

Ronny

Stasik0, 2016-03-15:
Hi Ronny, tnx for the bug, could you try my latest commit to verify it works (i have fixed some implcit conversions, but have no mixed-endian system at hands to test right now)

jpfr, 2016-03-15:
@Stasik0 The commit looks good.

In addition, we should fix two more things.

**Double-Encoding**
```// Todo: Architecture agnostic de- and encoding, like float has it```

IEEE float memory layouts can be very strange. And a simple endianness-swap doesn't cut it. For example, Linux has a funky big-endian layout of floats: https://android.googlesource.com/platform/prebuilts/gcc/linux-x86/host/x86_64-linux-glibc2.7-4.6/+/jb-dev/sysroot/usr/include/ieee754.h

**Array Encoding**
Imho we should be intelligent when setting the `.zero_copyable` attribute of the type description.
And in principle allow memcpy-encoding on all architectures. Then, at least `Byte` is always covered. :smile:
Let's just use reasonable defaults, macro-switch if possible, and **fail a unit test** if the architecture doesn't behave well.

Stasik0, 2016-03-15:
open an issue?

ronnydierckx, 2016-03-16:
Hi,

Maybe you could look at the sqlite source code, it has a simple solution for mixed endian doubles if SQLITE_MIXED_ENDIAN_64BIT_FLOAT is defined.

Best regards,

Ronny

Stasik0, 2016-03-16:
do you mean this stuff? https://github.com/mackyle/sqlite/blob/db72d7c40d4992aee4b5dc02169629754351d36d/src/vdbeaux.c#L3223 here a union is used to swap the halfs, right?

jpfr, 2016-03-16:
Swapping 32bit components may not be enough.
As the ieee754.h posted above shows, there may be more differences between little/big/mixed endian than just moving the last byte first or swapping halves.

For example here on x86 on Linux, the sign-bit can be either at the very front or the very end.
```c
    struct
      {
#if	__BYTE_ORDER == __BIG_ENDIAN
	unsigned int negative:1;
	unsigned int exponent:8;
	unsigned int mantissa:23;
#endif				/* Big endian.  */
#if	__BYTE_ORDER == __LITTLE_ENDIAN
	unsigned int mantissa:23;
	unsigned int exponent:8;
	unsigned int negative:1;
#endif				/* Little endian.  */
      } ieee;
```

Imho we should implement a solution that extracts the sign/exponent/mantissa using nothing but standard C tools and no union magic. Like here: https://github.com/axboe/fio/blob/master/lib/ieee754.c
And then provide a fastpath for known architectures as a special case.

jpfr, 2016-03-16:
And note that our current implementation is **not cross-platform**. We still expect the bits to be at specific places. I guess that's for some ARM7 architecture that was used at the time.
```c
    mantissa = (UA_Float)(**pos & 0xFF); // bits 0-7
    mantissa = (mantissa / 256.0) + (UA_Float)((*pos)[1] & 0xFF); // bits 8-15
    mantissa = (mantissa / 256.0) + (UA_Float)((*pos)[2] & 0x7F); // bits 16-22
    biasedExponent = ((*pos)[2] & 0x80) >> 7; // bits 23
    biasedExponent |= ((*pos)[3] & 0x7F) << 1; // bits 24-30
    sign = ((*pos)[3] & 0x80) ? -1.0 : 1.0; // bit 31
```

jpfr, 2016-03-29:
Closed. There is no more `UA_MIXED_ENDIAN` now. Instead, we have a fallback to a generic approach that always works.

closed 2016-03-29 by jpfr