open62541/open62541 #560: Cannot restart server

johnml1135, 2016-03-16:
In the code below (hacked from examples\server.cpp), I start a server and close it and start it again.  Using the Unified Automation UaExpert, I am unable to connect to the server after it was essentially restarted.  Is there a graceful way to stop and restart the server?

```c

#include <signal.h>
#include <stdio.h>

#ifdef UA_NO_AMALGAMATION
# include "ua_types.h"
# include "ua_server.h"
# include "logger_stdout.h"
# include "networklayer_tcp.h"
#else
# include "open62541.h"
#endif

UA_Boolean running = UA_TRUE;
UA_Logger logger = Logger_Stdout;

static void stopHandler(int sign) {
	UA_LOG_INFO(logger, UA_LOGCATEGORY_SERVER, "received ctrl-c");
	running = 0;
}

void doServer(void) {
	signal(SIGINT, stopHandler); /* catches ctrl-c */

	UA_ServerConfig config = UA_ServerConfig_standard;
	UA_ServerNetworkLayer nl = UA_ServerNetworkLayerTCP(UA_ConnectionConfig_standard, 16664);
	config.logger = Logger_Stdout;
	config.networkLayers = &nl;
	config.networkLayersSize = 1;
	UA_Server *server = UA_Server_new(config);

	// add a variable node to the adresspace
	UA_VariableAttributes attr;
	UA_VariableAttributes_init(&attr);
	UA_Int32 myInteger = 42;
	UA_Variant_setScalarCopy(&attr.value, &myInteger, &UA_TYPES[UA_TYPES_INT32]);
	attr.description = UA_LOCALIZEDTEXT_ALLOC("en_US", "the answer");
	attr.displayName = UA_LOCALIZEDTEXT_ALLOC("en_US", "the answer");
	UA_NodeId myIntegerNodeId = UA_NODEID_STRING_ALLOC(1, "the.answer");
	UA_QualifiedName myIntegerName = UA_QUALIFIEDNAME_ALLOC(1, "the answer");
	UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
	UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
	UA_Server_addVariableNode(server, myIntegerNodeId, parentNodeId,
		parentReferenceNodeId, myIntegerName,
		UA_NODEID_NULL, attr, NULL, NULL);

	/* allocations on the heap need to be freed */
	UA_VariableAttributes_deleteMembers(&attr);
	UA_NodeId_deleteMembers(&myIntegerNodeId);
	UA_QualifiedName_deleteMembers(&myIntegerName);

	UA_StatusCode retval = UA_Server_run(server, &running);
	UA_Server_delete(server);
	nl.deleteMembers(&nl);
}

void doServerShort(void) {
	signal(SIGINT, stopHandler); /* catches ctrl-c */

	UA_ServerConfig config = UA_ServerConfig_standard;
	UA_ServerNetworkLayer nl = UA_ServerNetworkLayerTCP(UA_ConnectionConfig_standard, 16664);
	config.logger = Logger_Stdout;
	config.networkLayers = &nl;
	config.networkLayersSize = 1;
	UA_Server *server = UA_Server_new(config);

	// add a variable node to the adresspace
	UA_VariableAttributes attr;
	UA_VariableAttributes_init(&attr);
	UA_Int32 myInteger = 42;
	UA_Variant_setScalarCopy(&attr.value, &myInteger, &UA_TYPES[UA_TYPES_INT32]);
	attr.description = UA_LOCALIZEDTEXT_ALLOC("en_US", "the answer");
	attr.displayName = UA_LOCALIZEDTEXT_ALLOC("en_US", "the answer");
	UA_NodeId myIntegerNodeId = UA_NODEID_STRING_ALLOC(1, "the.answer");
	UA_QualifiedName myIntegerName = UA_QUALIFIEDNAME_ALLOC(1, "the answer");
	UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
	UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
	UA_Server_addVariableNode(server, myIntegerNodeId, parentNodeId,
		parentReferenceNodeId, myIntegerName,
		UA_NODEID_NULL, attr, NULL, NULL);

	/* allocations on the heap need to be freed */
	UA_VariableAttributes_deleteMembers(&attr);
	UA_NodeId_deleteMembers(&myIntegerNodeId);
	UA_QualifiedName_deleteMembers(&myIntegerName);

	UA_StatusCode retval;
	retval = UA_Server_run_startup(server);
	int sample = 0;
	for (int i = 0; i < 100; i++) {
		retval = UA_Server_run_iterate(server, true);
	}
	UA_Server_delete(server);
	nl.deleteMembers(&nl);
}

int main(int argc, char** argv) {
	doServerShort();
	doServer();

	return 0;
}
```

johnml1135, 2016-03-16:
Actually, I solved it myself.  The shutdown procedure for the network for all the examples is wrong.  In the description for the UA_ServerNetworkLayer stop function pointer it says:
```c
     * Closes the network connection and returns all the jobs that need to be finished 
     * before the network layer can be safely deleted.
```
Assuming that killing the server stops all the jobs, this code works for me:
```C
	UA_Server_delete(server);
	//This is to shut down the network before you delete it.  
	UA_Job *jobs;
	nl.getJobs(&nl, &jobs, 100);
	nl.stop(&nl, &jobs);
	//now you can delete the network layer.
	nl.deleteMembers(&nl);
```

jpfr, 2016-03-17:
You do not shut down the server. Please call `UA_Server_run_shutdown` before you delete the server. That takes care of the network layers.

See here: https://github.com/open62541/open62541/blob/master/src/server/ua_server_worker.c#L644

Stasik0, 2016-03-17:
but this line is missing in the examples in this case, as @johnml1135 remarked

jpfr, 2016-03-17:
All examples use `UA_Server_run` that does the shutdown internally.
See here: https://github.com/open62541/open62541/blob/master/src/server/ua_server_worker.c#L673

@johnml1135 does not use the same approach as the examples and runs _startup and _iterate manually.

closed 2016-03-17 by jpfr

johnml1135, 2016-03-17:
Can there be an example made of using the iterate server so as when a person uses it (aka, me) they open ans shut it down properly?

jpfr, 2016-03-17:
Good idea.
The example was added in commit f2b5b84c0fbf3c36cdd6f384bd2631c25d0773c9.

juangburgos, 2019-06-10:
Hi guys,

I am having a similar issue. I would like to stop the server and then start it again **without** having to call `UA_Server_delete`. I do the regular loop:

```c
UA_StatusCode retval = UA_Server_run_startup(server);
while(running) 
{
   UA_Server_run_iterate(server, waitInternal);
}
UA_Server_run_shutdown(server);
```

But if I start the server again using `UA_Server_run_startup`, after calling `UA_Server_run_shutdown`, then the client cannot connect anymore. I keep getting the ` "Socket select failed with %s"` error in `ServerNetworkLayerTCP_listen`. 

It only works after deleting and recreating the server, but that is something I would like to avoid.

Thank you in advance,