open62541/open62541 #567: Question: Expected API Changes between now and 0.2

LEvertz, 2016-03-23:
Hey There,

right now, we have integrated a version 0.1 open62541-System into our own runtime environment. For planned enhancements (including the use of the call-service) we will update our integration soon. I think it makes sense to work with 0.2 as sonn as it is available and I do not want to make adaptations twice. So my question is:
Which parts of the API are expected not to change between now and 0.2? Or in other words, where should I start integration and where should I wait for changes to happen?

Stasik0, 2016-03-23:
From what I know we will change the signature of UA_AddExternalNamespace, don't we @FlorianPalm ?

FlorianPalm, 2016-03-23:
Yes, we will leave out the namespace id (UA_UInt16) in UA because it's the server which takes care of assigning these ids:
```C
UA_Server_addExternalNamespace(UA_Server *server, UA_UInt16 namespaceIndex, const UA_String *url,
                               UA_ExternalNodeStore *nodeStore);
````
becomes:
```C
UA_Server_addExternalNamespace(UA_Server *server, const UA_String *url,
                               UA_ExternalNodeStore *nodeStore);
````

LEvertz, 2016-03-23:
OK, leaving out is easy :-)
Anything on the more functional side?
Btw, forgot to say explicitliy: I was asking about the client-side as well.

ichrispa, 2016-03-23:
Hi everyone,

quick question: is it intended to significantly change any of the other high-level functions? I am having a bit of a disagreement with jpfr concerning instanatition callbacks, but other then that it would be very favorable to keep for example the current addNode calls intact.

I would consider it extremely beneficial both for us as well as the users if we could freeze our API at some (very near future) time to avoid high reimplementation overheads. Especially wrappers for high level languages are a pain to adapt every 3 months.

Best regards,

ichrispa

ichrispa, 2016-05-20:
Given #598 I would say this issue is done for?

LEvertz, 2016-05-20:
Yep. It's done.

closed 2016-05-20 by LEvertz