open62541/open62541 #568: VS2015 Code Analysis issues

johnml1135, 2016-03-24:
When running Visual Studio 2015's code analysis, a few issues popped up.  Are any of these actual bugs or just superfluous warnings?

Severity	Code	Description	Project	File	Line	Suppression State
Warning	C4146	unary minus operator applied to unsigned type, result still unsigned	OpcUaSvr	open62541.c	13859	
Warning	C6102	Using 'ft' from failed function call at line '2782'.	OpcUaSvr	open62541.c	2784	Active
Warning	C6308	'realloc' might return null pointer: assigning null pointer to 'server->namespaces', which is passed as an argument to 'realloc', will cause the original memory block to be leaked.	OpcUaSvr	open62541.c	6576	Active
Warning	C28182	Dereferencing NULL pointer. 'server->namespaces' contains the same NULL value as 'realloc()`6576' did. 	OpcUaSvr	open62541.c	6578	Active
Warning	C6011	Dereferencing NULL pointer 'ch'. 	OpcUaSvr	open62541.c	8092	Active
Warning	C6255	_alloca indicates failure by raising a stack overflow exception.  Consider using _malloca instead.	OpcUaSvr	open62541.c	8177	Active
Warning	C6255	_alloca indicates failure by raising a stack overflow exception.  Consider using _malloca instead.	OpcUaSvr	open62541.c	8226	Active
Warning	C6308	'realloc' might return null pointer: assigning null pointer to 'matchingTw', which is passed as an argument to 'realloc', will cause the original memory block to be leaked.	OpcUaSvr	open62541.c	8725	Active
Warning	C6255	_alloca indicates failure by raising a stack overflow exception.  Consider using _malloca instead.	OpcUaSvr	open62541.c	9521	Active
Warning	C6031	Return value ignored: 'WSAStartup'.	OpcUaSvr	open62541.c	13595	Active
Warning	C6386	Buffer overrun while writing to 'hostname':  the writable size is '512' bytes, but '-1' bytes might be written.	OpcUaSvr	open62541.c	13690	Active
Warning	C6031	Return value ignored: 'WSAStartup'.	OpcUaSvr	open62541.c	13695	Active
Warning	C4146	unary minus operator applied to unsigned type, result still unsigned	OpcUaSvr	open62541.c	13830

jpfr, 2016-03-24:
They look mostly fine.
But: We aim for zero warnings in the library.
Even if most of these warnings are probably not critical, we should get rid of them so that "true" bugs are more easily detected.

On gcc / clang, we achieve zero warnings already. Otherwise, the standard build will just abort.
On MSVC, this requires some volunteer work before we can activate "error on warning".

So if you come across a fix for some of these warnings, feel free to prepare a pull request.

jpfr, 2016-03-29:
Closed for now.
You can still reference the issue when a commit comes in.

closed 2016-03-29 by jpfr